evcc-io/evcc #12967: Huawei Sun2000 / Luna 2000

naseweiss99, 2024-03-15:
### Describe the bug

Wenn das Laden per EVCC mit "schnell" gestartet wird, geht der Huawei WR in den Lademodus - dabei wird auch aus dem Netz der Luna 2000 Hausakku geladen. Das sollte so nicht sein.

Dabei wird im Sun 2000 der Punkt Forced charge enabled. Der ist für AC Laden. PV Laden hat mit diesem Punkt nichts zu tun.

D.h. Akku Auto und Akku Haus werden bei zu wenig Sonnenstrom aus dem Netz geladen. 

### Steps to reproduce

1.Laden Schnell Starten - 
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

# unique installation id
plant: xx

interval: 30s # control cycle interval

#sponsortoken: 
# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: huawei-dongle-powersensor 
  id: 1  
  host: 192.168.178.40  
  port: 9000  
  usage: grid  
  modbus: tcpip  
  timeout: 31s  
  name: grid1
- type: template
  template: huawei-dongle-powersensor 
  id: 1  
  host: 192.168.178.40  
  port: 9000  
  usage: pv  
  modbus: tcpip  
  timeout: 31s  
  name: pv2
- type: template
  template: huawei-dongle-powersensor 
  id: 1  
  host: 192.168.178.40  
  port: 9000  
  usage: battery  
  modbus: tcpip  
  timeout: 31s  
  name: battery3

chargers:
- type: template
  template: go-e-v3 
  host: 192.168.178.102  
  name: wallbox5

vehicles:
- type: template
  template: citroen 
  title: eC4  
  user: 
  vin: 
  capacity: 45  
  name: ev4

loadpoints:
- title: Aussen
  charger: wallbox5
  mode: off
  phases: 0
  

site:
  title: Mein Zuhause
  residualPower: 100
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
```

### Log details

```text
hab ich momentan nicht reiche ich notfalls nach aber das Forced charge ist nicht ok.
```

### What type of operating system are you running?

Windows

### Version

124.10

naseweiss99, 2024-03-15:
Achso, Evcc läuft auf Debian

premultiply, 2024-03-15:
Nicht reproduzierbar und ohne Log nicht analysierbar.

closed 2024-03-15 by premultiply

naseweiss99, 2024-03-15:
danke, ich kann das momentan nicht nachstellen, aber Hauptsache der Fall es ist closed.
 #
Auch ohne Log, der Punkt AC Laden darf nicht im Huawei eingeschaltet werden, egal was passiert. Dann Lädt der den Hausakku aus teuerem Netzstrom.
Es hängt m.e. mit den Schiebern für Batterie zusammen.

mucki12, 2024-03-15:
Kann ich nicht nachstellen. In diesem Fall müsste evcc ja auch von normal/hold auf charge stellen, was imho noch gar nicht umgesetzt ist.

Ladevorgang Batteriesperre aktiv:
[image]

Ladevorgang Batteriesperre inaktiv:
[image]

naseweiss99, 2024-03-15:
Danke,
ich habe in der Huawei Eu5 Website geschaut und da war "Forced charge " gesetzt.
Habe das zurückgesetzt, aber das kam sofort wieder. 
Ich habe diese Topics nicht in der Heimautomation (FHEM) benutzt.

Ich werde das nächste Woche nochmal prüfen.

mucki12, 2024-03-15:
Viel Erfolg. Ich nutze ziemlich wilde Ladelogiken für den LUNA. Das eine Ladung durch evcc angestossen wurde konnte ich allerdings noch nie beobachten.