evcc-io/evcc #12974: Mehrere Shelly 3EM für die Grid Ermittlung

MilkoKA, 2024-03-15:
Ist es möglich 3 Shelly 3em einzutragen und in Summe den Grid Wert zu ermitteln?

Ich habe in meinen Zählerkasten nicht genügend Platz um ein Shell 3em mit den Messzangen anzuschließen.

Vom Zählerkasten geht es gleich in 3 Unterverteilungen (Wohnung, Garage und Keller).
In diesen Unterverteilungen habe ich jeweils ein Shelly 3em verbaut
und kann dann in Summe meinen Netzbezug sehen ( Sheyy App und Solaranzeige)

in evcc  bekomme ich das nicht konfiguriert.

Danke für eine Unterstützung

mdkeil, 2024-03-15:
Aus dem Bauch heraus würde ich sagen über ein custom meter + calc plugin.

https://docs.evcc.io/docs/reference/configuration/meters#custom
https://docs.evcc.io/docs/reference/plugins#calc-nur-lesen

VolkerK62, 2024-03-15:
Netzbezug ja, aber siehst du dann auch die Einspeisung? Die wird ja nicht über eine der Unterverteilungen laufen, oder?

mdkeil, 2024-03-15:
> Netzbezug ja, aber siehst du dann auch die Einspeisung? Die wird ja nicht über eine der Unterverteilungen laufen, oder?

das ist natürlich ein Argument.. :) 

Wenn nicht genug Platz im Zähler, wie sieht es denn aus, ggfs. direkt den Zähler z.B. über ein IR-Lesekopf auszulesen?

MilkoKA, 2024-03-15:
> > Netzbezug ja, aber siehst du dann auch die Einspeisung? Die wird ja nicht über eine der Unterverteilungen laufen, oder?
> 
> das ist natürlich ein Argument.. :)
> 
> Wenn nicht genug Platz im Zähler, wie sieht es denn aus, ggfs. direkt den Zähler z.B. über ein IR-Lesekopf auszulesen?

Ich habe noch einen alten Fer­ra­ris­zäh­ler. Wenn der mal ausgetauscht wird habe das Problem nicht mehr.

MilkoKA, 2024-03-15:
> Aus dem Bauch heraus würde ich sagen über ein custom meter + calc plugin.
> 
> https://docs.evcc.io/docs/reference/configuration/meters#custom https://docs.evcc.io/docs/reference/plugins#calc-nur-lesen
> 
> wie der shelly-3em ausgelesen wird, findest Du hier: https://github.com/evcc-io/evcc/blob/master/templates/definition/meter/shelly-3em.yaml

Ich werd mich damit mal befassen. Vieleicht bekomm ich das ja hin.

Danke für den Hinweis

mdkeil, 2024-03-15:
> > Netzbezug ja, aber siehst du dann auch die Einspeisung? Die wird ja nicht über eine der Unterverteilungen laufen, oder?
> 
> Die Solarerzeugung läuft über die Garage und wird direkt hinter der Solaranlage (keines Balkonkraftwerk) gemessen

bringt dich dann aber auch nicht weiter, da evcc damit nichts anfangen kann.. evcc versucht ja bei Überschuss (also bei Einspeisung) zu laden bzw. Geräte zu aktivieren.. Deine erzeugte Leistung ist ja nicht gleich Überschuss, da ja noch der Hausverbrauch davon ab geht..