evcc-io/evcc #12979: iOS: titel is overlapped by date, battery and WLAN iconc

eweri, 2024-03-15:
### Describe the bug

With 0.124.10 the title of evcc is overlapping with iOS headline, it is difficult or nearly impossible to open menu 

[image]
[image]

### Steps to reproduce

Install new 0.124.10 

### Configuration details

```yaml
- no configuration problem
```

### Log details

```text
- no logging
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.124.10

mucki12, 2024-03-15:
https://github.com/evcc-io/evcc/issues/12941

naltatis, 2024-03-15:
Yes, looks like a duplicate. Try reinstalling the homescreen bookmark.

closed 2024-03-15 by naltatis