evcc-io/evcc #12980: Geladene kWh nach Ladevorgang teilweise falsch berechnet

julred, 2024-03-15:
### Describe the bug

Hallo zusammen,

nach einigen Ladevorgängen habe ich eine falsch berechnete Gesamtladung in der Historie, als auch auf dem Startscreen unten links:
[image]

### Steps to reproduce

Nicht immer reproduzierbar

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: *****

interval: 10s # control cycle interval

sponsortoken: ***

# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

mqtt:
  broker: 192.168.178.15:1883
  topic: evcc # root topic for publishing, set empty to disable publishing
  # clientid: foo
  user: mqtt
  password: ***

meters:
- type: template
  template: sma-home-manager 
  usage: grid  
  host: 192.168.178.129
  name: grid1
- type: template
  template: sma-hybrid 
  id: 3  
  host: 192.168.178.77  
  port: 502  
  usage: pv  
  modbus: tcpip  
  name: pv2
- type: template
  template: sma-inverter 
  usage: pv  
  host: 192.168.178.128  
  password: ***
  name: pv3
- type: template
  template: sma-hybrid 
  id: 3  
  host: 192.168.178.77  
  port: 502  
  usage: battery  
  modbus: tcpip  
  name: battery4
- type: template
[... 293 lines cut ...]
Mar 15 13:27:40 evcc evcc[8621]: [lp-1  ] DEBUG 2024/03/15 13:27:40 charger status: B
Mar 15 13:27:40 evcc evcc[8621]: [lp-1  ] DEBUG 2024/03/15 13:27:40 pv charge current: 9.62A = 0A + 9.62A (-4425W @ 2p)
Mar 15 13:27:40 evcc evcc[8621]: [lp-1  ] DEBUG 2024/03/15 13:27:40 max charge current: 9.62A
Mar 15 13:27:50 evcc evcc[8621]: [site  ] DEBUG 2024/03/15 13:27:50 ----
Mar 15 13:27:50 evcc evcc[8621]: [lp-1  ] DEBUG 2024/03/15 13:27:50 charge power: 0W
Mar 15 13:27:50 evcc evcc[8621]: [site  ] DEBUG 2024/03/15 13:27:50 pv power: 4819W
Mar 15 13:27:50 evcc evcc[8621]: [site  ] DEBUG 2024/03/15 13:27:50 battery soc: 100%
Mar 15 13:27:50 evcc evcc[8621]: [site  ] DEBUG 2024/03/15 13:27:50 battery power: 0W
Mar 15 13:27:50 evcc evcc[8621]: [site  ] DEBUG 2024/03/15 13:27:50 grid meter: -4495W
Mar 15 13:27:50 evcc evcc[8621]: [site  ] DEBUG 2024/03/15 13:27:50 grid powers: [-1449 -1586 -1460]W
Mar 15 13:27:50 evcc evcc[8621]: [site  ] DEBUG 2024/03/15 13:27:50 grid currents: [-6.08 -6.54 -6.16]A
Mar 15 13:27:50 evcc evcc[8621]: [site  ] DEBUG 2024/03/15 13:27:50 site power: -4495W
Mar 15 13:27:50 evcc evcc[8621]: [lp-1  ] DEBUG 2024/03/15 13:27:50 charge total import: 517.600kWh
Mar 15 13:27:50 evcc evcc[8621]: [lp-1  ] DEBUG 2024/03/15 13:27:50 charger status: B
Mar 15 13:27:50 evcc evcc[8621]: [lp-1  ] DEBUG 2024/03/15 13:27:50 pv charge current: 9.77A = 0A + 9.77A (-4495W @ 2p)
Mar 15 13:27:50 evcc evcc[8621]: [lp-1  ] DEBUG 2024/03/15 13:27:50 max charge current: 9.77A
```

### What type of operating system are you running?

Linux

### Version

0.124.10

andig, 2024-03-16:
Was heisst „falsch“? Gibts ein debug log mit db: trace wo wir das sehen können? Ohne ist nix zu machen.

julred, 2024-03-16:
Hätte gedacht, dass aus dem Log diese Zeile relevant sein könnte:
`Mar 15 13:26:30 evcc evcc[8621]: [lp-1  ] DEBUG 2024/03/15 13:26:30 final charge energy: 1.04e+03kWh`

Ich habe hier mal ein log - welches ich aber grade erstellt habe, ohne das geladen wird etc.
Hilft das weiter?

```
[main  ] INFO 2024/03/16 09:46:26 using config file: /etc/evcc.yaml
[main  ] INFO 2024/03/16 09:46:26 starting ui and api at :7070
[db    ] INFO 2024/03/16 09:46:26 using sqlite database: /root/.evcc/evcc.db
[db    ] TRACE 2024/03/16 09:46:26 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2024/03/16 09:46:26 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/03/16 09:46:26 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/03/16 09:46:26 SELECT * FROM `settings` 6 <nil>
[mqtt  ] INFO 2024/03/16 09:46:26 connecting evcc-1242387417 at tcp://192.168.178.15:1883
[db    ] TRACE 2024/03/16 09:46:26 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2024/03/16 09:46:26 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2024/03/16 09:46:26 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2024/03/16 09:46:26 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/03/16 09:46:26 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/03/16 09:46:26 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2024/03/16 09:46:26 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
[db    ] TRACE 2024/03/16 09:46:26 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/03/16 09:46:26 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_configs_details"" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" OR sql LIKE "%CONSTRAINT        fk_configs_details      %") -1 <nil>
[db    ] TRACE 2024/03/16 09:46:26 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
[db    ] TRACE 2024/03/16 09:46:26 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_devices_details"" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT        fk_devices_details      %") -1 <nil>
[db    ] TRACE 2024/03/16 09:46:26 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%""device_id"" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "%  device_id       %") -1 <nil>
[sma   ] TRACE 2024/03/16 09:46:26 send 192.168.178.129: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[mqtt  ] DEBUG 2024/03/16 09:46:26 tcp://192.168.178.15:1883 connected
[sma   ] TRACE 2024/03/16 09:46:26 recv 192.168.178.129: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/03/16 09:46:26 new energy meter at 192.168.178.129 - Serial=3009870528
[sma   ] TRACE 2024/03/16 09:46:26 send 192.168.178.129: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/03/16 09:46:27 send discover package
[sma   ] TRACE 2024/03/16 09:46:27 recv 192.168.178.128: [proto.GroupPacketEntry, proto.UnknownPacketEntry, proto.SmaNet2PacketEntry, proto.DiscoveryRequestPacketEntry, proto.DiscoveryIPPacketEntry, proto.UnknownPacketEntry, proto.UnknownPacketEntry, proto.UnknownPacketEntry]
[sma   ] TRACE 2024/03/16 09:46:27 recv 192.168.178.129: [proto.GroupPacketEntry, proto.UnknownPacketEntry, proto.SmaNet2PacketEntry, proto.DiscoveryRequestPacketEntry, proto.DiscoveryIPPacketEntry, proto.UnknownPacketEntry, proto.UnknownPacketEntry, proto.UnknownPacketEntry]
[sma   ] TRACE 2024/03/16 09:46:27 send 192.168.178.129: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/03/16 09:46:27 recv 192.168.178.77: [proto.GroupPacketEntry, proto.UnknownPacketEntry, proto.SmaNet2PacketEntry, proto.DiscoveryRequestPacketEntry, proto.DiscoveryIPPacketEntry, proto.UnknownPacketEntry, proto.UnknownPacketEntry, proto.UnknownPacketEntry]
[sma   ] TRACE 2024/03/16 09:46:27 recv 192.168.178.129: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/03/16 09:46:27 new inverter at 192.168.178.129 - Serial=3009870528
[sma   ] TRACE 2024/03/16 09:46:27 found device 3009870528 at 192.168.178.129
[sma   ] TRACE 2024/03/16 09:46:27 send 192.168.178.128: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/03/16 09:46:27 recv 192.168.178.128: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/03/16 09:46:27 new inverter at 192.168.178.128 - Serial=3006546537
[sma   ] TRACE 2024/03/16 09:46:27 found device 3006546537 at 192.168.178.128
[sma   ] TRACE 2024/03/16 09:46:27 send 192.168.178.77: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[... cut ...]
[sma   ] TRACE 2024/03/16 09:48:59 recv 192.168.178.129: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
^C[db    ] TRACE 2024/03/16 09:49:00 INSERT INTO `settings` (`key`,`value`) VALUES ("savings.started","2023-04-28T11:43:09+02:00"),("savings.gridCharged","0"),("savings.gridCost","0"),("savings.gridSavedCost","0"),("savings.selfConsumptionCharged","0"),("savings.selfConsumptionCost","0"),("lp1.minCurrent","6"),("lp1.maxCurrent","16"),("lp1.phasesConfigured","3") ON CONFLICT (`key`) DO UPDATE SET `value`=`excluded`.`value` 9 <nil>
```

andig, 2024-03-16:
Sieht man denn wo die Werte her kommen? Wenn du nur db: trace machst wirds übersichtlicher.

julred, 2024-03-16:
Die Zählerwerte kommen ja vom angeschlossenen cfos MID Zähler via Modbus TCP. Gibts hier irgendwelche Konvertierungs- oder Timingprobleme? Wenn ich händisch den Zählerstand über Modbus TCP abfrage, funktioniert das alles reibungslos. Dieses Problem tritt ja auch nicht nach jedem Ladevorgang auf. Aber ab und zu schlägt der Fehler doch nochmal zu.,

```
root@evcc:~# evcc -l debug
[main  ] INFO 2024/03/16 14:02:46 evcc 0.124.10
[main  ] INFO 2024/03/16 14:02:46 using config file: /etc/evcc.yaml
[main  ] INFO 2024/03/16 14:02:46 starting ui and api at :7070
[db    ] INFO 2024/03/16 14:02:46 using sqlite database: /root/.evcc/evcc.db
[mqtt  ] INFO 2024/03/16 14:02:46 connecting evcc-1676981679 at tcp://192.168.178.15:1883
[mqtt  ] DEBUG 2024/03/16 14:02:46 tcp://192.168.178.15:1883 connected
[lp-1  ] WARN 2024/03/16 14:02:48 deprecated: minCurrent setting is ignored, please remove
[lp-1  ] WARN 2024/03/16 14:02:48 deprecated: maxcurrent setting is ignored, please remove
[lp-1  ] WARN 2024/03/16 14:02:48 deprecated: phases setting is ignored, please remove
[lp-1  ] DEBUG 2024/03/16 14:02:48 charge total import: 517.600kWh
[site  ] WARN 2024/03/16 14:02:48 battery configured but residualPower is missing or <= 0 (add residualPower: 100 to site), see https://docs.evcc.io/en/docs/reference/configuration/site#residualpower
[semp  ] WARN 2024/03/16 14:02:49 SEMP_BASE_URL unspecified, using http://192.168.178.12:7070 instead
[site  ] INFO 2024/03/16 14:02:49 site config:
[site  ] INFO 2024/03/16 14:02:49   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2024/03/16 14:02:49     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2024/03/16 14:02:49     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2024/03/16 14:02:49     pv 2:      power ✓ energy ✓ currents ✓
[site  ] INFO 2024/03/16 14:02:49     battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2024/03/16 14:02:49   vehicles:
[site  ] INFO 2024/03/16 14:02:49     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2024/03/16 14:02:49 loadpoint 1:
[lp-1  ] INFO 2024/03/16 14:02:49   mode:        pv
[lp-1  ] INFO 2024/03/16 14:02:49   charger:     power ✗ energy ✗ currents ✗ phases ✗ wakeup ✓
[lp-1  ] INFO 2024/03/16 14:02:49   meters:      charge ✓
[lp-1  ] INFO 2024/03/16 14:02:49     charge:    power ✓ energy ✓ currents ✗
[lp-1  ] DEBUG 2024/03/16 14:02:49 phase timer inactive
[lp-1  ] DEBUG 2024/03/16 14:02:49 pv timer inactive
[lp-1  ] INFO 2024/03/16 14:02:49 vehicle updated: unknown -> eGolf
[lp-1  ] DEBUG 2024/03/16 14:02:49 set charge mode: pv
[lp-1  ] DEBUG 2024/03/16 14:02:49 max charge current: 6A
[site  ] WARN 2024/03/16 14:02:49 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2024/03/16 14:02:49 ----
[lp-1  ] DEBUG 2024/03/16 14:02:49 charge power: 0W
[site  ] DEBUG 2024/03/16 14:02:49 pv power: 2832W
[site  ] DEBUG 2024/03/16 14:02:49 battery soc: 100%
[site  ] DEBUG 2024/03/16 14:02:49 battery power: 0W
[site  ] DEBUG 2024/03/16 14:02:49 grid meter: -2510W
[site  ] DEBUG 2024/03/16 14:02:49 grid powers: [-801 -919 -790]W
[site  ] DEBUG 2024/03/16 14:02:49 grid currents: [-3.47 -3.93 -3.49]A
[site  ] DEBUG 2024/03/16 14:02:49 site power: -2510W
[lp-1  ] DEBUG 2024/03/16 14:02:49 charge total import: 517.600kWh
[lp-1  ] DEBUG 2024/03/16 14:02:49 charger status: B
[lp-1  ] DEBUG 2024/03/16 14:02:49 charge start energy: 518kWh
[lp-1  ] INFO 2024/03/16 14:02:49 car connected
[lp-1  ] DEBUG 2024/03/16 14:02:49 pv timer elapse
[lp-1  ] DEBUG 2024/03/16 14:02:49 pv timer inactive
[lp-1  ] DEBUG 2024/03/16 14:02:49 charge total import: 517.600kWh
[lp-1  ] DEBUG 2024/03/16 14:02:49 pv charge current: 5.46A = 0A + 5.46A (-2510W @ 2p)
[lp-1  ] DEBUG 2024/03/16 14:02:49 pv disable timer reset
[lp-1  ] DEBUG 2024/03/16 14:02:49 pv timer inactive
[site  ] DEBUG 2024/03/16 14:02:59 ----
[lp-1  ] DEBUG 2024/03/16 14:02:59 charge power: 0W
[site  ] DEBUG 2024/03/16 14:02:59 pv power: 2672W
[site  ] DEBUG 2024/03/16 14:02:59 battery soc: 100%
[site  ] DEBUG 2024/03/16 14:02:59 battery power: 0W
[site  ] DEBUG 2024/03/16 14:02:59 grid meter: -2511W
[site  ] DEBUG 2024/03/16 14:02:59 grid powers: [-797 -921 -793]W
[site  ] DEBUG 2024/03/16 14:02:59 grid currents: [-3.44 -3.96 -3.5]A
[site  ] DEBUG 2024/03/16 14:02:59 site power: -2511W
[lp-1  ] DEBUG 2024/03/16 14:02:59 charge total import: 517.600kWh
[lp-1  ] DEBUG 2024/03/16 14:02:59 charger status: B
[lp-1  ] DEBUG 2024/03/16 14:02:59 pv charge current: 5.46A = 0A + 5.46A (-2511W @ 2p)
[site  ] DEBUG 2024/03/16 14:03:09 ----
[lp-1  ] DEBUG 2024/03/16 14:03:09 charge power: 0W
[site  ] DEBUG 2024/03/16 14:03:09 pv power: 2828W
[site  ] DEBUG 2024/03/16 14:03:09 battery soc: 100%
[site  ] DEBUG 2024/03/16 14:03:09 battery power: 0W
[site  ] DEBUG 2024/03/16 14:03:09 grid meter: -2503W
[site  ] DEBUG 2024/03/16 14:03:09 grid powers: [-799 -917 -787]W
[site  ] DEBUG 2024/03/16 14:03:09 grid currents: [-3.46 -3.93 -3.48]A
[site  ] DEBUG 2024/03/16 14:03:09 site power: -2503W
[lp-1  ] DEBUG 2024/03/16 14:03:09 charge total import: 517.600kWh
[lp-1  ] DEBUG 2024/03/16 14:03:09 charger status: B
[lp-1  ] DEBUG 2024/03/16 14:03:09 pv charge current: 5.44A = 0A + 5.44A (-2503W @ 2p)
```

andig, 2024-03-16:
Ich versuche es noch einmal. Gibt es ein Log, in dem die Herkunft der falschen Daten bis sie in die GB gescrhieben werden nachvollziehbar ist? Ich mache mal zu, bis es eins gibt...

closed 2024-03-16 by andig

julred, 2024-03-16:
Es gibt kein log dazu. Das ist ja eine schwierige Situation das Ding einzufangen. Oder kann man aus den vergangenen Ladevorgängen ein log reproduzieren wo die Datenherkunft ersichtlich ist? Intern müsste das doch ein Modbus tcp wert sein, der gelesen wird

julred, 2024-03-19:
@andig 
Ich habe die letzten Tage wieder fehlerhafte Werte erhalten.
Hast du einen Tipp wie ich an die gewünschten Daten im Nachgang rankomme?

StefanSchoof, 2024-03-20:
Mit `journalctl` und der Start- und Endzeit, siehe https://docs.evcc.io/docs/guides/setup/#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen

Sweti934, 2024-03-20:
Habe das gleiche Problem leider läuft mein evcc auf Home Assistant daher ist es schwer dienlich Dateien zu generieren. Hab einen Go-E charger und es passiert immer nur bei einem Fahrzeug das ich per MQTT angelegt habe. Das andere der beiden lieferte plausible Daten……

julred, 2024-03-22:
Ich habe hier nochmal einen Ausschnitt aus meiner letzten Ladehistorie eingefügt.
Folgende Passagen finde ich nicht nachvollziehbar:
`Mar 19 15:44:20 evcc evcc[9870]: [lp-1  ] DEBUG 2024/03/19 15:44:20 final charge energy: 536kWh`
`Mar 19 15:43:30 evcc evcc[9870]: [lp-1  ] DEBUG 2024/03/19 15:43:30 final charge energy: 536kWh`

Das Auto ist fertig mit Laden -> evcc versucht daraufhin erneut das Laden zu starten:
`Mar 19 15:36:20 evcc evcc[9870]: [lp-1  ] DEBUG 2024/03/19 15:36:20 wake-up charger, attempts left: 5`
`Mar 19 15:37:50 evcc evcc[9870]: [lp-1  ] DEBUG 2024/03/19 15:37:50 wake-up charger, attempts left: 5`
`Mar 19 15:39:30 evcc evcc[9870]: [lp-1  ] DEBUG 2024/03/19 15:39:30 wake-up charger, attempts left: 5`
`Mar 19 15:41:00 evcc evcc[9870]: [lp-1  ] DEBUG 2024/03/19 15:41:00 wake-up charger, attempts left: 5`
```
Mar 19 15:42:30 evcc evcc[9870]: [lp-1  ] DEBUG 2024/03/19 15:42:30 wake-up charger, attempts left: 5
Mar 19 15:42:40 evcc evcc[9870]: [site  ] DEBUG 2024/03/19 15:42:40 ----
Mar 19 15:42:47 evcc evcc[9870]: [lp-1  ] ERROR 2024/03/19 15:42:47 charge meter: read tcp 192.168.178.12:32888->192.168.178.73:4701: read: connection reset by peer
```
`Mar 19 15:45:00 evcc evcc[9870]: [lp-1  ] DEBUG 2024/03/19 15:45:00 wake-up charger, attempts left: 5`

Insgesamt versucht evcc 6 mal mein Auto wieder aufzuwecken -> ist das wirklich notwendig?
Evcc sollte erkennen, wenn sich das Auto nach einem aktiven Ladevorgang selbstständig "abmeldet" da der Ziel SoC erreicht wurde, dass dann keine Aufweckversuche gestartet werden.
Etwas anderes ist, wenn immer häppchenweise geladen wird, da ab und zu PV Überschuss zur Verfügung steht und das Auto beim nächsten Ladevorgang nicht mehr aufgeweckt werden kann. Dann macht die Aufweckroutine schon sinn. Siehe hier #11958

```
Mar 19 15:35:00 evcc evcc[9870]: [site  ] DEBUG 2024/03/19 15:35:00 ----
Mar 19 15:35:00 evcc evcc[9870]: [lp-1  ] DEBUG 2024/03/19 15:35:00 charge power: 0W
Mar 19 15:35:00 evcc evcc[9870]: [site  ] DEBUG 2024/03/19 15:35:00 pv power: 4153W
Mar 19 15:35:00 evcc evcc[9870]: [site  ] DEBUG 2024/03/19 15:35:00 battery soc: 100%
Mar 19 15:35:00 evcc evcc[9870]: [site  ] DEBUG 2024/03/19 15:35:00 battery power: 0W
Mar 19 15:35:00 evcc evcc[9870]: [site  ] DEBUG 2024/03/19 15:35:00 grid meter: -3955W
Mar 19 15:35:00 evcc evcc[9870]: [site  ] DEBUG 2024/03/19 15:35:00 grid powers: [-1363 -1314 -1278]W
Mar 19 15:35:00 evcc evcc[9870]: [site  ] DEBUG 2024/03/19 15:35:00 grid currents: [-5.78 -5.57 -5.51]A
Mar 19 15:35:00 evcc evcc[9870]: [site  ] DEBUG 2024/03/19 15:35:00 site power: -3955W
Mar 19 15:35:00 evcc evcc[9870]: [lp-1  ] DEBUG 2024/03/19 15:35:00 charge total import: 534.310kWh
Mar 19 15:35:00 evcc evcc[9870]: [lp-1  ] DEBUG 2024/03/19 15:35:00 charger status: C
Mar 19 15:35:00 evcc evcc[9870]: [lp-1  ] DEBUG 2024/03/19 15:35:00 charge start energy: 534kWh
Mar 19 15:35:00 evcc evcc[9870]: [lp-1  ] INFO 2024/03/19 15:35:00 start charging ->
Mar 19 15:35:00 evcc evcc[9870]: [lp-1  ] DEBUG 2024/03/19 15:35:00 wake-up timer: stop
Mar 19 15:35:00 evcc evcc[9870]: [lp-1  ] DEBUG 2024/03/19 15:35:00 pv charge current: 17.3A = 8.7A + 8.6A (-3955W @ 2p)
Mar 19 15:35:00 evcc evcc[9870]: [lp-1  ] DEBUG 2024/03/19 15:35:00 max charge current: 16A
Mar 19 15:35:10 evcc evcc[9870]: [site  ] DEBUG 2024/03/19 15:35:10 ----
Mar 19 15:35:10 evcc evcc[9870]: [lp-1  ] DEBUG 2024/03/19 15:35:10 charge power: 4W
Mar 19 15:35:10 evcc evcc[9870]: [site  ] DEBUG 2024/03/19 15:35:10 pv power: 4148W
Mar 19 15:35:10 evcc evcc[9870]: [site  ] DEBUG 2024/03/19 15:35:10 battery soc: 100%
Mar 19 15:35:10 evcc evcc[9870]: [site  ] DEBUG 2024/03/19 15:35:10 battery power: 0W
Mar 19 15:35:10 evcc evcc[9870]: [site  ] DEBUG 2024/03/19 15:35:10 grid meter: -3950W
Mar 19 15:35:10 evcc evcc[9870]: [site  ] DEBUG 2024/03/19 15:35:10 grid powers: [-1363 -1312 -1274]W
Mar 19 15:35:10 evcc evcc[9870]: [site  ] DEBUG 2024/03/19 15:35:10 grid currents: [-5.78 -5.56 -5.49]A
Mar 19 15:35:10 evcc evcc[9870]: [site  ] DEBUG 2024/03/19 15:35:10 site power: -3950W
Mar 19 15:35:10 evcc evcc[9870]: [lp-1  ] DEBUG 2024/03/19 15:35:10 charge total import: 534.310kWh
Mar 19 15:35:10 evcc evcc[9870]: [lp-1  ] DEBUG 2024/03/19 15:35:10 charger status: C
Mar 19 15:35:10 evcc evcc[9870]: [lp-1  ] DEBUG 2024/03/19 15:35:10 pv charge current: 24.6A = 16A + 8.59A (-3950W @ 2p)
Mar 19 15:35:20 evcc evcc[9870]: [site  ] DEBUG 2024/03/19 15:35:20 ----
Mar 19 15:35:20 evcc evcc[9870]: [lp-1  ] DEBUG 2024/03/19 15:35:20 charge power: 4W
Mar 19 15:35:20 evcc evcc[9870]: [site  ] DEBUG 2024/03/19 15:35:20 pv power: 4123W
Mar 19 15:35:20 evcc evcc[9870]: [site  ] DEBUG 2024/03/19 15:35:20 battery soc: 100%
Mar 19 15:35:20 evcc evcc[9870]: [site  ] DEBUG 2024/03/19 15:35:20 battery power: 0W
Mar 19 15:35:20 evcc evcc[9870]: [site  ] DEBUG 2024/03/19 15:35:20 grid meter: -3910W
Mar 19 15:35:20 evcc evcc[9870]: [site  ] DEBUG 2024/03/19 15:35:20 grid powers: [-1349 -1302 -1260]W
Mar 19 15:35:20 evcc evcc[9870]: [site  ] DEBUG 2024/03/19 15:35:20 grid currents: [-5.72 -5.52 -5.44]A
Mar 19 15:35:20 evcc evcc[9870]: [site  ] DEBUG 2024/03/19 15:35:20 site power: -3910W
Mar 19 15:35:20 evcc evcc[9870]: [lp-1  ] DEBUG 2024/03/19 15:35:20 charge total import: 534.310kWh
Mar 19 15:35:20 evcc evcc[9870]: [lp-1  ] DEBUG 2024/03/19 15:35:20 charger status: C
Mar 19 15:35:20 evcc evcc[9870]: [lp-1  ] DEBUG 2024/03/19 15:35:20 pv charge current: 24.5A = 16A + 8.5A (-3910W @ 2p)
[... cut ...]
Mar 19 15:45:50 evcc evcc[9870]: [site  ] DEBUG 2024/03/19 15:45:50 grid currents: [-8.57 -8.55 -8.29]A
Mar 19 15:45:50 evcc evcc[9870]: [site  ] DEBUG 2024/03/19 15:45:50 site power: -6029W
Mar 19 15:45:50 evcc evcc[9870]: [lp-1  ] DEBUG 2024/03/19 15:45:50 charge total import: 534.310kWh
Mar 19 15:45:50 evcc evcc[9870]: [lp-1  ] DEBUG 2024/03/19 15:45:50 charger status: B
Mar 19 15:45:50 evcc evcc[9870]: [lp-1  ] DEBUG 2024/03/19 15:45:50 pv charge current: 13.1A = 0A + 13.1A (-6029W @ 2p)
Mar 19 15:45:50 evcc evcc[9870]: [lp-1  ] DEBUG 2024/03/19 15:45:50 max charge current: 13.1A
```

andig, 2024-03-22:
Das

    charge start energy: 518kWh

und das

    final charge energy: 536kWh

sieht aus, als würde hier die cr.startEnergy fehlen:

	// get end energy amount
	if m, ok := cr.meter.(api.MeterEnergy); ok {
		if f, err := m.TotalEnergy(); err == nil {
			cr.chargedEnergy += f - cr.startEnergy
			cr.log.DEBUG.Printf("final charge energy: %.3gkWh", cr.chargedEnergy)
		} else {
			cr.log.ERROR.Printf("charge meter error %v", err)
		}
	}

Irgendwo wird diese also blöderweise genullt:

    charge start energy: 0kWh

Und tatsächlich scheint dein Ladezähler vorher nix zu liefern:

    charge total import: 0.000kWh

Jetzt müsste man man tracen wo das her kommt:

	f, err := m.TotalEnergy()
	if err != nil {
		lp.log.ERROR.Printf("charge total import: %v", err)
		return 0
	}

	lp.log.DEBUG.Printf("charge total import: %.3fkWh", f)

Problem des Ladezählers?

julred, 2024-03-22:
@andig 
Mag sein dass der Ladezähler ein Problem darstellt. Jedoch ist in der Ladehistorie einsehbar dass Start und Ende mit jeweils einem Zählerstand hinterlegt ist. Das ist doch sicherlich die gleiche Datenabfrage? Ggf timing Probleme? Oder nochmal eine Abfrage einfließen lassen ob der Weer plausibel ist?

andig, 2024-03-22:
Da hilft nur Ablauf mit dem Debugger nachzuverfolgen.

julred, 2024-04-02:
Gibts irgendwelche neue Erkenntnisse?
Es gibt ja den Startzählerwert und den Endzählerwert in der Ladehistorie.
Können diese Werte nicht als Berechnung des Ladedeltas genutzt werden?
Stattdessen hauen die exorbitant hohen Ladezählerwerte meine Historie um die Ohren.

andig, 2024-04-02:
> Da hilft nur Ablauf mit dem Debugger nachzuverfolgen.

Das muss jemand machen der das Problem hat...

julred, 2024-04-02:
Kann man eine Option einbauen, die das Ladedelta anderweitig berechnet?

premultiply, 2024-04-02:
Codeseitig ist das in der Implementierung so übersichtlich, dass es nahezu ausgeschlossen ist, dass dies ein Problem von evcc sein könnte.

https://github.com/evcc-io/evcc/blob/master/meter/cfos.go

Es wird genau ein Register für TotalEnergy von der Box eingelesen.
Im weiteren Ablauf wird daraus der Startzählerstand, die momentan bereits geladene Energie sowie mit dem Endzählerstand die gesamte Lademenge jeweils über die Differenz eindeutig berechnet.

Das ist so einfach und unzweifelhaft, dass hier ein Problem eigentlich(TM) nur von der Quelle herrühren kann.

premultiply, 2024-04-03:
Du könntest den Zähler einfach auf Loglevel trace setzen, dann sieht man die genaue Modbus-Kommunikation in der Logausgabe.

SolarPowerEV, 2024-04-03:
> Mit `journalctl` und der Start- und Endzeit, siehe https://docs.evcc.io/docs/guides/setup/#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen

Der Link funktioniert leider nicht.

StefanSchoof, 2024-04-04:
https://docs.evcc.io/docs/faq/#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen

andig, 2024-04-04:
> Und tatsächlich scheint dein Ladezähler vorher nix zu liefern:

Wir halten fest: es fehlt ein trace log um zu validieren, ob die hier verarbeiteten Daten plausibel sind.

kjs-DE, 2024-04-06:
Bei mir tritt ein ähnliches Problem auf aber nur bei "Stotterladung". Da ich nicht viel mit dem Terminal am Hut habe muß ich erst mal versuchen ein trace log zu erstellen wenn es wieder mal wechselhaft sonnig ist.

julred, 2024-04-06:
> Bei mir tritt ein ähnliches Problem auf aber nur bei "Stotterladung". Da ich nicht viel mit dem Terminal am Hut habe muß ich erst mal versuchen ein trace log zu erstellen wenn es wieder mal wechselhaft sonnig ist. Das wird aber dann garantiert ewig lang.

Bei welcher Wallbox/ welcher Ladezähler?

kjs-DE, 2024-04-06:
Wallbox EM2GO EM011AS0ONC. Sonnen ECO8 für alle übrigen Daten (PV/Netz/Akku)

julred, 2024-04-21:
> Du könntest den Zähler einfach auf Loglevel trace setzen, dann sieht man die genaue Modbus-Kommunikation in der Logausgabe.

Mein Ladezähler heißt in der yaml charger_brain_power
Ist das nun korrekt?
```

log: debug
levels:
  cache: error
  charge_brain_solar: trace
  lp-1: trace
  sma: error
```

julred, 2024-04-28:
@andig 
Habe gefühlt jede zweite Ladung das Problem. 
Habe im anderen issue logs hinzugefügt.
Leider hast du das hier geschlossen obwohl das problem nach wie vor besteht

MatzeBu, 2024-08-15:
Ich habe das Problem auch immer wieder mal mit meiner CFOS. Es zieht sich aber soweit durch, das diese zu den Zeitpunkten von EVCC falsch zählt, neugestartet ist. Also wahrscheinlich abgestürzt.
Dann erscheint in EVCC nicht mehr die Differenz sondern der ganze Zählerstand.
Leider problematisch, da dann das Auto vermeintlich voll ist und deshalb nicht weiter geladen wird. Tatsächlich aber noch zu wenig drin ist...