evcc-io/evcc #12987: OpenWB 2.0 Modbus TCP: Aktive Phasen werden auf auf Current-Topics geschrieben

IAmKonni, 2024-03-16:
### Describe the bug

Hallo,
ich habe bereits seit mehreren Wochen einen kleinen Fehler beobachtet, der erstmal keine große Auswirkungen hat. Ich wollte ihn trotzdem melden. Möglicherweise liegt der Fehler auch auf Seiten von OpenWB und wie dort intern die Verarbeitung von MQTT-Daten gesteuert wird. Ich habe in beiden Repositories mir den Code kurz angesehen, auf den ersten Blick aber weder bei evcc noch bei OpenWB einen Fehler entdecken können. Ich bin aber weder in Go noch in Python besonders erfahren, so dass ich das natürlich nur oberflächlich prüfen konnte.

Nun zur Beschreibung:
evcc (stets die aktuellste Release-Version) steuert meine OpenWB Series 2 Standard+ (Version 2.1.3, aber auch stets aktuell auf master) sowie eine OpenWB pro. Dies funktioniert prinzipiell auch fehlerfrei.

In den Logs kann ich allerdings über den Tag hinweg in Abhängigkeit von der PV-Leistung folgende Logs sehen:

```
2024-03-16 08:23:57,828 - {helpermodules.setdata:338} - {ERROR:Setdata} - Payload ungültig: Topic openWB/set/chargepoint/0/set/current, Payload 3 liegt in keinem der angegebenen Wertebereiche.
2024-03-16 08:25:17,937 - {helpermodules.setdata:338} - {ERROR:Setdata} - Payload ungültig: Topic openWB/set/chargepoint/0/set/current, Payload 1 liegt in keinem der angegebenen Wertebereiche.
2024-03-16 10:24:57,003 - {helpermodules.setdata:338} - {ERROR:Setdata} - Payload ungültig: Topic openWB/set/chargepoint/0/set/current, Payload 3 liegt in keinem der angegebenen Wertebereiche.
2024-03-16 10:26:58,111 - {helpermodules.setdata:338} - {ERROR:Setdata} - Payload ungültig: Topic openWB/set/chargepoint/0/set/current, Payload 1 liegt in keinem der angegebenen Wertebereiche.
```
Anhand der Payloads scheint es darauf hinzudeuten, dass an irgendeiner Stelle des Datenflusses das Mapping in der Verarbeitung falsch ist und die Phasenzahl auf das current-Topic geschrieben wird. Da die Werte unterhalb von theoretischen 6 Ampere liegen, wird dann vermutlich dieser Fehler geloggt.

Anhand der in Influx historisierten Daten lässt sich das auch ganz gut nachverfolgen. Anbei der zu diesen Logs passende Ausschnitt aus InfluxDB:

[image]

Es wäre schön, wenn das jemand verifizieren könnte. Die Modbus TCP Register im Code habe ich schon gecheckt, die scheinen zu passen. Wenn auf evcc-Seite kein Fehler zu finden ist, würde ich den Issue zu OpenWB schieben.

Danke und viele Grüße
Konni

### Steps to reproduce

1. OpenWB 2.0 über Modbus TCP konfigurieren
2. Wechselnde PV-Last simulieren, so dass die Phasenumschaltung aktiviert wird
3. main.log in der OpenWB checken

### Configuration details

```yaml
site:
- title: Zuhause # display name for UI
  meters:
    grid: z1
    pv: se17k_mqtt
    battery: smileb3
  residualPower: 100

telemetry: true
plant: xxx

meters:
  - name: z1
    type: custom
    power: # Leistung (W)
      source: mqtt
      topic: electricity/z1/status
      jq: .power_w
    energy:
      source: mqtt
      topic: electricity/z1/status
      jq: .energy_wh
      scale: 0.001
    currents: # Stromstärke (A) pro Phase
      - source: mqtt # Phase 1 Plugin Typ
        topic: electricity/z1/status
        jq: .current_l1_a
      - source: mqtt # Phase 2 Plugin Typ
        topic: electricity/z1/status
        jq: .current_l2_a
      - source: mqtt # Phase 3 Plugin Typ
        topic: electricity/z1/status
        jq: .current_l3_a
  - name: se17k_mqtt
    type: custom
    power: # Leistung (W)
      source: mqtt
      topic: pv/se17k/ac/power
    energy:
      source: mqtt
      topic: pv/se17k/energy_wh
      scale: 0.001
    currents: # Stromstärke (A) pro Phase
      - source: mqtt # Phase 1 Plugin Typ
        topic: pv/se17k/ac/current/l1
      - source: mqtt # Phase 2 Plugin Typ
        topic: pv/se17k/ac/current/l2
      - source: mqtt # Phase 3 Plugin Typ
        topic: pv/se17k/ac/current/l3
  - name: smileb3
    type: custom
    capacity: 8.25
    power:
      source: mqtt
      topic: alphaess/ac/power
    soc:
      source: mqtt
      topic: alphaess/soc

chargers:
  - name: openwb
    type: template
    template: openwb-2.0
    modbus: tcpip
    id: 1
    host: 10.1.1.10
    port: 5030
  - name: openwb-pro
    type: template
    template: openwb-pro 
    host: 10.1.1.244

loadpoints:
  - title: Garage # display name for UI
    charger: openwb # Wallbox Referenz
    mode: pv # charge mode (off, now, minpv, pv)
    vehicle: cupra_born
  - title: Carport
    charger: openwb-pro
    mode: pv
    vehicle: skoda_enyaq

vehicles:
  - name: cupra_born 
    type: template
    template: cupra
    title: # Wird in der Benutzeroberfläche angezeigt # Optional
    user: xxx # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.)
    password: xxx # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen)
    vin: xxx # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind # Optional
    capacity: 58 # Akku-Kapazität in kWh # Optional
  - name: skoda_enyaq
    type: template
    template: enyaq
    title: # Wird in der Benutzeroberfläche angezeigt # Optional
    user: xxx # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.)
    password: xxx # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen)
    vin: xxx # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind # Optional
    capacity: 77 # Akku-Kapazität in kWh # Optional
    identifiers:
      - 48:31:xxx

tariffs:
  grid:
    type: tibber
    token: "-xxx-cBA0DQzMc"
    homeid: "xxx-abb8-4321-989a-xxx"
  feedin:
    type: fixed
    price: 0.09

mqtt:
  broker: 10.1.1.10:1883
  topic: evcc # root topic for publishing, set empty to disable publishing
  user: xxx
  password: xxx

influx:
  url: http://10.1.1.10:8086
  database: evcc # InfluxDB v2.x uses term `bucket` but for compatibility still named `database` here
  token: xxx==
  org: myorg

sponsortoken: xxx

# log settings
log: info
levels:
  influx: info
```

### Log details

```text
Logs in evcc sind diesbezüglich unauffällig, habe aber auch noch nicht im Debug-Modus gecheckt (versuche ich noch nachzureichen)
```

### What type of operating system are you running?

Docker container

### Version

0.124.10

andig, 2024-03-16:
> main.log in der OpenWB checken

Ich verstehe nur Bahnhof. Was hat das Log der openWB mit evcc zu tun?

andig, 2024-03-16:
> Wenn auf evcc-Seite kein Fehler zu finden ist, würde ich den Issue zu OpenWB schieben.

Ich sehe keinen...

closed 2024-03-16 by andig