evcc-io/evcc #12994: Batteriesteuerung funktioniert nicht bei Solaredge

Cytron1980, 2024-03-16:
### Describe the bug

Batterie wird trotz Anzeige "gesperrt" beim Modus "Schnell" entladen.
Aufgefallen heute. Ha die letzten Tage aber funktioniert. Abwahl der Funktion, Neustart von EVCC und Modus-Umstellung hat nicht geholfen.

### Steps to reproduce

1. Auto anschließen
2.Modus "Schnell" aktivieren

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 40e7a34a253eb356e07322451451f552e79457aa1fd68cf282472088ac0f1c4b

interval: 10s # control cycle interval

sponsortoken: XXXXXXX

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: solaredge-hybrid 
  id: 1  
  host: 192.168.2.201  
  port: 1502  
  usage: grid  
  modbus: tcpip  
  name: grid1
- type: template
  template: solaredge-hybrid 
  id: 1  
  host: 192.168.2.201  
  port: 1502  
  usage: pv  
  modbus: tcpip  
  name: pv2
- type: template
  template: solaredge-hybrid 
  id: 1  
  host: 192.168.2.201  
  port: 1502  
  usage: battery  
  modbus: tcpip  
  name: battery3

chargers:
- type: template
  template: easee 
  user: XXXteme.com  
  password: XXX  
  charger: XXX 
  timeout: 10s  
  name: wallbox
  

[... 246 lines cut ...]
[site  ] DEBUG 2024/03/16 17:08:13 battery soc: 29%
[site  ] DEBUG 2024/03/16 17:08:13 battery power: 2950W
[site  ] DEBUG 2024/03/16 17:08:13 grid meter: -3232W
[site  ] DEBUG 2024/03/16 17:08:13 site power: -282W
[lp-1  ] DEBUG 2024/03/16 17:08:13 charge currents: [0.108 0.105 0.109]A
[lp-1  ] DEBUG 2024/03/16 17:08:13 charge total import: 2419.467kWh
[lp-1  ] DEBUG 2024/03/16 17:08:13 charger status: B
[lp-1  ] INFO 2024/03/16 17:08:13 stop charging <-
[lp-1  ] DEBUG 2024/03/16 17:08:13 charge total import: 2419.467kWh
[easee ] TRACE 2024/03/16 17:08:13 ProductUpdate EHNRT639: (2024-03-16 16:08:13 +0000 UTC) OUT_VOLT_PIN1_4 0.13500000536441803
[easee ] TRACE 2024/03/16 17:08:13 ProductUpdate EHNRT639: (2024-03-16 16:08:13 +0000 UTC) OUT_VOLT_PIN1_5 1.9240000247955322
[easee ] TRACE 2024/03/16 17:08:13 ProductUpdate EHNRT639: (2024-03-16 16:08:13 +0000 UTC) OUT_VOLT_PIN1_3 0.9940000176429749
[lp-1  ] DEBUG 2024/03/16 17:08:14 soc estimated: 46.13% (vehicle: 46.00%)
[lp-1  ] DEBUG 2024/03/16 17:08:14 vehicle soc: 46%
[lp-1  ] DEBUG 2024/03/16 17:08:14 vehicle soc limit: 80%
[lp-1  ] DEBUG 2024/03/16 17:08:14 vehicle range: 174km
```

### What type of operating system are you running?

Windows

### Version

0.124.10

andig, 2024-03-16:
```
[site  ] DEBUG 2024/03/16 17:07:54 set battery mode: hold
[site  ] DEBUG 2024/03/16 17:08:03 battery power: 5001W
```

@premultiply hast Du eine Idee?

mucki12, 2024-03-16:
Gibt es da vielleicht einen Zusammenhang?
https://github.com/evcc-io/evcc/discussions/12990

Gotcha1980, 2024-03-16:
Es gibt eine neue Funktion bei solaredge die nennt sich Time of Use wenn man die zum testen einstellt und danach wieder auf Maximum Self consumption wechselt steht der Control Mode nicht mehr auf Remote
[image]

Cytron1980, 2024-03-16:
Das kann es tatsächlich sein. Habe die Tage den Modus wieder auf "maximalen eigenverbrauch" gestellt. Werde morgen nochmal zurück stellen und testen.

Cytron1980, 2024-03-16:
Gerade getestet. Im Modus "Nutzungszeit" klappt alles wie gewollt. Scheint also daher zu kommen.

Gotcha1980, 2024-03-16:
Genau weil beim Time of Use mode wird der Wechselrichter auf Remote gestellt. In der Setapp heißt der Modus Time of Use + Remote. 

Mir ist das auch nur aufgefallen weil ich ebenfalls einen solaredge Wechselrichter habe und genau das gleiche Problem. 😬

Cytron1980, 2024-03-16:
Also einfach nur die Zeiten raus nehmen und in dem Modus belassen?

Gotcha1980, 2024-03-16:
Entweder so, wobei ich glaube ohne Zeiten geht das nicht, oder über die Modbus Multi Integration den Control Mode händisch auf Remote stellen. Oder mit der setapp direkt am Wechselrichter das einstellen. 
Hier mal ein Bild aus Homeassistant von meinem Wechselrichter.
[image]

Cytron1980, 2024-03-16:
Habe nur den ioBroker. Ich glaube dort ist das der Mode 4 beim Storage Control Mode... ich werde das mal testen. Aber eigentlich sollte EVCC unabhängig vom Batterie Management doch den Speicher sperren können (Mode auf 4 schreiben) so habe ich das vorher ja auch gelöst obwohl der Modus auf maximaler Eigenverbrauch stand, hat er den Mode 4 geschrieben und auch beibehalte.

Cytron1980, 2024-03-16:
Habe jetzt den Modus erstmal wieder aktiviert und die nutzungzeit nur auf Dezember und Januar begrenzt...somit sollte es erstmal funktionieren.

andig, 2024-03-17:
Hier scheint im Template ja ein Hinweis zu fehlen, welcher Modus verwendet werden darf oder auch nicht. Was wäre das konkret? Ein PR wäre Klasse!

Cytron1980, 2024-03-17:
Hilf mir auf die Sprünge- PR?! Kann ich da weiterhelfen? Was bedeutet dies in dem Zusammenhang hier? VG

andig, 2024-03-17:
Pull Request. Zur Ergänzung des SE Templates um den entsprechenden Hinweis.

Cytron1980, 2024-03-17:
Okay. Werde ich machen.

StefanSchoof, 2024-03-17:
Am einfachsten im Browser https://github.com/evcc-io/evcc/blob/master/templates/definition/meter/solaredge-hybrid.yaml öffnen und dann edit in place klicken. Dann Text anpassen und dann commit und pull request öffnen. Gibt unter https://docs.github.com/de/repositories/working-with-files/managing-files/editing-files eine Anleitung

Cytron1980, 2024-03-18:
Habe ich jetzt mal gemacht.hoffe das war so richtig.

andig, 2024-03-19:
Ist drin- vielen Dank!

closed 2024-03-19 by andig