evcc-io/evcc #12999: Vehicle reconnection logic gives up

Gifford47, 2024-03-17:
### Describe the bug

If the car is offline or the server cannot be reached, the logic attempts to re-establish the connection. However, if this does not work for a certain period of time, no further connection attempt is made and the car remains "offline". The behavior is apparently logic and not brand-related. See also [here](https://github.com/evcc-io/evcc/discussions/10706).

### Steps to reproduce

1. Connect a car and start evcc. Car is "online"
2. try to block the connection from the car to the server. Time period unknown. Try for at least 24h.
3. unblock the connection and reconnect the car to the server
4. EVCC should stay "offline" although it should be connected

### Configuration details

```yaml
not provided
```

### Log details

```text
not logged yet, because it´s a long term issue
```

### What type of operating system are you running?

Linux

### Version

0.124.7

andig, 2024-03-17:
Missing log

closed 2024-03-17 by andig

andig, 2024-03-17:
Let's see if we can incorporate at least a manual retry in https://github.com/evcc-io/evcc/pull/12962

Gifford47, 2024-03-20:
i could not get the reason until now, but i logged with debug level and added it to the log details above... the trace log floods my log files after 18 hours... do i need anyway?