evcc-io/evcc #13000: "Drain Battery" (or "Max Self Consumption") Charging Mode

MarkusMit, 2024-03-17:
**Is your feature request related to a problem? Please describe.**
When I don't need a fully charged car for my travel, before leaving home on a sunny day's morning, I'd just like to transfer all available energy from my home energy storage into my car so that the home energy storage can be (fully) recharged during the rest of the day.

**Describe the solution you'd like**
As a variation of the 'Fast' charging mode, evcc should provide a "Drain Battery" (or "Max Self Consumption") mode, where the max available power of PV and Battery is used for charging, with no (or just minimal) power from Grid.

With active battery management by evcc, a separate battery discarge limit might be specified for this "Drain Battery" mode.

As a bonus feature I'd like to suggest an "auto-activation" mode, where I can activate the "Drain Battery" mode already during the night, but it only starts charging when PV starts producing a certain amount of power. This limit might either be configurable (so that the average household load is covered), or just the one used for the "PV" Charging mode.

**Describe alternatives you've considered**
I tried to achive the same result by manually limitting the max charging current, but with varying load of the household or strong variations of PV production on a partially cloudy day, it required permanent monitioring to maximize battery discharge while minimizing Grid use.

**Additional context**
-

VolkerK62, 2024-03-17:
seems to be the same  https://github.com/evcc-io/evcc/discussions/12981

MarkusMit, 2024-03-17:
Indeed. Didn't think of checking the discussions.

closed 2024-03-17 by andig