evcc-io/evcc #13001: Back-Off when Charge Point is not available

MarkusMit, 2024-03-17:
**Is your feature request related to a problem? Please describe.**
I have a "mobile" charger (_go-eCharger Gemini flex_) and it happens quite often that it gets disconnected. 
When that happens, evcc starts flooding its log file with error messages like this one, because it tries getting the status every few seconds:
```
[lp-1  ] ERROR 2024/03/16 15:11:34 charge meter: Get "http://192.***.***.***/api/status?filter=alw,car,eto,nrg,wh,trx,cards": dial tcp 192.***.***.***:80: connect: no route to host
```

**Describe the solution you'd like**
evcc should back-off when failing to connect to the charger to at least a minute or so (instead of a few seconds). additionally it might reduce log-messages even more when the charger is not availble for a longer time.
However, it shouldn't take longer than a minute or so, that evcc discovers when the charge point becomes available again.

**Describe alternatives you've considered**
Since I use evcc only for charging my car, I could manually deactivate the Home Assistant evcc add-on in my setup, but when evcc also using for other purposes (e.g. controlling a water heater) this isn't feasible.

i

andig, 2024-03-17:
> because it tries getting the status every few seconds:

Why is that? Shoudn't happen more often than once per interval imho.

MarkusMit, 2024-03-17:
I can't answer the why, I can just say that it happens.
My interval is configured to 13 seconds.

```
interval: 13s # control cycle interval
```

But a log entry occurs about every three seconds. Taking a closer look, it seems there are happening different requests ("charge total import", "charger", "charge meter", "charge rater") although the url is always the same, but still "charge meter" happens twice within 3 seconds, then again 12 seconds later:
```
[lp-1  ] ERROR 2024/03/16 15:11:28 charge total import: Get "http://192.***.***.***/api/status?filter=alw,car,eto,nrg,wh,trx,cards": dial tcp 192.***.***.***:80: connect: no route to host
[lp-1  ] ERROR 2024/03/16 15:11:31 charger: Get "http://192.***.***.***/api/status?filter=alw,car,eto,nrg,wh,trx,cards": dial tcp 192.***.***.***:80: connect: no route to host
[lp-1  ] ERROR 2024/03/16 15:11:34 charge meter: Get "http://192.***.***.***/api/status?filter=alw,car,eto,nrg,wh,trx,cards": dial tcp 192.***.***.***:80: connect: no route to host
[lp-1  ] ERROR 2024/03/16 15:11:37 charge meter: Get "http://192.***.***.***/api/status?filter=alw,car,eto,nrg,wh,trx,cards": dial tcp 192.***.***.***:80: connect: no route to host
[lp-1  ] ERROR 2024/03/16 15:11:40 charge rater: Get "http://192.***.***.***/api/status?filter=alw,car,eto,nrg,wh,trx,cards": dial tcp 192.***.***.***:80: connect: no route to host
[lp-1  ] ERROR 2024/03/16 15:11:43 charge total import: Get "http://192.***.***.***/api/status?filter=alw,car,eto,nrg,wh,trx,cards": dial tcp 192.***.***.***:80: connect: no route to host
[lp-1  ] ERROR 2024/03/16 15:11:46 charger: Get "http://192.***.***.***/api/status?filter=alw,car,eto,nrg,wh,trx,cards": dial tcp 192.***.***.***:80: connect: no route to host
[lp-1  ] ERROR 2024/03/16 15:11:49 charge meter: Get "http://192.***.***.***/api/status?filter=alw,car,eto,nrg,wh,trx,cards": dial tcp 192.***.***.***:80: connect: no route to host
[lp-1  ] ERROR 2024/03/16 15:11:52 charge meter: Get "http://192.***.***.***/api/status?filter=alw,car,eto,nrg,wh,trx,cards": dial tcp 192.***.***.***:80: connect: no route to host
[lp-1  ] ERROR 2024/03/16 15:11:56 charge rater: Get "http://192.***.***.***/api/status?filter=alw,car,eto,nrg,wh,trx,cards": dial tcp 192.***.***.***:80: connect: no route to host
[lp-1  ] ERROR 2024/03/16 15:11:59 charge total import: Get "http://192.***.***.***/api/status?filter=alw,car,eto,nrg,wh,trx,cards": dial tcp 192.***.***.***:80: connect: no route to host
[lp-1  ] ERROR 2024/03/16 15:12:02 charger: Get "http://192.***.***.***/api/status?filter=alw,car,eto,nrg,wh,trx,cards": dial tcp 192.***.***.***:80: connect: no route to host
[lp-1  ] ERROR 2024/03/16 15:12:05 charge meter: Get "http://192.***.***.***/api/status?filter=alw,car,eto,nrg,wh,trx,cards": dial tcp 192.***.***.***:80: connect: no route to host
[lp-1  ] ERROR 2024/03/16 15:12:08 charge meter: Get "http://192.***.***.***/api/status?filter=alw,car,eto,nrg,wh,trx,cards": dial tcp 192.***.***.***:80: connect: no route to host
[lp-1  ] ERROR 2024/03/16 15:12:11 charge rater: Get "http://192.***.***.***/api/status?filter=alw,car,eto,nrg,wh,trx,cards": dial tcp 192.***.***.***:80: connect: no route to host
[lp-1  ] ERROR 2024/03/16 15:12:14 charge total import: Get "http://192.***.***.***/api/status?filter=alw,car,eto,nrg,wh,trx,cards": dial tcp 192.***.***.***:80: connect: no route to host
[lp-1  ] ERROR 2024/03/16 15:12:17 charger: Get "http://192.***.***.***/api/status?filter=alw,car,eto,nrg,wh,trx,cards": dial tcp 192.***.***.***:80: connect: no route to host
[lp-1  ] ERROR 2024/03/16 15:12:20 charge meter: Get "http://192.***.***.***/api/status?filter=alw,car,eto,nrg,wh,trx,cards": dial tcp 192.***.***.***:80: connect: no route to host
[lp-1  ] ERROR 2024/03/16 15:12:23 charge meter: Get "http://192.***.***.***/api/status?filter=alw,car,eto,nrg,wh,trx,cards": dial tcp 192.***.***.***:80: connect: no route to host
[lp-1  ] ERROR 2024/03/16 15:12:26 charge rater: Get "http://192.***.***.***/api/status?filter=alw,car,eto,nrg,wh,trx,cards": dial tcp 192.***.***.***:80: connect: no route to host
[lp-1  ] ERROR 2024/03/16 15:12:29 charge total import: Get "http://192.***.***.***/api/status?filter=alw,car,eto,nrg,wh,trx,cards": dial tcp 192.***.***.***:80: connect: no route to host
[lp-1  ] ERROR 2024/03/16 15:12:32 charger: Get "http://192.***.***.***/api/status?filter=alw,car,eto,nrg,wh,trx,cards": dial tcp 192.***.***.***:80: connect: no route to host
[lp-1  ] ERROR 2024/03/16 15:12:36 charge meter: Get "http://192.***.***.***/api/status?filter=alw,car,eto,nrg,wh,trx,cards": dial tcp 192.***.***.***:80: connect: no route to host
[lp-1  ] ERROR 2024/03/16 15:12:39 charge meter: Get "http://192.***.***.***/api/status?filter=alw,car,eto,nrg,wh,trx,cards": dial tcp 192.***.***.***:80: connect: no route to host
[lp-1  ] ERROR 2024/03/16 15:12:42 charge rater: Get "http://192.***.***.***/api/status?filter=alw,car,eto,nrg,wh,trx,cards": dial tcp 192.***.***.***:80: connect: no route to host
[lp-1  ] ERROR 2024/03/16 15:12:45 charge total import: Get "http://192.***.***.***/api/status?filter=alw,car,eto,nrg,wh,trx,cards": dial tcp 192.***.***.***:80: connect: no route to host
[lp-1  ] ERROR 2024/03/16 15:12:48 charger: Get "http://192.***.***.***/api/status?filter=alw,car,eto,nrg,wh,trx,cards": dial tcp 192.***.***.***:80: connect: no route to host
[... cut ...]
   broker: localhost:1883
  # topic: evcc # root topic for publishing, set empty to disable
   user: Mosquito
   password: *****

# influx database
influx:
  # url: http://localhost:8086
  # database: evcc
  # user:
  # password:

# eebus credentials
eebus:
  # uri: # :4712
  # interfaces: # limit eebus to specific network interfaces
  # - en0
  # certificate: # local signed certificate, required, can be generated via `evcc eebus-cert`
  #   public: # public key
  #   private: # private key
```

MarkusMit, 2024-03-21:
Hi, now set the intervall to `30s` but the behaviour stays the same:
```
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/03/21 11:14:19 evcc 0.124.10 
[main  ] INFO 2024/03/21 11:14:19 using config file: /config/evcc.yaml 
[main  ] INFO 2024/03/21 11:14:19 starting ui and api at :7070 
[mqtt  ] INFO 2024/03/21 11:14:20 connecting evcc-649746822 at tcp://localhost:1883 
[mqtt  ] DEBUG 2024/03/21 11:14:20 tcp://localhost:1883 connected
[lp-1  ] ERROR 2024/03/21 11:14:29 charge total import: Get "http://192.***.***.***/status": dial tcp 192.***.***.***:80: connect: no route to host 
[site  ] INFO 2024/03/21 11:14:30 site config:
[site  ] INFO 2024/03/21 11:14:30   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2024/03/21 11:14:30     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2024/03/21 11:14:30     pv 1:      power ✓ energy ✓ currents ✓
[site  ] INFO 2024/03/21 11:14:30     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓
[site  ] INFO 2024/03/21 11:14:30   vehicles:
[site  ] INFO 2024/03/21 11:14:30     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[lp-1  ] ERROR 2024/03/21 11:14:32 charger: Get "http://192.***.***.***/status": dial tcp 192.***.***.***:80: connect: no route to host
[lp-1  ] ERROR 2024/03/21 11:14:35 charge meter: Get "http://192.***.***.***/status": dial tcp 192.***.***.***:80: connect: no route to host
[lp-1  ] ERROR 2024/03/21 11:14:38 charge meter: Get "http://192.***.***.***/status": dial tcp 192.***.***.***:80: connect: no route to host
[lp-1  ] ERROR 2024/03/21 11:14:41 charge meter: Get "http://192.***.***.***/status": dial tcp 192.***.***.***:80: connect: no route to host
[lp-1  ] ERROR 2024/03/21 11:14:44 charge rater: Get "http://192.***.***.***/status": dial tcp 192.***.***.***:80: connect: no route to host
[lp-1  ] ERROR 2024/03/21 11:14:47 charge total import: Get "http://192.***.***.***/status": dial tcp 192.***.***.***:80: connect: no route to host
[lp-1  ] ERROR 2024/03/21 11:14:50 charger: Get "http://192.***.***.***/status": dial tcp 192.***.***.***:80: connect: no route to host
[lp-1  ] ERROR 2024/03/21 11:15:05 charge meter: Get "http://192.***.***.***/status": dial tcp 192.***.***.***:80: connect: no route to host
[lp-1  ] ERROR 2024/03/21 11:15:08 charge meter: Get "http://192.***.***.***/status": dial tcp 192.***.***.***:80: connect: no route to host
[lp-1  ] ERROR 2024/03/21 11:15:11 charge meter: Get "http://192.***.***.***/status": dial tcp 192.***.***.***:80: connect: no route to host
[lp-1  ] ERROR 2024/03/21 11:15:14 charge rater: Get "http://192.***.***.***/status": dial tcp 192.***.***.***:80: connect: no route to host
[lp-1  ] ERROR 2024/03/21 11:15:17 charge total import: Get "http://192.***.***.***/status": dial tcp 192.***.***.***:80: connect: no route to host
[lp-1  ] ERROR 2024/03/21 11:15:20 charger: Get "http://192.***.***.***/status": dial tcp 192.***.***.***:80: connect: no route to host
[lp-1  ] ERROR 2024/03/21 11:15:35 charge meter: Get "http://192.***.***.***/status": dial tcp 192.***.***.***:80: connect: no route to host
[lp-1  ] ERROR 2024/03/21 11:15:38 charge meter: Get "http://192.***.***.***/status": dial tcp 192.***.***.***:80: connect: no route to host
[lp-1  ] ERROR 2024/03/21 11:15:41 charge meter: Get "http://192.***.***.***/status": dial tcp 192.***.***.***:80: connect: no route to host
[lp-1  ] ERROR 2024/03/21 11:15:44 charge rater: Get "http://192.***.***.***/status": dial tcp 192.***.***.***:80: connect: no route to host
[lp-1  ] ERROR 2024/03/21 11:15:47 charge total import: Get "http://192.***.***.***/status": dial tcp 192.***.***.***:80: connect: no route to host
```

andig, 2024-03-22:
Spannend. Müsste man mal mit dem Debugger rein schauen, warum das passiert.

closed 2024-03-28 by andig

andig, 2024-07-31:
Tatsächlich sieht das wie retry aufgrund technischen Fehlers aus. Die Fehlerursache hier ist wohl Netzwerk