evcc-io/evcc #13004: Der im UI angezeigte Zähler der Pausieren-Meldung wegen mangelndem Überschuss ist nicht konsistent (ein Refresh resettet den Zähler)

SolarPowerEV, 2024-03-17:
### Describe the bug

Wenn die Sonne nicht mehr ausreicht, um Überschuss-Laden aufrecht zu erhalten (z.B. weil eine dicke Wolke erschienen oder ein fetter Verbraucher angesprungen ist), dann erscheint ein Timer, der von 3 Minuten an rückwärts zählt:

[image]

Soweit so gut und auch erwartetes Verhalten. Aber wenn ich dann einen Browser-Refresh mache oder z.B. evcc parallel dazu auf meinem Handy aufrufe, dann läuft der angezeigte Timer im entsprechenden Browser wieder bei 3:00 los.

[image]

Das ist für mich unerwartetes Verhalten, weil ich dachte, die 3 Minuten werden generell heruntergezählt, ab dem Moment, an dem der PV-Ertrag eingebrochen ist und nicht abhängig davon, wann ich evcc im Browser aufgerufen habe bzw. wann der letzte Aufruf oder Refresh stattgefunden hat. Und dass jede Browser-Instanz entsprechend denselben Wert anzeigen würde, weil evcc diesen intern(!) herunterzählt.

### Steps to reproduce

1. Einen PV-Überschuss Ladevorgang starten
2. Warten, bis zu wenig Überschuss vorhanden ist (wegen hohem Verbraucher oder Wolke vor der Sonne)
3. Der Zähler erscheint und fängt an von der eingestellten Zeit (disable delay - in meinem Fall 3 Minuten) an rückwärts zu zählen. 
4. Jetzt einfach z.B. nach ein paar Sekunden oder einer Minute in einem zweiten Browser evcc aufrufen und schon hat man zwei verschiedene angezeigte Zeiten. Der erste Browser zählt weiter runter, der später gestartete Browser fängt wieder von neuem (in meinem Fall bei 3 Minuten) an rückwärts zu zählen. 
5. Ein Reload von evcc im aktuellen Browser zeigt ein ähnlich inkonsistentes Anzeigeverhalten. Auch dann springt der Zähler wieder zurück auf (in meinem Fall) 3 Minuten und zählt von da an neu rückwärt herunter.
...

### Configuration details

```yaml
Liefere ich bei Bedarf gerne nach, ist aber Huawei / Alfen Standard.
```

### Log details

```text
evcc-evcc-latest
date	stream	content
2024/03/17 14:01:56	stdout	[lp-1  ] ERROR 2024/03/17 14:01:56 vehicle range: unexpected status: 403 (Forbidden)
2024/03/17 14:01:56	stdout	[lp-1  ] WARN 2024/03/17 14:01:56 vehicle soc: unexpected status: 403 (Forbidden) (ignored by estimator)
2024/03/17 14:01:56	stdout	[lp-1  ] INFO 2024/03/17 14:01:56 start charging ->
2024/03/17 13:52:56	stdout	[lp-1  ] INFO 2024/03/17 13:52:56 stop charging <-
2024/03/17 13:41:41	stdout	[lp-1  ] INFO 2024/03/17 13:41:41 start charging ->
2024/03/17 13:36:30	stdout	[site  ] ERROR 2024/03/17 13:36:30 pv 1 power: read failed: modbus: exception '4' (server device failure), function '3'
2024/03/17 13:32:53	stdout	[site  ] ERROR 2024/03/17 13:32:53 battery 1 power: read failed: read tcp 192.168.2.21:40414->192.168.2.5:502: i/o timeout
2024/03/17 13:32:04	stdout	[site  ] ERROR 2024/03/17 13:32:04 pv 1 power: read failed: read tcp 192.168.2.21:40364->192.168.2.5:502: i/o timeout
2024/03/17 13:30:55	stdout	[site  ] ERROR 2024/03/17 13:30:55 grid meter: read failed: read tcp 192.168.2.21:40340->192.168.2.5:502: i/o timeout
2024/03/17 13:30:39	stdout	[site  ] ERROR 2024/03/17 13:30:39 battery 1 soc: read failed: read tcp 192.168.2.21:40330->192.168.2.5:502: i/o timeout
2024/03/17 13:30:23	stdout	[site  ] ERROR 2024/03/17 13:30:23 battery 1 energy: read failed: read tcp 192.168.2.21:40316->192.168.2.5:502: i/o timeout
2024/03/17 13:30:05	stdout	[site  ] ERROR 2024/03/17 13:30:05 pv 2 power: read failed: read tcp 192.168.2.21:40302->192.168.2.5:502: i/o timeout
2024/03/17 13:29:49	stdout	[site  ] ERROR 2024/03/17 13:29:49 pv 1 power: read failed: read tcp 192.168.2.21:40158->192.168.2.5:502: i/o timeout
2024/03/17 13:28:19	stdout	[lp-1  ] INFO 2024/03/17 13:28:19 stop charging <-
2024/03/17 13:26:28	stdout	[site  ] ERROR 2024/03/17 13:26:28 pv 2 power: read failed: read tcp 192.168.2.21:40146->192.168.2.5:502: i/o timeout
2024/03/17 13:26:12	stdout	[site  ] ERROR 2024/03/17 13:26:12 pv 1 power: read failed: read tcp 192.168.2.21:40132->192.168.2.5:502: i/o timeout
2024/03/17 13:25:55	stdout	[site  ] ERROR 2024/03/17 13:25:55 battery 1 power: read failed: read tcp 192.168.2.21:40118->192.168.2.5:502: i/o timeout
2024/03/17 13:25:39	stdout	[site  ] ERROR 2024/03/17 13:25:39 pv 2 power: read failed: read tcp 192.168.2.21:40106->192.168.2.5:502: i/o timeout
2024/03/17 13:25:23	stdout	[site  ] ERROR 2024/03/17 13:25:23 pv 1 power: read failed: read tcp 192.168.2.21:40090->192.168.2.5:502: i/o timeout
2024/03/17 13:25:07	stdout	[site  ] ERROR 2024/03/17 13:25:07 battery 1 power: read failed: read tcp 192.168.2.21:40076->192.168.2.5:502: i/o timeout
2024/03/17 13:24:51	stdout	[site  ] ERROR 2024/03/17 13:24:51 pv 2 power: read failed: read tcp 192.168.2.21:40048->192.168.2.5:502: i/o timeout
2024/03/17 13:24:35	stdout	[site  ] ERROR 2024/03/17 13:24:35 pv 1 power: read failed: read tcp 192.168.2.21:40016->192.168.2.5:502: i/o timeout
2024/03/17 13:24:02	stdout	[site  ] ERROR 2024/03/17 13:24:02 grid meter: read failed: read tcp 192.168.2.21:39228->192.168.2.5:502: i/o timeout
2024/03/17 13:07:19	stdout	[site  ] ERROR 2024/03/17 13:07:19 pv 1 power: read failed: read tcp 192.168.2.21:32842->192.168.2.5:502: i/o timeout
2024/03/17 12:59:02	stdout	[lp-1  ] ERROR 2024/03/17 12:59:02 vehicle range: unexpected status: 403 (Forbidden)
2024/03/17 12:59:02	stdout	[lp-1  ] WARN 2024/03/17 12:59:02 vehicle soc: unexpected status: 403 (Forbidden) (ignored by estimator)
2024/03/17 11:04:23	stdout	[lp-1  ] INFO 2024/03/17 11:04:23 vehicle updated: unknown -> FIAT 500e
2024/03/17 11:04:23	stdout	[main  ] ERROR 2024/03/17 11:04:23 vehicle status: asleep
2024/03/17 11:02:28	stdout	[main  ] ERROR 2024/03/17 11:02:28 vehicle status: asleep
2024/03/17 11:02:27	stdout	[lp-1  ] INFO 2024/03/17 11:02:27 start charging ->
2024/03/17 11:02:05	stdout	[main  ] ERROR 2024/03/17 11:02:05 vehicle status: asleep
2024/03/17 11:02:04	stdout	[lp-1  ] INFO 2024/03/17 11:02:04 car connected
2024/03/17 10:56:47	stdout	[lp-3  ] INFO 2024/03/17 10:56:47 car disconnected
2024/03/17 10:56:05	stdout	[lp-2  ] INFO 2024/03/17 10:56:05 car disconnected
2024/03/17 10:55:20	stdout	[lp-1  ] INFO 2024/03/17 10:55:20 car disconnected
2024/03/17 10:55:03	stdout	[lp-3  ] INFO 2024/03/17 10:55:03     charge:    power ✓ energy ✓ currents ✓
2024/03/17 10:55:03	stdout	[lp-3  ] INFO 2024/03/17 10:55:03   meters:      charge ✓
2024/03/17 10:55:03	stdout	[lp-3  ] INFO 2024/03/17 10:55:03   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[... 90 lines cut ...]
2024/03/13 16:19:50	stdout	[lp-1  ] INFO 2024/03/13 16:19:50   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
2024/03/13 16:19:50	stdout	[lp-1  ] INFO 2024/03/13 16:19:50   mode:        pv
2024/03/13 16:19:50	stdout	[lp-1  ] INFO 2024/03/13 16:19:50 loadpoint 1:
2024/03/13 16:19:50	stdout	[site  ] INFO 2024/03/13 16:19:50     vehicle 3: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
2024/03/13 16:19:50	stdout	[site  ] INFO 2024/03/13 16:19:50     vehicle 2: range ✓ finish ✗ status ✓ climate ✗ wakeup ✗
2024/03/13 16:19:50	stdout	[site  ] INFO 2024/03/13 16:19:50     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✗
2024/03/13 16:19:50	stdout	[site  ] INFO 2024/03/13 16:19:50   vehicles:
2024/03/13 16:19:50	stdout	[site  ] INFO 2024/03/13 16:19:50     battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✗
2024/03/13 16:19:50	stdout	[site  ] INFO 2024/03/13 16:19:50     pv 2:      power ✓ energy ✓ currents ✗
2024/03/13 16:19:50	stdout	[site  ] INFO 2024/03/13 16:19:50     pv 1:      power ✓ energy ✓ currents ✗
2024/03/13 16:19:50	stdout	[site  ] INFO 2024/03/13 16:19:50     grid:      power ✓ energy ✓ currents ✓
2024/03/13 16:19:50	stdout	[site  ] INFO 2024/03/13 16:19:50   meters:      grid ✓ pv ✓ battery ✓
2024/03/13 16:19:50	stdout	[site  ] INFO 2024/03/13 16:19:50 site config:
2024/03/13 16:19:47	stdout	[main  ] INFO 2024/03/13 16:19:47 starting ui and api at :7070
2024/03/13 16:19:47	stdout	[main  ] INFO 2024/03/13 16:19:47 using config file: /etc/evcc.yaml
2024/03/13 16:19:47	stdout	[main  ] INFO 2024/03/13 16:19:47 evcc 0.124.10
```

### What type of operating system are you running?

Docker container

### Version

0.124.10

andig, 2024-03-17:
@naltatis er müsste egl. beim aktuellen wert des pvTimer los laufen?

SolarPowerEV, 2024-03-18:
Noch ein ähnlicher Fall, der heute aufgetreten und relativ irritierend ist, selbst wenn man evcc nur in einem einzigen Fenster betreibt:
evcc hatte den Ladevorgang laut meiner Tesla-App UND der Anzeige des Loadpoints in evcc zufolge längst beendet, der Zähler im Loadpoint zählte aber immer noch fleissig runter:

[image]

[image]

Die Anzeige des Timers ist also auch nicht in sich konsistent, selbst wenn man nur einen einzigen Browser benutzt.

andig, 2024-03-18:
Das kann gut sein. Bei einem Invervall von 30s gibts eben nur alls 60s einen neuen Status für das UI. Tut aber der Funktionalität keinen Abbruch.

SolarPowerEV, 2024-03-18:
Ok. Mir war nicht bewusst, dass das Update-Intervall für die Abfrage des Wechselrichters durch evcc auch den Update-Zyklus des Web-UI betrifft und sich beide in 30 Sekunden-Intervallen aktualisieren. 
Ich hatte den Eindruck, dass die Verbindung zwischen evcc und dem Browser schneller getaktet sein würde, weil immer wenn ich meine evcc-Instanz im Docker abschieße, das UI binnen 1 Sekunde anzeigt, dass es die Verbindung zum Server verloren hat.

GrimmiMeloni, 2024-03-18:
@SolarPowerEV das ist auch korrekt. Die Events von der UI gehen in Echtzeit an den Core. Deswegen kannst Du auch den Lademodus von AUS auf SCHNELL setzen, und es geht (mehr oder weniger) sofort los, ohne das Intervall abzuwarten.
Der Core aber hat die Taktung des Intervals, und schreibt in dieser Taktung die Restlaufzeit der Timer raus an die UI. Diese erzeugt dann durch einen Countdown die Illusion das da permanent Updates kommen. Rein technisch kommt unter der Haube aber nur immer pro Intervall die Information.

naltatis, 2024-03-19:
Ok, heißt, hier läuft eigentlich alles wie erwartet bzw. zumindest in den Rahmen in denen wir das designed haben. Also Differenz maximal ein Update-Interval. Korrekt?

Ich kann die Irritation verstehen, aber da es sich dann ja geraderuckelt und es keine Auswirkungen auf die eigentliche Funktion hat, würde ich jetzt hier erstmal beim aktuellen Verhalten bleiben.

andig, 2024-03-20:
> Der Zähler mit der Meldung "Zu wenig Überschuss. Pausiere in 3:00 m ..." erscheint und fängt an von der eingestellten Zeit (disable delay - in meinem Fall entsprechend 3 Minuten) an rückwärts zu zählen.

> Ok, heißt, hier läuft eigentlich alles wie erwartet bzw. zumindest in den Rahmen in denen wir das designed haben. Also Differenz maximal ein Update-Interval. Korrekt?

Korrekt- ich hatte die 3min anders interpretiert. Hier funktioniert also alles wie erwartet. Die Timer-Updates ins UI kommen nur im Rahmen von `interval`, die Abweichung ist also prinzipbedingt.

closed 2024-03-20 by andig

SolarPowerEV, 2024-03-28:
Was ich halt trotz allem seltsam finde, ist dass evcc schon weiß und anzeigt, dass der Ladepunkt nur noch 0 Watt zieht (sprich die Ladung beendet wurde), der Timer aber trotzdem weiter heruntergezählt wird. Kann man nicht zumindest in dem Moment, wo faktisch das Signal zum Beenden des Ladevorganges an die Wallbox gesendet wurde UND die Wallbox schon die 0 Watt Leistung an evcc zurückgemeldet hat, zumindest den Abschalt-Timer einfach sofort vom GUI runternehmen? 

Das müsste doch möglich sein, wenn das GUI auch sofort mitbekommt, wenn ich am Handy z.B. von PV auf AUS oder SCHNELL umschalte und das in Echtzeit auf dem Desktop-Browser anzeigt?

andig, 2024-03-28:
@naltatis ?