evcc-io/evcc #13007: Tesla: in PV mode, charge always starts at 5A despite min current being set to 1A, causing start/stop oscillation

FraBoCH, 2024-03-17:
### Describe the bug

I have a Tesla charging on a Tesla Wall Connector Gen 3 (twc3). I configured the min current on the loadpoint to 1A (~700w with 3 phases) to charge even with few excess power.
The problem is that in PV mode, the charge always starts at 5A, thus using more current than available and stopping, then starting again, and stopping, ....
In Min+PV mode, the 1A is properly sent to the vehicle and taken into account. So the charge starts at ~700w and then increase/decrease properly with the available solar power in excess.

### Steps to reproduce

1. Configure min current of loadpoint to 1amp: 
2. Wait for enough solar current being generated to start charging (400w with my config)
3. Charging starts, but at 5amps => 3.6kw and not at 1amp (~700w): 
[image]
4. As the used power is now higher than the threshold (1200w with my config), it immediately decides to stop charging
5. During all the disable delay, no try is made to reduce the current, it remains at 5amp during the whole delay (10min with my config).
6. I am able to force the current being reduced to the expected 700w value by: setting min current of loadpoint to 2amp, wait for the order to be sent to the tesla, then set it again to 1amp. Or by switching to min+pv.

### Configuration details

```yaml
chargers:
- type: template
  template: twc3 
  host: 192.168.1.16
  name: wallbox

loadpoints:
- title: Garage
  charger: wallbox
  mode: pv
  enable:
    threshold: -400
    delay: 1m
  disable:
    threshold: 1200
    delay: 10m
    
vehicles:
- type: template
  template: tesla
  accessToken: ****
  refreshToken: ****
  vin: ****
  control: true
  name: tesla
```

### Log details

```text
[site  ] DEBUG 2024/03/17 14:45:42 grid currents: [-1.68 -1.31 1.84]A
[site  ] DEBUG 2024/03/17 14:45:42 site power: -396W
[lp-1  ] DEBUG 2024/03/17 14:45:42 charge voltages: [0 1.7 0]V
[lp-1  ] DEBUG 2024/03/17 14:45:42 charge currents: [0 0.3 0.4]A
[lp-1  ] DEBUG 2024/03/17 14:45:42 charger status: B
[lp-1  ] DEBUG 2024/03/17 14:45:42 pv charge current: 0.574A = 0A + 0.574A (-396W @ 3p)
[site  ] DEBUG 2024/03/17 14:46:12 ----
[lp-1  ] DEBUG 2024/03/17 14:46:12 charge power: 0W
[site  ] DEBUG 2024/03/17 14:46:12 pv power: 1410W
[site  ] DEBUG 2024/03/17 14:46:12 grid meter: -462W
[site  ] DEBUG 2024/03/17 14:46:12 grid powers: [-400 -331 268]W
[site  ] DEBUG 2024/03/17 14:46:12 grid currents: [-1.72 -1.6 1.89]A
[site  ] DEBUG 2024/03/17 14:46:12 site power: -462W
[lp-1  ] DEBUG 2024/03/17 14:46:12 charge voltages: [0 1.7 0]V
[lp-1  ] DEBUG 2024/03/17 14:46:12 charge currents: [0 0.3 0.4]A
[lp-1  ] DEBUG 2024/03/17 14:46:12 charger status: B
[lp-1  ] DEBUG 2024/03/17 14:46:12 pv charge current: 0.67A = 0A + 0.67A (-462W @ 3p)
[lp-1  ] DEBUG 2024/03/17 14:46:12 site power -462W <= -400W enable threshold
[lp-1  ] DEBUG 2024/03/17 14:46:12 pv enable timer start: 1m0s
[lp-1  ] DEBUG 2024/03/17 14:46:12 pv enable in 1m0s
[site  ] DEBUG 2024/03/17 14:46:42 ----
[lp-1  ] DEBUG 2024/03/17 14:46:42 charge power: 0W
[site  ] DEBUG 2024/03/17 14:46:42 pv power: 1446W
[site  ] DEBUG 2024/03/17 14:46:42 grid meter: -428W
[site  ] DEBUG 2024/03/17 14:46:42 grid powers: [-411 -272 254]W
[site  ] DEBUG 2024/03/17 14:46:42 grid currents: [-1.78 -1.44 1.89]A
[site  ] DEBUG 2024/03/17 14:46:42 site power: -428W
[lp-1  ] DEBUG 2024/03/17 14:46:42 charge voltages: [0 1.7 0]V
[lp-1  ] DEBUG 2024/03/17 14:46:42 charge currents: [0 0.3 0.4]A
[lp-1  ] DEBUG 2024/03/17 14:46:42 charger status: B
[lp-1  ] DEBUG 2024/03/17 14:46:42 pv charge current: 0.621A = 0A + 0.621A (-428W @ 3p)
[lp-1  ] DEBUG 2024/03/17 14:46:42 site power -428W <= -400W enable threshold
[lp-1  ] DEBUG 2024/03/17 14:46:42 pv enable in 30s
[lp-1  ] DEBUG 2024/03/17 14:46:42 pv enable timer remaining: 30s
[site  ] DEBUG 2024/03/17 14:47:12 ----
[lp-1  ] DEBUG 2024/03/17 14:47:12 charge power: 0W
[site  ] DEBUG 2024/03/17 14:47:42 pv power: 1551W
[site  ] DEBUG 2024/03/17 14:47:42 grid meter: -563W
[site  ] DEBUG 2024/03/17 14:47:42 grid powers: [-445 -347 230]W
[site  ] DEBUG 2024/03/17 14:47:42 grid currents: [-1.91 -1.5 1.78]A
[site  ] DEBUG 2024/03/17 14:47:42 site power: -563W
[lp-1  ] DEBUG 2024/03/17 14:47:42 charge voltages: [0 1.7 0]V
[lp-1  ] DEBUG 2024/03/17 14:47:42 charge currents: [0 0.3 0.4]A
[lp-1  ] DEBUG 2024/03/17 14:47:42 charger status: B
[lp-1  ] DEBUG 2024/03/17 14:47:42 pv charge current: 0.816A = 0A + 0.816A (-563W @ 3p)
[lp-1  ] DEBUG 2024/03/17 14:47:42 site power -563W <= -400W enable threshold
[lp-1  ] DEBUG 2024/03/17 14:47:42 pv enable in 0s
[lp-1  ] DEBUG 2024/03/17 14:47:42 pv enable timer elapsed
[lp-1  ] DEBUG 2024/03/17 14:47:44 charger enable: waking up vehicle
[lp-1  ] ERROR 2024/03/17 14:47:45 charger enable: asleep
[site  ] DEBUG 2024/03/17 14:48:12 ----
[lp-1  ] DEBUG 2024/03/17 14:48:12 charge power: 0W
[site  ] DEBUG 2024/03/17 14:48:12 pv power: 1605W
[site  ] DEBUG 2024/03/17 14:48:12 grid meter: -539W
[site  ] DEBUG 2024/03/17 14:48:12 grid powers: [-464 -317 248]W
[site  ] DEBUG 2024/03/17 14:48:12 grid currents: [-1.99 -1.63 1.77]A
[site  ] DEBUG 2024/03/17 14:48:12 site power: -539W
[lp-1  ] DEBUG 2024/03/17 14:48:12 charge voltages: [0 1.7 0]V
[lp-1  ] DEBUG 2024/03/17 14:48:12 charge currents: [0 0.3 0.4]A
[lp-1  ] DEBUG 2024/03/17 14:48:12 charger status: B
[lp-1  ] DEBUG 2024/03/17 14:48:12 pv charge current: 0.782A = 0A + 0.782A (-539W @ 3p)
[lp-1  ] DEBUG 2024/03/17 14:48:12 site power -539W <= -400W enable threshold
[lp-1  ] DEBUG 2024/03/17 14:48:12 pv enable in 0s
[lp-1  ] DEBUG 2024/03/17 14:48:12 pv enable timer elapsed
[lp-1  ] DEBUG 2024/03/17 14:48:13 charger enable
[lp-1  ] DEBUG 2024/03/17 14:48:13 wake-up timer: start
[site  ] DEBUG 2024/03/17 14:48:42 ----
[lp-1  ] DEBUG 2024/03/17 14:48:42 charge power: 3638W
[site  ] DEBUG 2024/03/17 14:48:42 pv power: 1708W
[site  ] DEBUG 2024/03/17 14:48:42 grid meter: 2922W
[site  ] DEBUG 2024/03/17 14:48:42 grid powers: [718 787 1416]W
[site  ] DEBUG 2024/03/17 14:48:42 grid currents: [3.11 3.72 6.12]A
[site  ] DEBUG 2024/03/17 14:48:42 site power: 2922W
[lp-1  ] DEBUG 2024/03/17 14:48:42 charge voltages: [237 238 238]V
[lp-1  ] DEBUG 2024/03/17 14:48:42 detected connected phases: 3p
[lp-1  ] DEBUG 2024/03/17 14:48:42 charge currents: [5.1 5.1 5.1]A
[lp-1  ] DEBUG 2024/03/17 14:48:42 charger status: C
[lp-1  ] INFO 2024/03/17 14:48:42 start charging ->
[lp-1  ] DEBUG 2024/03/17 14:48:42 wake-up timer: stop
[lp-1  ] DEBUG 2024/03/17 14:48:43 vehicle soc: 60%
[lp-1  ] DEBUG 2024/03/17 14:48:43 vehicle soc limit: 80%
[lp-1  ] DEBUG 2024/03/17 14:48:43 vehicle range: 286km
[lp-1  ] DEBUG 2024/03/17 14:48:43 pv charge current: 0A = 1A + -4.23A (2922W @ 3p)
[lp-1  ] DEBUG 2024/03/17 14:48:43 projected site power 2922W >= 1200W disable threshold
[lp-1  ] DEBUG 2024/03/17 14:48:43 pv disable in 7m30s
[lp-1  ] DEBUG 2024/03/17 14:48:43 pv disable timer remaining: 7m30s
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.124.10

StefanSchoof, 2024-03-17:
Take a look at https://github.com/evcc-io/evcc/issues/12427

FraBoCH, 2024-03-17:
Hi. I checked #12427 which indeed look similar, but I think I'm in a slightly different situation here, as the issue I observe is cycle of  start/stop of charge. I wouldn't mind if it was starting at 5A and reduce the current afterward, this is not the case. Unless I manually change something, the charging is not working properly.

andig, 2024-03-17:
@FraBoCH could you kindly set `tesla: trace` to see what's going on here?

andig, 2024-03-17:
In https://github.com/evcc-io/evcc/issues/12427#issuecomment-1962548105 it seemed as if evcc sent the right command but is was ignored by the vehicle. Please provide log of MinPV (working) and PV (not working).

FraBoCH, 2024-03-17:
Here is some additional logs from today. I will add later new logs with trace option on.

First, while the disable timer timer is running, without any apparent request to reduce the current (and it will eventually just stop the charge): 
[lp-1  ] DEBUG 2024/03/17 14:50:42 vehicle range: 286km
[lp-1  ] DEBUG 2024/03/17 14:50:42 pv charge current: 0A = 1A + -10.6A (7285W @ 3p)
[lp-1  ] DEBUG 2024/03/17 14:50:42 projected site power 7285W >= 1200W disable threshold
[lp-1  ] DEBUG 2024/03/17 14:50:42 pv disable in 5m30s
[lp-1  ] DEBUG 2024/03/17 14:50:42 pv disable timer remaining: 5m30s
[site  ] DEBUG 2024/03/17 14:51:12 ----
[lp-1  ] DEBUG 2024/03/17 14:51:12 charge power: 3648W
[site  ] DEBUG 2024/03/17 14:51:12 pv power: 1785W
[site  ] DEBUG 2024/03/17 14:51:12 grid meter: 2882W
[site  ] DEBUG 2024/03/17 14:51:12 grid powers: [698 763 1417]W
[site  ] DEBUG 2024/03/17 14:51:12 grid currents: [3.03 3.39 6.16]A
[site  ] DEBUG 2024/03/17 14:51:12 site power: 2882W
[lp-1  ] DEBUG 2024/03/17 14:51:12 charge voltages: [238 239 239]V
[lp-1  ] DEBUG 2024/03/17 14:51:12 detected connected phases: 3p
[lp-1  ] DEBUG 2024/03/17 14:51:12 charge currents: [5.1 5.1 5.1]A
[lp-1  ] DEBUG 2024/03/17 14:51:12 detected active phases: 3p
[lp-1  ] DEBUG 2024/03/17 14:51:12 charger status: C
[lp-1  ] DEBUG 2024/03/17 14:51:12 vehicle soc: 60%
[lp-1  ] DEBUG 2024/03/17 14:51:12 vehicle soc limit: 80%
[lp-1  ] DEBUG 2024/03/17 14:51:12 vehicle range: 286km
[lp-1  ] DEBUG 2024/03/17 14:51:12 pv charge current: 0A = 1A + -4.18A (2882W @ 3p)
[lp-1  ] DEBUG 2024/03/17 14:51:12 projected site power 2882W >= 1200W disable threshold
[lp-1  ] DEBUG 2024/03/17 14:51:12 pv disable in 5m0s
[lp-1  ] DEBUG 2024/03/17 14:51:12 pv disable timer remaining: 5m0s
[site  ] DEBUG 2024/03/17 14:51:42 ----
[lp-1  ] DEBUG 2024/03/17 14:51:42 charge power: 3651W
[site  ] DEBUG 2024/03/17 14:51:42 pv power: 1753W
[site  ] DEBUG 2024/03/17 14:51:42 grid meter: 2916W
[site  ] DEBUG 2024/03/17 14:51:42 grid powers: [709 765 1417]W
[site  ] DEBUG 2024/03/17 14:51:42 grid currents: [3.07 3.39 6.17]A
[site  ] DEBUG 2024/03/17 14:51:42 site power: 2916W
[lp-1  ] DEBUG 2024/03/17 14:51:42 charge voltages: [238 239 239]V
[lp-1  ] DEBUG 2024/03/17 14:51:42 detected connected phases: 3p
[lp-1  ] DEBUG 2024/03/17 14:51:42 charge currents: [5.1 5.1 5.1]A
[lp-1  ] DEBUG 2024/03/17 14:51:42 detected active phases: 3p
[lp-1  ] DEBUG 2024/03/17 14:51:42 charger status: C
[lp-1  ] DEBUG 2024/03/17 14:51:42 vehicle soc: 60%
[lp-1  ] DEBUG 2024/03/17 14:51:42 vehicle soc limit: 80%
[lp-1  ] DEBUG 2024/03/17 14:51:42 vehicle range: 286km
[lp-1  ] DEBUG 2024/03/17 14:51:42 pv charge current: 0A = 1A + -4.23A (2916W @ 3p)
[lp-1  ] DEBUG 2024/03/17 14:51:42 projected site power 2916W >= 1200W disable threshold
[lp-1  ] DEBUG 2024/03/17 14:51:42 pv disable in 4m30s
[lp-1  ] DEBUG 2024/03/17 14:51:42 pv disable timer remaining: 4m30s
[site  ] DEBUG 2024/03/17 14:52:12 ----
[lp-1  ] DEBUG 2024/03/17 14:52:12 charge power: 3658W
[site  ] DEBUG 2024/03/17 14:52:12 pv power: 1765W
[site  ] DEBUG 2024/03/17 14:52:12 grid meter: 2897W
[site  ] DEBUG 2024/03/17 14:52:12 grid powers: [707 773 1412]W
[site  ] DEBUG 2024/03/17 14:52:12 grid currents: [3.05 3.54 6.1]A
[site  ] DEBUG 2024/03/17 14:52:12 site power: 2897W
[lp-1  ] DEBUG 2024/03/17 14:52:12 charge voltages: [238 239 240]V
[lp-1  ] DEBUG 2024/03/17 14:52:12 detected connected phases: 3p
[lp-1  ] DEBUG 2024/03/17 14:52:12 charge currents: [5.1 5.1 5.1]A
[lp-1  ] DEBUG 2024/03/17 14:52:12 detected active phases: 3p
[lp-1  ] DEBUG 2024/03/17 14:52:12 charger status: C
[lp-1  ] DEBUG 2024/03/17 14:52:12 vehicle soc: 60%
[lp-1  ] DEBUG 2024/03/17 14:52:12 vehicle soc limit: 80%
[lp-1  ] DEBUG 2024/03/17 14:52:12 vehicle range: 286km
[lp-1  ] DEBUG 2024/03/17 14:52:12 pv charge current: 0A = 1A + -4.2A (2897W @ 3p)
[lp-1  ] DEBUG 2024/03/17 14:52:12 projected site power 2897W >= 1200W disable threshold
[lp-1  ] DEBUG 2024/03/17 14:52:12 pv disable in 4m0s
[lp-1  ] DEBUG 2024/03/17 14:52:12 pv disable timer remaining: 4m0s
[site  ] DEBUG 2024/03/17 14:52:42 ----
[lp-1  ] DEBUG 2024/03/17 14:52:42 charge power: 3657W
[site  ] DEBUG 2024/03/17 14:52:42 pv power: 1779W
[site  ] DEBUG 2024/03/17 14:52:42 grid meter: 2911W
[site  ] DEBUG 2024/03/17 14:52:42 grid powers: [704 778 1422]W
[... 92 lines cut ...]
[lp-1  ] DEBUG 2024/03/17 15:05:12 charge currents: [1.2 1.2 1.2]A
[lp-1  ] DEBUG 2024/03/17 15:05:12 detected active phases: 3p
[lp-1  ] DEBUG 2024/03/17 15:05:12 charger status: C
[lp-1  ] DEBUG 2024/03/17 15:05:12 vehicle soc: 61%
[lp-1  ] DEBUG 2024/03/17 15:05:12 vehicle soc limit: 80%
[lp-1  ] DEBUG 2024/03/17 15:05:12 vehicle range: 289km
[lp-1  ] DEBUG 2024/03/17 15:05:12 pv charge current: 0.712A = 1A + -0.288A (199W @ 3p)
[site  ] DEBUG 2024/03/17 15:05:42 ----
[lp-1  ] DEBUG 2024/03/17 15:05:42 charge power: 859W
[site  ] DEBUG 2024/03/17 15:05:42 pv power: 1453W
[site  ] DEBUG 2024/03/17 15:05:42 grid meter: 251W
[site  ] DEBUG 2024/03/17 15:05:42 grid powers: [-175 -131 557]W
[site  ] DEBUG 2024/03/17 15:05:42 grid currents: [-1.08 -1.19 2.88]A
[site  ] DEBUG 2024/03/17 15:05:42 site power: 251W
[lp-1  ] DEBUG 2024/03/17 15:05:42 charge voltages: [238 239 238]V
[lp-1  ] DEBUG 2024/03/17 15:05:42 detected connected phases: 3p
[lp-1  ] DEBUG 2024/03/17 15:05:42 charge currents: [1.2 1.2 1.2]A
[lp-1  ] DEBUG 2024/03/17 15:05:42 detected active phases: 3p
[lp-1  ] DEBUG 2024/03/17 15:05:42 charger status: C
[lp-1  ] DEBUG 2024/03/17 15:05:42 vehicle soc: 61%
[lp-1  ] DEBUG 2024/03/17 15:05:42 vehicle soc limit: 80%
[lp-1  ] DEBUG 2024/03/17 15:05:42 vehicle range: 289km
[lp-1  ] DEBUG 2024/03/17 15:05:42 pv charge current: 0.637A = 1A + -0.363A (251W @ 3p)

It's visible in minPV that there is a "max charge current: 1A" line in the beginning which is not visible in the pv mode

FraBoCH, 2024-03-17:
I cannot test with pv mode now as there is no more sun. However, I've made some progress investigating this issue with the following steps: 
- solar -> now : charge starts at 8A (matching my setting)
- now -> minPV: set_charging_amps 1 is sent to tesla, but power is reduced to 3580W (-> 5A) ! So here, the tesla ignored partially the request: it reduced the power, but not to 1A as requested
- change min current in loadpoint popup to 2: set_charging_amps 2 is sent to tesla, and power is reduced accordingly (1483W)
- change min current in loadpoint popup to 1: set_charging_amps 1 is sent to tesla, and power is reduced accordingly (779W)

So as I see it, tesla reduce the power but is somewhat "blocking" at 5A when going down, requiring an additional request to go further down in power. In other words, evcc cannot assume tesla executes properly the set_charging_amps and should resend the request until the actual charge power match the expectation.

Here is the corresponding log, with the tesla trace:
[lp-1  ] DEBUG 2024/03/17 20:05:26 set charge mode: now
[site  ] DEBUG 2024/03/17 20:05:26 ----
[lp-1  ] DEBUG 2024/03/17 20:05:26 charge power: 3576W
[site  ] DEBUG 2024/03/17 20:05:26 pv power: 0W
[site  ] DEBUG 2024/03/17 20:05:26 grid meter: 4045W
[site  ] DEBUG 2024/03/17 20:05:27 grid powers: [1334 1297 1415]W
[site  ] DEBUG 2024/03/17 20:05:27 grid currents: [5.7 5.6 6.16]A
[site  ] DEBUG 2024/03/17 20:05:27 site power: 4045W
[lp-1  ] DEBUG 2024/03/17 20:05:27 charge voltages: [233 234 234]V
[lp-1  ] DEBUG 2024/03/17 20:05:27 detected connected phases: 3p
[lp-1  ] DEBUG 2024/03/17 20:05:27 charge currents: [5.1 5.1 5.1]A
[lp-1  ] DEBUG 2024/03/17 20:05:27 detected active phases: 3p
[lp-1  ] DEBUG 2024/03/17 20:05:27 charger status: C
[lp-1  ] DEBUG 2024/03/17 20:05:27 vehicle soc: 62%
[lp-1  ] DEBUG 2024/03/17 20:05:27 vehicle soc limit: 80%
[lp-1  ] DEBUG 2024/03/17 20:05:27 vehicle range: 294km
[tesla ] TRACE 2024/03/17 20:05:27 POST https://tesla.evcc.io/api/1/vehicles/********/command/set_charging_amps
[tesla ] TRACE 2024/03/17 20:05:27 {"charging_amps": 8}
--
{"response":{"result":true,"reason":""}}
[lp-1  ] DEBUG 2024/03/17 20:05:27 max charge current: 8A
[site  ] DEBUG 2024/03/17 20:05:53 ----
[lp-1  ] DEBUG 2024/03/17 20:05:53 charge power: 5664W
[site  ] DEBUG 2024/03/17 20:05:54 pv power: 0W
[site  ] DEBUG 2024/03/17 20:05:54 grid meter: 6244W
[site  ] DEBUG 2024/03/17 20:05:54 grid powers: [2059 2054 2131]W
[site  ] DEBUG 2024/03/17 20:05:54 grid currents: [8.75 8.81 9.18]A
[site  ] DEBUG 2024/03/17 20:05:54 site power: 6244W
[lp-1  ] DEBUG 2024/03/17 20:05:54 charge voltages: [232 233 234]V
[lp-1  ] DEBUG 2024/03/17 20:05:54 detected connected phases: 3p
[lp-1  ] DEBUG 2024/03/17 20:05:54 charge currents: [8.1 8.1 8.1]A
[lp-1  ] DEBUG 2024/03/17 20:05:54 detected active phases: 3p
[lp-1  ] DEBUG 2024/03/17 20:05:54 charger status: C
[lp-1  ] DEBUG 2024/03/17 20:05:54 vehicle soc: 62%
[lp-1  ] DEBUG 2024/03/17 20:05:54 vehicle soc limit: 80%
[lp-1  ] DEBUG 2024/03/17 20:05:54 vehicle range: 294km

[lp-1  ] DEBUG 2024/03/17 20:07:25 set charge mode: minpv
[site  ] DEBUG 2024/03/17 20:07:25 ----
[lp-1  ] DEBUG 2024/03/17 20:07:25 charge power: 5667W
[site  ] DEBUG 2024/03/17 20:07:25 pv power: 0W
[site  ] DEBUG 2024/03/17 20:07:25 grid meter: 6232W
[site  ] DEBUG 2024/03/17 20:07:25 grid powers: [2050 2056 2132]W
[site  ] DEBUG 2024/03/17 20:07:25 grid currents: [8.73 8.75 9.16]A
[site  ] DEBUG 2024/03/17 20:07:25 site power: 6232W
[lp-1  ] DEBUG 2024/03/17 20:07:25 charge voltages: [233 233 234]V
[lp-1  ] DEBUG 2024/03/17 20:07:25 detected connected phases: 3p
[lp-1  ] DEBUG 2024/03/17 20:07:25 charge currents: [8.1 8.1 8.1]A
[lp-1  ] DEBUG 2024/03/17 20:07:25 detected active phases: 3p
[lp-1  ] DEBUG 2024/03/17 20:07:25 charger status: C
[lp-1  ] DEBUG 2024/03/17 20:07:25 vehicle soc: 62%
[lp-1  ] DEBUG 2024/03/17 20:07:25 vehicle soc limit: 80%
[lp-1  ] DEBUG 2024/03/17 20:07:25 vehicle range: 294km
[lp-1  ] DEBUG 2024/03/17 20:07:25 pv charge current: 0A = 8A + -9.03A (6232W @ 3p)
[tesla ] TRACE 2024/03/17 20:07:25 POST https://tesla.evcc.io/api/1/vehicles/********/command/set_charging_amps
[tesla ] TRACE 2024/03/17 20:07:26 {"charging_amps": 1}
--
{"response":{"result":true,"reason":""}}
[lp-1  ] DEBUG 2024/03/17 20:07:26 max charge current: 1A
[site  ] DEBUG 2024/03/17 20:07:53 ----
[lp-1  ] DEBUG 2024/03/17 20:07:53 charge power: 3580W
[... 81 lines cut ...]
[lp-1  ] DEBUG 2024/03/17 20:09:54 charge voltages: [236 236 236]V
[lp-1  ] DEBUG 2024/03/17 20:09:54 detected connected phases: 3p
[lp-1  ] DEBUG 2024/03/17 20:09:54 charge currents: [1.1 1.1 1.1]A
[lp-1  ] DEBUG 2024/03/17 20:09:54 detected active phases: 3p
[lp-1  ] DEBUG 2024/03/17 20:09:54 charger status: C
[lp-1  ] DEBUG 2024/03/17 20:09:54 vehicle soc: 62%
[lp-1  ] DEBUG 2024/03/17 20:09:54 vehicle soc limit: 80%
[lp-1  ] DEBUG 2024/03/17 20:09:54 vehicle range: 294km
[lp-1  ] DEBUG 2024/03/17 20:09:54 pv charge current: 0A = 1A + -1.77A (1218W @ 3p)
[site  ] DEBUG 2024/03/17 20:10:23 ----
[lp-1  ] DEBUG 2024/03/17 20:10:23 charge power: 779W
[site  ] DEBUG 2024/03/17 20:10:24 pv power: 0W
[site  ] DEBUG 2024/03/17 20:10:24 grid meter: 1225W
[site  ] DEBUG 2024/03/17 20:10:24 grid powers: [372 394 459]W
[site  ] DEBUG 2024/03/17 20:10:24 grid currents: [1.73 1.82 2.24]A
[site  ] DEBUG 2024/03/17 20:10:24 site power: 1225W
[lp-1  ] DEBUG 2024/03/17 20:10:24 charge voltages: [236 236 236]V
[lp-1  ] DEBUG 2024/03/17 20:10:24 detected connected phases: 3p
[lp-1  ] DEBUG 2024/03/17 20:10:24 charge currents: [1.1 1.1 1.1]A
[lp-1  ] DEBUG 2024/03/17 20:10:24 detected active phases: 3p
[lp-1  ] DEBUG 2024/03/17 20:10:24 charger status: C
[lp-1  ] DEBUG 2024/03/17 20:10:24 vehicle soc: 62%
[lp-1  ] DEBUG 2024/03/17 20:10:24 vehicle soc limit: 80%
[lp-1  ] DEBUG 2024/03/17 20:10:24 vehicle range: 294km
[lp-1  ] DEBUG 2024/03/17 20:10:24 pv charge current: 0A = 1A + -1.78A (1225W @ 3p)

Adminius, 2024-03-19:
a while ago, "under 5A" worked only if the set_charging_amps was sended twice (this was only requierd for values 1-4A)

andig, 2024-03-19:
Good input, thank you. I was already wondering what made the difference. Sounds like a bug in the Tesla Fleet api that should be fixed?

Adminius, 2024-03-19:
That was with the old unofficial API.
I'm using now EVCC with TeslaLogger with FleetAPI via MQTT with 2A-16A setting and I mean it works with 2A without problem... 
But never really looked if there any problems or not. I have to check next days, but car battery is full now

andig, 2024-03-19:
> So as I see it, tesla reduce the power but is somewhat "blocking" at 5A when going down, requiring an additional request to go further down in power. In other words, evcc cannot assume tesla [...]

We could try doing something like "send command twice if power <5A and last value >=5A".

FraBoCH, 2024-03-19:
> We could try doing something like "send command twice if power <5A and last value >=5A".

This looks like a possible workaround. 
I also thought about:  
1. resending the set_target_amp before starting the charge, so in case of "start/stop" cycles it may eventually work
2. adding a control loop to read back the current amp st on the vehicle and send command again a few times if that's not matching the expected value

I would be happy to beta-test a fix.

mucki12, 2024-03-19:
Würde das als damaliger Ersteller von https://github.com/evcc-io/evcc/issues/12427 natürlich auch gerne testen.

Raudi1, 2024-03-21:
Vielleicht hängt das auch damit zusammen, dass man in der Tesla App den Strom nur auf minimal 5A stellen kann. Ich hoffe, dass die Möglichkeit per API da drunter zu gehen nicht nur ein Bug ist, den Tesla irgendwann schließt...

premultiply, 2024-03-21:
Naja, wirklich Sinn macht das ohnehin nicht, denn auch Tesla kann nicht zaubern.
Der Wirkungsgrad bei niedrigen Leistungen ist in der Regel schon schlecht, irgendwann kommt sogar die Schwelle dass das Fahrzeug mehr durch seine laufenden Systeme verbraucht als das Ladesystem abzüglich seines an diesem Arbeitspunkt miserablen Wirkungsgrad überhaupt ins Bordnetz einspeist.
Daher ist im IEC-Standard aus gutem Grund 6A als Mindestleistungsvorgabe gewählt.

Raudi1, 2024-03-21:
Ich habe leider nur eine 1,75kWp PV. Da bin ich auf die niedrige Stromstärke angewiesen. Und es funktioniert auch besser als erwartet. Die Effizienz ist so bei 75% Minimum laut Teslamate. Normalerweise ist sie sogar eher bei mindestens 80-90%. Hat zumindest selbst an eher durchwachsenen Tagen wie den letzten 2 dafür gereicht 10% in die Batterie zu bekommen. Wenn das Wetter konstant sonnig ist, kann ich so wahrscheinlich teilweise komplett nur mit PV laden, da ich regulär nicht viel Auto fahre, besonders bei gutem Wetter. Und ich kann mir einen Heimspeicher sparen.

andig, 2024-03-22:
Könnte jemand https://github.com/evcc-io/evcc/pull/13099 testen wenn ich das in einem Docker Build zur Verfügung stelle?

FraBoCH, 2024-03-22:
> Könnte jemand #13099 testen wenn ich das in einem Docker Build zur Verfügung stelle?

Thanks a lot @andig for working on a fix !
I would test it with pleasure, even if I need to figure out how to run that ;-). 
Is there a way to provide a "beta version" that can be installed in home assistant ?

mucki12, 2024-03-22:
@andig 
Kann ich auf jeden Fall testen, aber leider erst in einer Woche. Würde sich das mit dem testing Docker für das Batterieladen beißen? Wenn nicht wäre das natürlich mein Traum ;)

Raudi1, 2024-03-22:
> Könnte jemand #13099 testen wenn ich das in einem Docker Build zur Verfügung stelle?

Sollte machbar sein. Es muss aber natürlich auch das Wetter mitspielen.

andig, 2024-03-23:
Build läuft, Tag `:tesla`

FraBoCH, 2024-03-23:
It's the first time I'm running evcc docker image so I may be doing something wrong. 
I launched using: docker run -d --name evcc -v .\evcc.yaml:/etc/evcc.yaml -v /.evcc:/root/.evcc -p 7070:7070 evcc/evcc:tesla

But the vehicle fails at creation: 
[main  ] INFO 2024/03/23 12:23:58 evcc 0.124.10 (cfb35a455)
[main  ] INFO 2024/03/23 12:23:58 using config file: /etc/evcc.yaml
[main  ] INFO 2024/03/23 12:23:58 starting ui and api at :7070
[db    ] INFO 2024/03/23 12:24:00 using sqlite database: /root/.evcc/evcc.db
[main  ] ERROR 2024/03/23 12:24:00 creating vehicle tesla failed: cannot create vehicle type 'template': invalid vehicle type: tesla

So then, quite logically, I get a "vehicle not capable of current control" whenever trying to change something.

andig, 2024-03-23:
Ups- scheint mein local build wäre nicht erfolgreich gewesen. Probiers bitte nochmal.

FraBoCH, 2024-03-23:
Ok. Trying with the :testing docker image, as it looks like it's the one you just updated ;-)
[image]

FraBoCH, 2024-03-23:
Same result with :testing image. The logs are quite similar to what happens when I try to build & run locally without a proper TESLA_CLIENT_ID, maybe this is what's missing in your build ?

andig, 2024-03-23:
Sorry for the confusion. `:tesla` should be fixed now.

FraBoCH, 2024-03-23:
I've been able to run and quickly test the :tesla branch.
I see no effect with the current steps: 
1. solar -> now : charge starts at 8A (matching my setting)
2. now -> minPV: set_charging_amps 1 is sent, but only once as per the log, and the actual current used be the tesla remains 5A and not the expected 1A.
3. I've let it run couple of minutes to see if a new set_charging_amps request would be sent but it was not the case, and then manually stopped the charge.. 

Here is the corresponding log: 
[main  ] INFO 2024/03/23 17:50:10 evcc 0.124.10 (f344dcb94)
[main  ] INFO 2024/03/23 17:50:10 using config file: /etc/evcc.yaml
[main  ] INFO 2024/03/23 17:50:10 starting ui and api at :7070
[db    ] INFO 2024/03/23 17:50:12 using sqlite database: /root/.evcc/evcc.db
[tesla ] TRACE 2024/03/23 17:50:12 GET https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/users/region
[tesla ] TRACE 2024/03/23 17:50:15 {"response":{"region":"eu","fleet_api_base_url":"https://fleet-api.prd.eu.vn.cloud.tesla.com"}}
[tesla ] TRACE 2024/03/23 17:50:15 GET https://fleet-api.prd.eu.vn.cloud.tesla.com/api/1/vehicles
[tesla ] TRACE 2024/03/23 17:50:17 {"response":[{"id":1493081361345419,....}],"pagination":{"previous":null,"next":null,"current":1,"per_page":2,"count":2,"pages":1},"count":2}
[site  ] INFO 2024/03/23 17:50:22 site config:
[site  ] INFO 2024/03/23 17:50:22   meters:      grid Γ£ô pv Γ£ô battery Γ£ù
[site  ] INFO 2024/03/23 17:50:22     grid:      power Γ£ô energy Γ£ô currents Γ£ô
[site  ] INFO 2024/03/23 17:50:22     pv 1:      power Γ£ô energy Γ£ô currents Γ£ô
[site  ] INFO 2024/03/23 17:50:22   vehicles:
[site  ] INFO 2024/03/23 17:50:22     vehicle 1: range Γ£ô finish Γ£ù status Γ£ô climate Γ£ù wakeup Γ£ù
[site  ] INFO 2024/03/23 17:50:22     vehicle 2: range Γ£ô finish Γ£ô status Γ£ô climate Γ£ù wakeup Γ£ô
[lp-1  ] INFO 2024/03/23 17:50:22 loadpoint 1:
[lp-1  ] INFO 2024/03/23 17:50:22   mode:        pv
[lp-1  ] INFO 2024/03/23 17:50:22   charger:     power Γ£ô energy Γ£ù currents Γ£ô phases Γ£ù wakeup Γ£ù
[lp-1  ] INFO 2024/03/23 17:50:22   meters:      charge Γ£ô
[lp-1  ] INFO 2024/03/23 17:50:22     charge:    power Γ£ô energy Γ£ù currents Γ£ô
[lp-1  ] DEBUG 2024/03/23 17:50:22 phase timer inactive
[lp-1  ] DEBUG 2024/03/23 17:50:22 pv timer inactive
[site  ] DEBUG 2024/03/23 17:50:22 ----
[lp-1  ] DEBUG 2024/03/23 17:50:22 charge power: 0W
[site  ] DEBUG 2024/03/23 17:50:22 pv power: 99W
[site  ] DEBUG 2024/03/23 17:50:23 grid meter: 178W
[site  ] DEBUG 2024/03/23 17:50:23 grid powers: [40 -26 164]W
[site  ] DEBUG 2024/03/23 17:50:23 grid currents: [0.56 -0.52 1.25]A
[site  ] DEBUG 2024/03/23 17:50:23 site power: 178W
[lp-1  ] DEBUG 2024/03/23 17:50:24 charge voltages: [0 1.7 0]V
[lp-1  ] DEBUG 2024/03/23 17:50:24 charge currents: [0 0.3 0.4]A
[lp-1  ] DEBUG 2024/03/23 17:50:24 charger status: B
[lp-1  ] INFO 2024/03/23 17:50:24 car connected
[lp-1  ] DEBUG 2024/03/23 17:50:24 vehicle api refresh
[lp-1  ] DEBUG 2024/03/23 17:50:24 pv timer elapse
[lp-1  ] DEBUG 2024/03/23 17:50:24 pv timer inactive
[main  ] ERROR 2024/03/23 17:50:26 vehicle status: unexpected status: 403 (Forbidden)
[tesla ] TRACE 2024/03/23 17:50:26 GET https://fleet-api.prd.eu.vn.cloud.tesla.com/api/1/vehicles/***/vehicle_data
[tesla ] TRACE 2024/03/23 17:50:26 {"response":****}}}
[main  ] DEBUG 2024/03/23 17:50:26 vehicle status: B (Ynot)
[lp-1  ] INFO 2024/03/23 17:50:26 vehicle updated: unknown -> Ynot
[lp-1  ] DEBUG 2024/03/23 17:50:26 vehicle soc: 75%
[lp-1  ] DEBUG 2024/03/23 17:50:26 vehicle soc limit: 80%
[lp-1  ] DEBUG 2024/03/23 17:50:26 vehicle range: 355km
[lp-1  ] DEBUG 2024/03/23 17:50:27 pv charge current: 0A = 0A + -0.257A (178W @ 3p)
[lp-1  ] DEBUG 2024/03/23 17:50:27 pv enable timer reset
[lp-1  ] DEBUG 2024/03/23 17:50:27 pv timer inactive
[site  ] DEBUG 2024/03/23 17:50:52 ----
[lp-1  ] DEBUG 2024/03/23 17:50:52 charge power: 0W
[site  ] DEBUG 2024/03/23 17:50:52 pv power: 103W
[site  ] DEBUG 2024/03/23 17:50:52 grid meter: 170W
[site  ] DEBUG 2024/03/23 17:50:52 grid powers: [38 -27 160]W
[site  ] DEBUG 2024/03/23 17:50:53 grid currents: [0.56 -0.52 1.28]A
[site  ] DEBUG 2024/03/23 17:50:53 site power: 170W
[lp-1  ] DEBUG 2024/03/23 17:50:53 vehicle odometer: 35430km
[lp-1  ] DEBUG 2024/03/23 17:50:53 charge voltages: [0 1.7 0]V
[lp-1  ] DEBUG 2024/03/23 17:50:53 charge currents: [0 0.3 0.4]A
[lp-1  ] DEBUG 2024/03/23 17:50:53 charger status: B
[lp-1  ] DEBUG 2024/03/23 17:50:53 pv charge current: 0A = 0A + -0.247A (170W @ 3p)
[site  ] DEBUG 2024/03/23 17:51:22 ----
[lp-1  ] DEBUG 2024/03/23 17:51:22 charge power: 0W
[site  ] DEBUG 2024/03/23 17:51:22 pv power: 104W
[site  ] DEBUG 2024/03/23 17:51:22 grid meter: 165W
[... 189 lines cut ...]
[lp-1  ] DEBUG 2024/03/23 17:56:23 charger status: C
[lp-1  ] DEBUG 2024/03/23 17:56:23 vehicle soc: 75%
[lp-1  ] DEBUG 2024/03/23 17:56:23 vehicle soc limit: 80%
[lp-1  ] DEBUG 2024/03/23 17:56:23 vehicle range: 355km
[lp-1  ] DEBUG 2024/03/23 17:56:23 pv charge current: 0A = 1A + -5.43A (3748W @ 3p)
[lp-1  ] DEBUG 2024/03/23 17:56:48 set charge mode: off
[site  ] DEBUG 2024/03/23 17:56:48 ----
[lp-1  ] DEBUG 2024/03/23 17:56:50 charge power: 3570W
[site  ] DEBUG 2024/03/23 17:56:50 pv power: 110W
[site  ] DEBUG 2024/03/23 17:56:50 grid meter: 3691W
[site  ] DEBUG 2024/03/23 17:56:50 grid powers: [1171 1171 1349]W
[site  ] DEBUG 2024/03/23 17:56:50 grid currents: [5 4.97 5.82]A
[site  ] DEBUG 2024/03/23 17:56:50 site power: 3691W
[... cut ...]
[lp-1  ] DEBUG 2024/03/23 17:56:50 charger status: C
[lp-1  ] DEBUG 2024/03/23 17:56:50 vehicle soc: 75%
[lp-1  ] DEBUG 2024/03/23 17:56:50 vehicle soc limit: 80%
[lp-1  ] DEBUG 2024/03/23 17:56:50 vehicle range: 355km
[tesla ] TRACE 2024/03/23 17:56:50 POST https://tesla.evcc.io/api/1/vehicles/***/command/charge_stop
[tesla ] TRACE 2024/03/23 17:56:52 {"response":{"result":true,"reason":""}}
[lp-1  ] DEBUG 2024/03/23 17:56:52 charger disable
[lp-1  ] DEBUG 2024/03/23 17:56:52 wake-up timer: stop

andig, 2024-03-24:
Thanks. Added the function but never called ;). New `:tesla` image online.

FraBoCH, 2024-03-24:
We are almost there ! 
The behavior is now correct (charging amp go down correctly without stopping at 5A), but there are constant "charger logic error: current mismatch", even when the current is actually correct. 
By reading more closely the trace, I found out that you are not reading the correct property. You should use "charge_amps" and not "charge_rate". There is also a "charger_actual_current" property which could be interesting, but I observe that it shows different values than "charge_amps" during the transition between amps.

Here is the corresponding logs extract:

`````
[tesla ] TRACE 2024/03/24 10:45:08 GET https://fleet-api.prd.eu.vn.cloud.tesla.com/api/1/vehicles/***/vehicle_data
[tesla ] TRACE 2024/03/24 10:45:09 {"response":{"id":***,"user_id":***,"vehicle_id":***,"vin":"***","color":null,"access_type":"OWNER","granular_access":{"hide_private":false,"training_wheels":false},"tokens":[***],"state":"online","in_service":false,"id_s":"***","calendar_enabled":true,"api_version":73,"backseat_token":null,"backseat_token_updated_at":null,"ble_autopair_enrolled":false,"charge_state":{"battery_heater_on":false,"battery_level":78,"battery_range":229.28,"charge_amps":3,"charge_current_request":3,"charge_current_request_max":16,"charge_enable_request":true,"charge_energy_added":1.97,"charge_limit_soc":90,"charge_limit_soc_max":100,"charge_limit_soc_min":50,"charge_limit_soc_std":80,"charge_miles_added_ideal":8.0,"charge_miles_added_rated":8.0,"charge_port_cold_weather_mode":false,"charge_port_color":"\u003cinvalid\u003e","charge_port_door_open":true,"charge_port_latch":"Engaged","charge_rate":7.6,"charger_actual_current":4,"charger_phases":2,"charger_pilot_current":16,"charger_power":2,"charger_voltage":237,"charging_state":"Charging","conn_charge_cable":"IEC","est_battery_range":261.08,"fast_charger_brand":"","fast_charger_present":false,"fast_charger_type":"ACSingleWireCAN","ideal_battery_range":229.28,"max_range_charge_counter":0,"minutes_to_full_charge":490,"not_enough_power_to_heat":null,"off_peak_charging_enabled":false," ......}}}
[lp-1  ] WARN 2024/03/24 10:45:09 charger logic error: current mismatch (got 7.6A, expected 3A)
[lp-1  ] DEBUG 2024/03/24 10:45:09 pv charge current: 3.42A = 5.1A + -1.68A (1162W @ 3p)
[tesla ] TRACE 2024/03/24 10:45:09 POST https://tesla.evcc.io/api/1/vehicles/***/command/set_charging_amps
[tesla ] TRACE 2024/03/24 10:45:09 {"charging_amps": 3}
--
{"response":{"result":true,"reason":""}}
[lp-1  ] DEBUG 2024/03/24 10:45:09 max charge current: 3A
`````

FraBoCH, 2024-03-24:
Test is successful with latest :tesla image. Thanks a lot @andig !
Here is the log proving that the current mismatch logic is correctly detecting and fixing the tesla bug when going down from "now" to "pv" mode with a min current set at 1A.

Looking forward to see this fix in the next release

````
[lp-1  ] DEBUG 2024/03/24 12:37:59 set charge mode: now
[site  ] DEBUG 2024/03/24 12:37:59 ----
[lp-1  ] DEBUG 2024/03/24 12:37:59 charge power: 790W
[site  ] DEBUG 2024/03/24 12:37:59 pv power: 1673W
[site  ] DEBUG 2024/03/24 12:37:59 grid meter: -264W
[site  ] DEBUG 2024/03/24 12:37:59 grid powers: [190 -320 -134]W
[site  ] DEBUG 2024/03/24 12:37:59 grid currents: [1.43 -1.48 -1.17]A
[site  ] DEBUG 2024/03/24 12:38:00 site power: -264W
[lp-1  ] DEBUG 2024/03/24 12:38:00 charge voltages: [239 239 240]V
[lp-1  ] DEBUG 2024/03/24 12:38:00 detected connected phases: 3p
[lp-1  ] DEBUG 2024/03/24 12:38:00 charge currents: [1.1 1.1 1.1]A
[lp-1  ] DEBUG 2024/03/24 12:38:00 detected active phases: 3p
[lp-1  ] DEBUG 2024/03/24 12:38:00 charger status: C
[lp-1  ] DEBUG 2024/03/24 12:38:00 vehicle soc: 81%
[lp-1  ] DEBUG 2024/03/24 12:38:00 vehicle soc limit: 90%
[lp-1  ] DEBUG 2024/03/24 12:38:00 vehicle range: 383km
[tesla ] TRACE 2024/03/24 12:38:00 POST https://tesla.evcc.io/api/1/vehicles/******/command/set_charging_amps
[tesla ] TRACE 2024/03/24 12:38:00 {"charging_amps": 8}
--
{"response":{"result":true,"reason":""}}
[lp-1  ] DEBUG 2024/03/24 12:38:00 max charge current: 8A
[site  ] DEBUG 2024/03/24 12:38:18 ----
[lp-1  ] DEBUG 2024/03/24 12:38:18 charge power: 5768W
[site  ] DEBUG 2024/03/24 12:38:18 pv power: 1719W
[site  ] DEBUG 2024/03/24 12:38:18 grid meter: 4387W
[site  ] DEBUG 2024/03/24 12:38:18 grid powers: [1441 1389 1558]W
[site  ] DEBUG 2024/03/24 12:38:18 grid currents: [6.66 5.83 6.59]A
[site  ] DEBUG 2024/03/24 12:38:18 site power: 4387W
[lp-1  ] DEBUG 2024/03/24 12:38:18 charge voltages: [237 238 237]V
[lp-1  ] DEBUG 2024/03/24 12:38:18 detected connected phases: 3p
[lp-1  ] DEBUG 2024/03/24 12:38:18 charge currents: [8.1 8.1 8.1]A
[lp-1  ] DEBUG 2024/03/24 12:38:18 detected active phases: 3p
[lp-1  ] DEBUG 2024/03/24 12:38:18 charger status: C
[lp-1  ] DEBUG 2024/03/24 12:38:18 vehicle soc: 81%
[lp-1  ] DEBUG 2024/03/24 12:38:18 vehicle soc limit: 90%
[lp-1  ] DEBUG 2024/03/24 12:38:18 vehicle range: 383km
[site  ] DEBUG 2024/03/24 12:38:48 ----
[lp-1  ] DEBUG 2024/03/24 12:38:48 charge power: 5790W
[site  ] DEBUG 2024/03/24 12:38:48 pv power: 1867W
[site  ] DEBUG 2024/03/24 12:38:48 grid meter: 4690W
[site  ] DEBUG 2024/03/24 12:38:48 grid powers: [1847 1340 1503]W
[site  ] DEBUG 2024/03/24 12:38:48 grid currents: [7.77 5.61 6.37]A
[site  ] DEBUG 2024/03/24 12:38:48 site power: 4690W
[lp-1  ] DEBUG 2024/03/24 12:38:48 charge voltages: [238 238 238]V
[lp-1  ] DEBUG 2024/03/24 12:38:48 detected connected phases: 3p
[lp-1  ] DEBUG 2024/03/24 12:38:48 charge currents: [8.1 8.1 8.1]A
[lp-1  ] DEBUG 2024/03/24 12:38:48 detected active phases: 3p
[lp-1  ] DEBUG 2024/03/24 12:38:48 charger status: C
[lp-1  ] DEBUG 2024/03/24 12:38:48 vehicle soc: 81%
[lp-1  ] DEBUG 2024/03/24 12:38:48 vehicle soc limit: 90%
[lp-1  ] DEBUG 2024/03/24 12:38:48 vehicle range: 383km
[lp-1  ] DEBUG 2024/03/24 12:38:58 set charge mode: pv
[site  ] DEBUG 2024/03/24 12:38:58 ----
[lp-1  ] DEBUG 2024/03/24 12:38:58 charge power: 5809W
[site  ] DEBUG 2024/03/24 12:38:58 pv power: 2014W
[site  ] DEBUG 2024/03/24 12:38:58 grid meter: 4578W
[site  ] DEBUG 2024/03/24 12:38:58 grid powers: [1805 1296 1476]W
[site  ] DEBUG 2024/03/24 12:38:58 grid currents: [7.54 5.43 6.21]A
[site  ] DEBUG 2024/03/24 12:38:58 site power: 4578W
[lp-1  ] DEBUG 2024/03/24 12:38:58 charge voltages: [239 238 239]V
[lp-1  ] DEBUG 2024/03/24 12:38:58 detected connected phases: 3p
[lp-1  ] DEBUG 2024/03/24 12:38:58 charge currents: [8.1 8.1 8.1]A
[lp-1  ] DEBUG 2024/03/24 12:38:58 detected active phases: 3p
[lp-1  ] DEBUG 2024/03/24 12:38:58 charger status: C
[lp-1  ] DEBUG 2024/03/24 12:38:58 vehicle soc: 81%
[lp-1  ] DEBUG 2024/03/24 12:38:58 vehicle soc limit: 90%
[lp-1  ] DEBUG 2024/03/24 12:38:58 vehicle range: 383km
[lp-1  ] DEBUG 2024/03/24 12:38:58 pv charge current: 1.37A = 8A + -6.63A (4578W @ 3p)
[tesla ] TRACE 2024/03/24 12:38:58 POST https://tesla.evcc.io/api/1/vehicles/******/command/set_charging_amps
[tesla ] TRACE 2024/03/24 12:38:59 {"charging_amps": 1}
--
{"response":{"result":true,"reason":""}}
[lp-1  ] DEBUG 2024/03/24 12:38:59 max charge current: 1A
[site  ] DEBUG 2024/03/24 12:39:18 ----
[lp-1  ] DEBUG 2024/03/24 12:39:18 charge power: 3683W
[site  ] DEBUG 2024/03/24 12:39:18 pv power: 3521W
[site  ] DEBUG 2024/03/24 12:39:18 grid meter: 324W
[site  ] DEBUG 2024/03/24 12:39:18 grid powers: [52 48 223]W
[site  ] DEBUG 2024/03/24 12:39:18 grid currents: [0.89 0.73 1.49]A
[site  ] DEBUG 2024/03/24 12:39:18 site power: 324W
[lp-1  ] DEBUG 2024/03/24 12:39:18 charge voltages: [241 241 240]V
[lp-1  ] DEBUG 2024/03/24 12:39:18 detected connected phases: 3p
[lp-1  ] DEBUG 2024/03/24 12:39:18 charge currents: [5.1 5.1 5.1]A
[lp-1  ] DEBUG 2024/03/24 12:39:18 detected active phases: 3p
[lp-1  ] DEBUG 2024/03/24 12:39:18 charger status: C
[lp-1  ] DEBUG 2024/03/24 12:39:18 vehicle soc: 81%
[lp-1  ] DEBUG 2024/03/24 12:39:18 vehicle soc limit: 90%
[lp-1  ] DEBUG 2024/03/24 12:39:18 vehicle range: 383km
[tesla ] TRACE 2024/03/24 12:39:18 GET https://fleet-api.prd.eu.vn.cloud.tesla.com/api/1/vehicles/******/vehicle_data
[tesla ] TRACE 2024/03/24 12:39:19 {"response":....
[lp-1  ] WARN 2024/03/24 12:39:19 charger logic error: current mismatch (got 5A, expected 1A)
[lp-1  ] DEBUG 2024/03/24 12:39:19 pv charge current: 4.53A = 5A + -0.469A (324W @ 3p)
[tesla ] TRACE 2024/03/24 12:39:19 POST https://tesla.evcc.io/api/1/vehicles/******/command/set_charging_amps
[tesla ] TRACE 2024/03/24 12:39:19 {"charging_amps": 4}
--
{"response":{"result":true,"reason":""}}
````

closed 2024-03-24 by andig

Neffe1337, 2024-08-02:
Hallo zusammen,

Ich habe leider den gleichen Fehler und finde das Problem leider nicht. 
Kann mor bitte einer seine Config senden mit der der Tesla unter 5A laden kann?

Neffe1337, 2024-08-03:
vehicles:
  - name: my_car
    type: template
    template: tesla
    title: # Wird in der Benutzeroberfläche angezeigt (optional)
    icon: car # Icon in der Benutzeroberfläche (optional)
    accessToken: ***
    refreshToken: ***
    capacity: 60 # Akkukapazität in kWh (optional)
    phases: 3 # Die maximale Anzahl der Phasen welche genutzt werden können (optional)
    commandProxy: https://tesla.evcc.io/ # Bei Verwendung eines TWC3 (oder eines anderen 'dummen' Ladegeräts, das nicht steuerbar ist) kann evcc die Ladung direkt verwalten, indem es über einen Command Proxy mit dem Fahrzeug kommuniziert. Standardmäßig wird der von evcc bereitgestellte Proxy verwendet. Dieses param>
    mode: # Möglich sind Off, Now, MinPV und PV, oder leer wenn keiner definiert werden soll (optional)
    control: true
    minCurrent: 1 # Definiert die minimale Stromstärke pro angeschlossener Phase mit welcher das Fahrzeug geladen werden soll (optional)
    maxCurrent: 16 # Definiert die maximale Stromstärke pro angeschlossener Phase mit welcher das Fahrzeug geladen werden soll (optional)
    identifiers: # Kann meist erst später eingetragen werden, siehe: https://docs.evcc.io/docs/features/vehicle (optional)
    priority: # Priorität des Ladepunktes oder Fahrzeugs in Relation zu anderen Ladepunkten oder Fahrzeugen für die Zuweisung von PV-Energie (optional)

andig, 2024-08-03:
@Neffe1337 mach bitte ein neues Thema. Dieses hier ist gelöst.