evcc-io/evcc #13017: set mqtt.0.client0.evcc/site/residualPower/set: strconv.ParseFloat: parsing \"null\": invalid syntax

rene-wieland, 2024-03-17:
### Describe the bug

I'am using MQTT within ioroker to supply all the meter data to evcc (pv, battery). evcc outputs warning messages for residualPower/set as this parameter is only for setting residualPower within MQTT adapter. This value shows always (null)

### Steps to reproduce

1. no special steps required
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070  
network:  
  schema: http  
  host: evcc.local # .local suffix announces the hostname on MDNS  
  port: 7070  
  
log: debug  
levels:  
  cache: error  
  
# unique installation id  
plant: *****  
  
interval: 30s # control cycle interval  
  
sponsortoken: *****  
  
# sponsors can set telemetry: true to enable anonymous data aggregation  
# see https://github.com/evcc-io/evcc/discussions/4554  
telemetry: false  
  
meters:  
- name: grid0  
  type: custom  
  power:  
    source: mqtt  
    topic: mqtt/0/meters/grid  
    #timeout: 30s  
    scale: -1  
    
- name: pv1  
  type: custom  
  power:  
    source: mqtt  
    topic: mqtt/0/meters/solaredge  
    #timeout: 30s  
    scale: 1  
    
- name: pv2  
  type: custom  
  power:  
    source: mqtt  
    topic: mqtt/0/meters/mic2500  
    #timeout: 30s  
    scale: 1  
    
- name: pv3  
  type: custom  
  power:  
    source: mqtt  
    topic: mqtt/0/meters/mod4000  
    #timeout: 30s  
    scale: 1  
    
- name: battery4  
  type: custom  
  power:  
    source: mqtt  
    topic: mqtt/0/meters/battery  
    #timeout: 30s  
    scale: -1  
  soc:  
    source: mqtt  
    topic: mqtt/0/meters/batterySoc  
  capacity: 5.0  
      
    
chargers:  
- type: template  
  template: easee   
  user: *****  
  password: *****  
  charger: EHEXAFFW    
  timeout: 30s    
  name: wallbox5  
  authorize: true  
  
vehicles:  
- type: template  
  template: tesla-command  
  title: Jarvis    
  icon: car    
  accessToken: *****  
  refreshToken: *****  
  vin: *****  
  capacity: 78    
  phases: 3    
  mode: pv    
  minCurrent: 5    
  maxCurrent: 16    
  priority: 1    
  name: ev5  
  
loadpoints:  
- title: Easee Jarvis  
  charger: wallbox5  
  vehicle: ev5  
  mode: pv  
  guardduration: 3m  
  
site:  
  title: RWzuHause  
  meters:  
    grid: grid0  
    pv:  
    - pv1  
    - pv2  
    - pv3  
    battery:  
    - battery4  
  residualPower: 100  
  maxGridSupplyWhileBatteryCharging: 0  
    
mqtt:  
  broker: 192.168.144.100:1883  
  topic: mqtt.0.client0.evcc  
  clientid: evcc  
  
tariffs:  
  currency: EUR  
  grid:  
    type: fixed  
    price: 0.3287  
  feedin:  
    type: fixed  
    price: 0.10  
  # co2:  
    # type: grünstromindex  
    # zip: 53773
```

### Log details

```text
main  ] INFO 2024/03/17 22:54:21 evcc 0.124.10
[main  ] INFO 2024/03/17 22:54:21 using config file: /etc/evcc.yaml
[main  ] INFO 2024/03/17 22:54:21 starting ui and api at :7070
[db    ] INFO 2024/03/17 22:54:22 using sqlite database: /root/.evcc/evcc.db
[mqtt  ] INFO 2024/03/17 22:54:22 connecting evcc at tcp://192.168.144.100:1883
[mqtt  ] DEBUG 2024/03/17 22:54:22 tcp://192.168.144.100:1883 connected
[lp-1  ] DEBUG 2024/03/17 22:54:23 charge total import: 2430.870kWh
[mqtt  ] ERROR 2024/03/17 22:54:24 set mqtt.0.client0.evcc/site/residualPower/set: strconv.ParseFloat: parsing "null": invalid syntax
[mqtt  ] ERROR 2024/03/17 22:54:24 set mqtt.0.client0.evcc/site/residualPower/set: invalid float value: NaN
[mqtt  ] ERROR 2024/03/17 22:54:24 set mqtt.0.client0.evcc/site/residualPower/set: strconv.ParseFloat: parsing "null": invalid syntax
[mqtt  ] ERROR 2024/03/17 22:54:24 set mqtt.0.client0.evcc/site/residualPower/set: strconv.ParseFloat: parsing "null": invalid syntax
[mqtt  ] ERROR 2024/03/17 22:54:25 set mqtt.0.client0.evcc/site/residualPower/set: strconv.ParseFloat: parsing "null": invalid syntax
[mqtt  ] ERROR 2024/03/17 22:54:25 set mqtt.0.client0.evcc/site/residualPower/set: strconv.ParseFloat: parsing "null": invalid syntax
[site  ] INFO 2024/03/17 22:54:25 site config:
[site  ] INFO 2024/03/17 22:54:25   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2024/03/17 22:54:25     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/03/17 22:54:25     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/03/17 22:54:25     pv 2:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/03/17 22:54:25     pv 3:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/03/17 22:54:25     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓
[site  ] INFO 2024/03/17 22:54:25   vehicles:
[site  ] INFO 2024/03/17 22:54:25     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[lp-1  ] INFO 2024/03/17 22:54:25 loadpoint 1:
[lp-1  ] INFO 2024/03/17 22:54:25   mode:        pv
[lp-1  ] INFO 2024/03/17 22:54:25   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2024/03/17 22:54:25   meters:      charge ✓
[lp-1  ] INFO 2024/03/17 22:54:25     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2024/03/17 22:54:25 phase timer inactive
[lp-1  ] DEBUG 2024/03/17 22:54:25 pv timer inactive
[lp-1  ] INFO 2024/03/17 22:54:25 vehicle updated: unknown -> Jarvis
[lp-1  ] DEBUG 2024/03/17 22:54:25 set charge mode: pv
[site  ] DEBUG 2024/03/17 22:54:25 ----
[lp-1  ] DEBUG 2024/03/17 22:54:25 charge power: 0W
[site  ] DEBUG 2024/03/17 22:54:25 pv power: 0W
[site  ] DEBUG 2024/03/17 22:54:25 battery soc: 3%
[site  ] DEBUG 2024/03/17 22:54:25 battery power: 0W
[site  ] DEBUG 2024/03/17 22:54:25 grid meter: 556W
[site  ] DEBUG 2024/03/17 22:54:25 site power: 606W
[mqtt  ] ERROR 2024/03/17 22:54:25 set mqtt.0.client0.evcc/site/residualPower/set: strconv.ParseFloat: parsing "null": invalid syntax
[lp-1  ] ERROR 2024/03/17 22:54:25 vehicle odometer: asleep
[lp-1  ] DEBUG 2024/03/17 22:54:25 charge currents: [0.008 0.008 0.011]A
[lp-1  ] DEBUG 2024/03/17 22:54:25 charge total import: 2430.870kWh
[lp-1  ] DEBUG 2024/03/17 22:54:25 charger status: B
[lp-1  ] INFO 2024/03/17 22:54:25 car connected
[lp-1  ] DEBUG 2024/03/17 22:54:25 pv timer elapse
[lp-1  ] DEBUG 2024/03/17 22:54:25 pv timer inactive
[lp-1  ] DEBUG 2024/03/17 22:54:25 charge total import: 2430.870kWh
[lp-1  ] ERROR 2024/03/17 22:54:25 vehicle soc: asleep
[lp-1  ] DEBUG 2024/03/17 22:54:25 available power -606W < 4140W min 3p threshold
[lp-1  ] DEBUG 2024/03/17 22:54:25 phase scale1p in 0s
[mqtt  ] ERROR 2024/03/17 22:54:25 set mqtt.0.client0.evcc/site/residualPower/set: strconv.ParseFloat: parsing "null": invalid syntax
[lp-1  ] DEBUG 2024/03/17 22:54:25 switched phases: 1p
[lp-1  ] DEBUG 2024/03/17 22:54:25 phase timer inactive
[lp-1  ] DEBUG 2024/03/17 22:54:25 pv charge current: 0A = 0A + -2.63A (606W @ 1p)
[lp-1  ] DEBUG 2024/03/17 22:54:25 pv enable timer reset
[lp-1  ] DEBUG 2024/03/17 22:54:25 pv timer inactive
[mqtt  ] ERROR 2024/03/17 22:54:25 set mqtt.0.client0.evcc/site/residualPower/set: strconv.ParseFloat: parsing "null": invalid syntax
[mqtt  ] ERROR 2024/03/17 22:54:25 set mqtt.0.client0.evcc/site/residualPower/set: strconv.ParseFloat: parsing "null": invalid syntax
[mqtt  ] ERROR 2024/03/17 22:54:25 set mqtt.0.client0.evcc/site/residualPower/set: strconv.ParseFloat: parsing "null": invalid syntax
[mqtt  ] ERROR 2024/03/17 22:54:25 set mqtt.0.client0.evcc/site/residualPower/set: strconv.ParseFloat: parsing "null": invalid syntax
[mqtt  ] ERROR 2024/03/17 22:54:25 set mqtt.0.client0.evcc/site/residualPower/set: strconv.ParseFloat: parsing "null": invalid syntax
[mqtt  ] ERROR 2024/03/17 22:54:25 set mqtt.0.client0.evcc/site/residualPower/set: strconv.ParseFloat: parsing "null": invalid syntax
[mqtt  ] ERROR 2024/03/17 22:54:25 set mqtt.0.client0.evcc/site/residualPower/set: strconv.ParseFloat: parsing "null": invalid syntax
[mqtt  ] ERROR 2024/03/17 22:54:25 set mqtt.0.client0.evcc/site/residualPower/set: strconv.ParseFloat: parsing "null": invalid syntax
[mqtt  ] ERROR 2024/03/17 22:54:25 set mqtt.0.client0.evcc/site/residualPower/set: strconv.ParseFloat: parsing "null": invalid syntax
[mqtt  ] ERROR 2024/03/17 22:54:25 set mqtt.0.client0.evcc/site/residualPower/set: strconv.ParseFloat: parsing "null": invalid syntax
[mqtt  ] ERROR 2024/03/17 22:54:25 set mqtt.0.client0.evcc/site/residualPower/set: strconv.ParseFloat: parsing "null": invalid syntax
[mqtt  ] ERROR 2024/03/17 22:54:26 set mqtt.0.client0.evcc/site/residualPower/set: strconv.ParseFloat: parsing "null": invalid syntax
[mqtt  ] ERROR 2024/03/17 22:54:26 set mqtt.0.client0.evcc/site/residualPower/set: strconv.ParseFloat: parsing "null": invalid syntax
[mqtt  ] ERROR 2024/03/17 22:54:26 set mqtt.0.client0.evcc/site/residualPower/set: strconv.ParseFloat: parsing "null": invalid syntax
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.124.10

mdkeil, 2024-03-18:
Did you set `residualPower` within iOBroker via MQTT?

andig, 2024-03-18:
> strconv.ParseFloat: parsing "null": invalid syntax

Woher kommt die "null"?

> set mqtt.0.client0.evcc/site/residualPower/set: invalid float value: NaN

Woher kommt das NaN? Du musst bitte auch passende Werte da rein schreiben!