evcc-io/evcc #13028: MQTT message order

MarkusGH, 2024-03-18:
### Describe the bug

It looks like mqtt message are not always processed in the order they are sent.

I send:

```
site/bufferStartSoc
(1 sec pause)
site/bufferSoc
site/bufferStartSoc

```
mosquitto_sub sees:

```
evcc/site/bufferStartSoc/set 99
evcc/site/bufferStartSoc/set (null)
evcc/site/bufferSoc/set 90
evcc/site/bufferStartSoc/set 95
evcc/site/bufferStartSoc/set (null)
evcc/site/bufferSoc/set (null)
```

evcc log shows:

```
[mqtt  ] TRACE 2024/03/18 16:37:51 recv evcc/site/bufferStartSoc/set: '99'
[mqtt  ] TRACE 2024/03/18 16:37:51 recv evcc/site/bufferStartSoc/set: ''
[mqtt  ] TRACE 2024/03/18 16:37:52 recv evcc/site/bufferStartSoc/set: '95'
[mqtt  ] TRACE 2024/03/18 16:37:52 recv evcc/site/bufferSoc/set: '90'
[mqtt  ] TRACE 2024/03/18 16:37:52 recv evcc/site/bufferStartSoc/set: ''
[mqtt  ] TRACE 2024/03/18 16:37:52 recv evcc/site/bufferSoc/set: ''
```

It looks like bufferStartSoc is processed before bufferSoc gets processed, hence leading to error messages: `set evcc/site/bufferStartSoc/set: buffer start soc must be larger than buffer soc` and bufferStartSoc not being set properly.

### Steps to reproduce

See above.

### Configuration details

```yaml
Not relevant.
```

### Log details

```text
[mqtt  ] TRACE 2024/03/18 16:37:51 recv evcc/site/bufferStartSoc/set: '99'
[mqtt  ] TRACE 2024/03/18 16:37:51 recv evcc/site/bufferStartSoc/set: ''
[mqtt  ] TRACE 2024/03/18 16:37:52 recv evcc/site/bufferStartSoc/set: '95'
[mqtt  ] TRACE 2024/03/18 16:37:52 recv evcc/site/bufferSoc/set: '90'
[mqtt  ] TRACE 2024/03/18 16:37:52 recv evcc/site/bufferStartSoc/set: ''
[mqtt  ] TRACE 2024/03/18 16:37:52 recv evcc/site/bufferSoc/set: ''
[mqtt  ] TRACE 2024/03/18 16:48:01 recv evcc/site/bufferStartSoc/set: '99'
[mqtt  ] TRACE 2024/03/18 16:48:01 recv evcc/site/bufferStartSoc/set: ''
[mqtt  ] TRACE 2024/03/18 16:48:02 recv evcc/site/bufferStartSoc/set: '60'
[mqtt  ] ERROR 2024/03/18 16:48:02 set evcc/site/bufferStartSoc/set: buffer start soc must be larger than buffer soc
[mqtt  ] TRACE 2024/03/18 16:48:02 recv evcc/site/bufferSoc/set: '50'
[mqtt  ] TRACE 2024/03/18 16:48:02 recv evcc/site/bufferStartSoc/set: ''
[mqtt  ] TRACE 2024/03/18 16:48:02 recv evcc/site/bufferSoc/set: ''
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.124.10 (d3ae3350)

andig, 2024-03-18:
Looks like this was intentionally disabled:

    options.SetOrderMatters(false)

MarkusGH, 2024-03-18:
That may be the reason, but:

```
[mqtt  ] TRACE 2024/03/18 16:37:52 recv evcc/site/bufferStartSoc/set: '95'
[mqtt  ] TRACE 2024/03/18 16:37:52 recv evcc/site/bufferSoc/set: '90'
[mqtt  ] TRACE 2024/03/18 16:37:52 recv evcc/site/bufferStartSoc/set: ''
[mqtt  ] TRACE 2024/03/18 16:37:52 recv evcc/site/bufferSoc/set: ''
```

evcc also sends it's empty messages out of order.

andig, 2024-03-18:
Don't let you be fooled by the logging. The Go routines are not deterministic.

MarkusGH, 2024-03-18:
> Don't let you be fooled by the logging. The Go routines are not deterministic. Long story short: don't rely on mqtt message order.

OK - but in the end it is evcc that relies on order when bufferSoc and bufferStartSoc are set.

andig, 2024-03-18:
Eigentlich nicht- es schaut ja ohnehin nur alle Interval Sekunden drauf.

closed 2024-03-18 by andig

MarkusGH, 2024-03-19:
> Eigentlich nicht- es schaut ja ohnehin nur alle Interval Sekunden drauf.

Das stimmt leider nicht.

In der Situation schlägt das Setzen von bufferStartSoc fehl, weil evcc die Nachrichten reproduzierbar in umgekehrter Reihenfolge verarbeitet. Normalerweise kann man sich bei MQTT darauf verlassen, dass Nachrichten selber QOS "in order" verarbeitet werden.

Ist es denn wirklich notwendig, options.SetOrderMatters(false) zu setzen oder kommt diese Einstellung aus einer theoretischen Überlegung?

andig, 2024-03-19:
Keine Ahnung. Wir können das gerne wieder raus werfen. Kann gut sein, dass das eine falsche Fehlerbehebung war.

MarkusGH, 2024-03-25:
Das Problem scheint mit https://github.com/evcc-io/evcc/pull/13037 verschwunden zu sein. Danke!