evcc-io/evcc #13033: Shelly 1 mini gen3 kann nicht eingebunden werden

Mathias83, 2024-03-18:
### Describe the bug

Ich würde gerne meinen Shelly als Schalter für die WP nehmen.
Beim einbinden nach der Vorlage kommt leider folgender Fehler

`cannot create charger 'my_charger'
cannot create charger type 'template'
cannot create charger type 'shelly'
(S3SW-001X8EU) unknown api generation (3)`

### Steps to reproduce

1. Konfiguration
2. neustart des Dienstes

### Configuration details

```yaml
chargers:
  - name: wallbox
    type: template
    template: go-e-v3
    host: 192.168.0.212
  - name: my_charger
    type: template
    template: shelly
    host: 192.168.0.214
```

### Log details

```text
Mar 18 20:52:38 evcc evcc[23663]: [main  ] FATAL 2024/03/18 20:52:38 cannot create charger 'my_charger': cannot create charger type 'template': cannot create charger type 'shelly':  (S3SW-001X8EU) unknown api generation (3)
Mar 18 20:52:38 evcc evcc[23663]: [main  ] FATAL 2024/03/18 20:52:38 will attempt restart in: 5m0s
```

### What type of operating system are you running?

Linux

### Version

0.124.10

Mathias83, 2024-03-18:
Ich vermute mal es liegt an der Cloud API https://shelly-api-docs.shelly.cloud/cloud-control-api/

andig, 2024-03-19:
/cc @thierolm

thierolm, 2024-03-19:
> Ich vermute mal es liegt an der Cloud API https://shelly-api-docs.shelly.cloud/cloud-control-api/

Das ist wie die Fehler-Meldung aussagt, eine neue Shelly Variante, die evcc (noch) nicht unterstützt.

Kannst du bitte mal ein Trace-Log deines Shelly Mini gen3 chargers schicken?

`evcc -l trace charger`

Mathias83, 2024-03-19:
`[main  ] INFO 2024/03/19 16:28:52 evcc 0.124.10
[main  ] INFO 2024/03/19 16:28:52 using config file: /etc/evcc.yaml
[db    ] INFO 2024/03/19 16:28:52 using sqlite database: /root/.evcc/evcc.db
[db    ] TRACE 2024/03/19 16:28:52 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2024/03/19 16:28:52 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/03/19 16:28:52 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/03/19 16:28:52 SELECT * FROM `settings` 6 <nil>
[db    ] TRACE 2024/03/19 16:28:52 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2024/03/19 16:28:52 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2024/03/19 16:28:52 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2024/03/19 16:28:52 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/03/19 16:28:52 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/03/19 16:28:52 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2024/03/19 16:28:52 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
[db    ] TRACE 2024/03/19 16:28:52 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/03/19 16:28:52 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_configs_details"" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" OR sql LIKE "%CONSTRAINT 	fk_configs_details	%") -1 <nil>
[db    ] TRACE 2024/03/19 16:28:52 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
[db    ] TRACE 2024/03/19 16:28:52 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_devices_details"" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT 	fk_devices_details	%") -1 <nil>
[db    ] TRACE 2024/03/19 16:28:52 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%""device_id"" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "%	device_id	%") -1 <nil>
[db    ] TRACE 2024/03/19 16:28:52 SELECT * FROM `configs` WHERE `configs`.`class` = 1 ORDER BY id 0 <nil>
[go-e  ] TRACE 2024/03/19 16:28:52 GET http://192.168.0.212/api/status?filter=alw
[shelly] TRACE 2024/03/19 16:28:52 GET http://192.168.0.214/shelly
[shelly] TRACE 2024/03/19 16:28:52 {"name":null,"id":"shelly1minig3-5432044fc790","mac":"5432044FC790","slot":0,"model":"S3SW-001X8EU","gen":3,"fw_id":"20240223-141905/1.2.2-g7c39781","ver":"1.2.2","app":"Mini1G3","auth_en":false,"auth_domain":null}
[go-e  ] TRACE 2024/03/19 16:28:52 {"alw":false}
[main  ] FATAL 2024/03/19 16:28:52 cannot create charger 'my_charger': cannot create charger type 'template': cannot create charger type 'shelly':  (S3SW-001X8EU) unknown api generation (3)`

Webhook aufrufe gehen noch tadellos 
http://192.168.0.214/relay/0?turn=off
http://192.168.0.214/relay/0?turn=on

thierolm, 2024-03-19:
> Webhook aufrufe gehen noch tadellos

Warum NOCH? Hast du den Mini 1P Gen 3 schon mal erfolgreich eingebunden gehabt?

Mathias83, 2024-03-19:
> > Webhook aufrufe gehen noch tadellos
> 
> Warum NOCH? Hast du den Mini 1P Gen 3 schon mal erfolgreich eingebunden gehabt?

nein hatte aber sorge das man für alles die cloud brauch.

thierolm, 2024-03-20:
Nein, brauchst du nicht. Ich versuche mal bis heute Abend eine PR bereitzustellen.

Mathias83, 2024-03-20:
Das wäre ja super. Falls ich was testen kann sagt gerne Bescheid.

thierolm, 2024-03-20:
Wenn du Nightlys installieren kannst, kannst du die Erweiterung testen, sobald die Änderung gemerged ist.

thierolm, 2024-03-20:
Schau mal unter #13057. Wenn @andig den PR gemerged hat, kannst du die Nightly Version testen.

closed 2024-03-20 by andig

Mathias83, 2024-03-23:
Also Schalten scheint zu gehen lustigerweise schaltet er aber wenn ich von "pv" auf "aus" gehe direkt wieder zurück auf "pv". Der Shelly reagiert aber und ist kurzzeitig deaktiviert

chuck2941, 2024-03-29:
Ich habe heute meinen Shelly plus 1PM mini gegen einen PM mini an dem BKW getauscht und dann gemerkt das EVCC ihn nicht erkennt. Habe dann wieder zurück getauscht.
Ich bekomme die gleichen Fehler.
Gibt es dazu schon was neues oder weiß jemand wann die Erweiterung kommt das man den PM mini Gen 3 einbinden kann?

chuck2941, 2024-03-29:
Funktioniert das Nightly oder gibt's Unterschiede zur normalen Version?

rennsven, 2025-04-25:
In der Version 0.203.1 funktioniert der Shelly PM Mini Gen 3 als Messung für das BKW noch tadellos. Ich bleibe bei der Version bis eine unterstützende Version erscheint!

thierolm, 2025-04-25:
@rennsven bitte teste mal #20855 wenn du kannst.

PR #20855 (2025-04-25): Shelly: re-enable PM Mini Gen3
Fixes  #13033 comment of @rennsven