evcc-io/evcc #13041: Wallbox: charger: invalid status: 7

btrfb, 2024-03-19:
### Describe the bug

Hello Community,
I have evcc running on a Raspberry, in the latest version (0.124.10). I am also using a Webasto Next Wallbox.
We are currently charging a Skoda Octavia RS Hybrid as a guest vehicle.
Deliberately as a guest vehicle, as the car has poor LTE reception and is therefore not well recognized.
If I charge via PV surplus and the sun goes away for a short time and there is no surplus, evcc goes into the error from the headline: 
Wallbox: charger: invalid status: 7
The car does not start charging again either.
You first have to unplug it and plug it in again - in most cases it then starts charging again.
Do you have an approach for me?
Thank you very much in advance.
[image]
[image]

### Steps to reproduce

If I charge via PV surplus and the sun goes away for a short time and there is no surplus, evcc goes into the error from the headline: 
Wallbox: charger: invalid status: 7
The car does not start charging again either.
You first have to unplug it and plug it in again - in most cases it then starts charging again.

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: xxx

interval: 10s # control cycle interval

sponsortoken: xxx

######## Experiment MQTT TCP #########                                                                                   
meters:
  - name: my_grid
    type: template
    template: sofarsolar-g3
    usage: grid

    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.2.99 # Hostname/IP
    port: 8899 # Port
#    delay: 1s # optional

  - name: my_pv
    type: template
    template: sofarsolar-g3
    usage: pv
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.2.99 # Hostname/IP
    port: 8899 # Port
#    delay: 1s # optional

  - name: my_battery
    type: template
    template: sofarsolar-g3
    usage: battery

    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.2.99 # Hostname
    port: 8899 # Port
#    delay: 1s # optional
######## Experiment MQTT TCP END #########  

chargers:
- name: Wallbox
  type: template
  template: webasto-next
  host: 192.168.2.97
  port: 502

#vehicles:
#  - name: my_car
#    type: template
#    template: skoda
#    title: Skoda Octavia # Wird in der Benutzeroberfläche angezeigt (optional)
#    user: xxx@icloud.com # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.)
#    password: bxxx # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen)
#    vin: xxx # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind (optional)
#    capacity: 11 # Akkukapazität in kWh (optional) 

loadpoints:
- title: Wallbox
  charger: Wallbox
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16

site:
  title: Holunderweg 12
  meters:
    grid: my_grid
    pv: my_pv

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.326 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.08 # [currency]/kWh
```

### Log details

```text
pi@solaranzeige:~ $ evcc --log debug
[main  ] INFO 2024/03/19 14:30:53 evcc 0.124.10
[main  ] INFO 2024/03/19 14:30:53 using config file: /etc/evcc.yaml
[main  ] INFO 2024/03/19 14:30:53 starting ui and api at :7070
[db    ] INFO 2024/03/19 14:30:53 using sqlite database: /home/pi/.evcc/evcc.db
[lp-1  ] WARN 2024/03/19 14:30:53 deprecated: minCurrent setting is ignored, please remove
[lp-1  ] WARN 2024/03/19 14:30:53 deprecated: maxcurrent setting is ignored, please remove
[lp-1  ] WARN 2024/03/19 14:30:53 deprecated: phases setting is ignored, please remove
[lp-1  ] DEBUG 2024/03/19 14:30:53 charge total import: 273.305kWh
[site  ] WARN 2024/03/19 14:30:53 battery configured but residualPower is missing or <= 0 (add residualPower: 100 to site), see https://docs.evcc.io/en/docs/reference/configuration/site#residualpower
[site  ] INFO 2024/03/19 14:30:53 site config:
[site  ] INFO 2024/03/19 14:30:53   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2024/03/19 14:30:53     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2024/03/19 14:30:53     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2024/03/19 14:30:53     battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✗
[lp-1  ] INFO 2024/03/19 14:30:53 loadpoint 1:
[lp-1  ] INFO 2024/03/19 14:30:53   mode:        pv
[lp-1  ] INFO 2024/03/19 14:30:53   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2024/03/19 14:30:53   meters:      charge ✓
[lp-1  ] INFO 2024/03/19 14:30:53     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2024/03/19 14:30:53 phase timer inactive
[lp-1  ] DEBUG 2024/03/19 14:30:53 pv timer inactive
[lp-1  ] ERROR 2024/03/19 14:30:53 charger: invalid status: 7
[site  ] WARN 2024/03/19 14:30:53 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2024/03/19 14:30:53 ----
[lp-1  ] DEBUG 2024/03/19 14:30:53 charge power: 0W
[site  ] DEBUG 2024/03/19 14:30:54 pv power: 7100W
[site  ] DEBUG 2024/03/19 14:30:55 battery soc: 100%
[site  ] DEBUG 2024/03/19 14:30:55 battery power: 0W
[site  ] DEBUG 2024/03/19 14:30:56 grid meter: -6980W
[site  ] DEBUG 2024/03/19 14:30:57 grid powers: [-2340 -2350 -2330]W
[site  ] DEBUG 2024/03/19 14:30:58 grid currents: [-10.3 -10.2 -10.5]A
[site  ] DEBUG 2024/03/19 14:30:58 site power: -6980W
[lp-1  ] DEBUG 2024/03/19 14:30:58 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/03/19 14:30:58 charge total import: 273.305kWh
[lp-1  ] ERROR 2024/03/19 14:30:58 charger: invalid status: 7
[site  ] DEBUG 2024/03/19 14:31:03 ----
[lp-1  ] DEBUG 2024/03/19 14:31:03 charge power: 0W
[site  ] DEBUG 2024/03/19 14:31:10 pv power: 7500W
[site  ] DEBUG 2024/03/19 14:31:11 battery soc: 100%
[site  ] DEBUG 2024/03/19 14:31:11 battery power: 0W
[site  ] DEBUG 2024/03/19 14:31:12 grid meter: -7170W
[site  ] DEBUG 2024/03/19 14:31:13 grid powers: [-2390 -2390 -2390]W
[site  ] DEBUG 2024/03/19 14:31:14 grid currents: [-10.6 -10.5 -10.8]A
[site  ] DEBUG 2024/03/19 14:31:15 site power: -7170W
[lp-1  ] DEBUG 2024/03/19 14:31:15 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/03/19 14:31:15 charge total import: 273.305kWh
[lp-1  ] ERROR 2024/03/19 14:31:15 charger: invalid status: 7
[site  ] DEBUG 2024/03/19 14:31:15 ----
[lp-1  ] DEBUG 2024/03/19 14:31:15 charge power: 0W
[site  ] DEBUG 2024/03/19 14:31:16 pv power: 7500W
[site  ] DEBUG 2024/03/19 14:31:17 battery soc: 100%
[site  ] DEBUG 2024/03/19 14:31:17 battery power: 0W
[site  ] DEBUG 2024/03/19 14:31:17 grid meter: -7220W
```

### What type of operating system are you running?

Linux

premultiply, 2024-03-19:
Die Wallbox meldet damit einen internen Fehler:

```
State of the charging
device
0: No vehicle attached
(EVSE available, EVSE
unavailable)
1: Vehicle attached, no
permission (preparing)
3: charging
4: charging paused
(suspending EVSE,
suspending EV,
finishing)
7: charging error
8: charging station
reserved
```
Was sollten wir damit anderes tun als ihn ausgeben?

andig, 2024-03-20:
Ich würde vorschlagen wir geben in dem Fall die konkrete Fehlermeldung dafür aus, sowas wie

    charging error: status %d

Damit wäre zumindest klar, dass es nicht nur ein sonst ungemappter Status ist? Könnten wir nach und nach bei allen Boxen machen bei denen es tatsächlich zu solchen Fehlern kommt. Ich schaus mir an.

closed 2024-03-20 by andig

bo-cheng, 2025-03-04:
Bei uns taucht der Fehler leider auch auf. Es wird ein charging error status 7 angezeigt. Abhilfe schafft nur das Neustarten der Wallbox über die Benutzeroberfläche.

Der PV Modus ist aktiv und sobald eine Wolke über der PV-Anlage ist, wird nicht mehr so viel Strom erzeugt und evcc reguliert die Wallbox herunter. Sobald die Wolke weg ist, wird wieder mehr Strom erzeugt und die Wallbox kann dann leider nicht mehr den Ladevorgang erneut starten.

Wenn ich das richtig verstehe, bedeutet dies, dass der Modus PV nicht benutzt werden kann, sondern nur Min+PV, damit ein durchgehender Stromfluss sichergestellt wird und der Ladevorgang nicht gekappt wird?

andig, 2025-03-04:
Die Frage ist doch warum die Box in einen Fehlerstatus geht? Da muss der Hersteller ran!

andig, 2025-03-04:
Thank you for the valuable feedback!

bo-cheng, 2025-03-12:
We recently tried another method to connect to the Webasto Next Wallbox.
We used the OCPP interface instead of the default Modbus interface. So we replaced the Monta (former Webasto ChargeConnect) Link in the Wallbox to connect to evcc directly.

We tested it now nearly a week and the problem has not occured again. It seems like it is now working stable. There seems like a problem in the Modbus interface of the Webasto Next.