evcc-io/evcc #13042: enphase battery does not report charge and discharge watt values

thomas-hutterer-tik, 2024-03-19:
### Describe the bug

For a battery connected via enphase template inverter, the soc values are reported but charge and discharge watt are always 0

### Steps to reproduce

1. deploy config
2. start evcc
3. check logs for 0 values
...

### Configuration details

```yaml
---
network:
  schema: "http"
  host: "0.0.0.0"
  port: 7070
interval: "2s"
sponsortoken: "..."
log: "trace"
meters:
- name: "Gattringer"
  type: "custom"
  power:
    source: "mqtt"
    topic: "wa-pi/grid"
  currents:
  - source: "mqtt"
    topic: "wa-pi/grid/p1"
  - source: "mqtt"
    topic: "wa-pi/grid/p2"
  - source: "mqtt"
    topic: "wa-pi/grid/p3"
- name: "Batterie"
  type: "template"
  template: "enphase"
  usage: "battery"
  host: "10.0.0.98"
  token: "..."
- name: "PV"
  type: "template"
  template: "enphase"
  usage: "pv"
  host: "10.0.0.98"
  token: "..."
chargers:
- name: "Go-E-Simulator"
  type: "template"
  template: "go-e-v3"
  host: "localhost:8091"
site:
  title: "Gattringer"
  meters:
    grid: "Gattringer"
    pv:
    - "PV"
    battery:
    - "Batterie"
  residualPower: 100
loadpoints:
- title: "Go-E-Simulator"
  charger: "Go-E-Simulator"
  phases: 0
mqtt:
  broker: "127.0.0.1"
  topic: "evcc"
  user: "wm-DwS8L5ZoQQhx"
  password: "..."
[... 107 lines cut ...]
Mar 19 16:09:41 iWattControllerV3 evcc[32534]:                 "dc_switch_off": false,
Mar 19 16:09:41 iWattControllerV3 evcc[32534]:                 "encharge_rev": 2,
Mar 19 16:09:41 iWattControllerV3 evcc[32534]:                 "encharge_capacity": 3500
Mar 19 16:09:41 iWattControllerV3 evcc[32534]:             }
Mar 19 16:09:41 iWattControllerV3 evcc[32534]:         ]
Mar 19 16:09:41 iWattControllerV3 evcc[32534]:     }
Mar 19 16:09:41 iWattControllerV3 evcc[32534]: ]
Mar 19 16:09:41 iWattControllerV3 evcc[32534]: [site  ] DEBUG 2024/03/19 16:09:41 battery soc: 32%
Mar 19 16:09:41 iWattControllerV3 evcc[32534]: [site  ] DEBUG 2024/03/19 16:09:41 battery power: 0W
Mar 19 16:09:41 iWattControllerV3 evcc[32534]: [cache ] TRACE 2024/03/19 16:09:41 batteryCapacity: 0
Mar 19 16:09:41 iWattControllerV3 evcc[32534]: [site  ] DEBUG 2024/03/19 16:09:41 grid meter: -174W
Mar 19 16:09:41 iWattControllerV3 evcc[32534]: [site  ] DEBUG 2024/03/19 16:09:41 grid currents: [1.8 -1.51 -1.04]A
Mar 19 16:09:41 iWattControllerV3 evcc[32534]: [site  ] DEBUG 2024/03/19 16:09:41 site power: -74W
Mar 19 16:09:41 iWattControllerV3 evcc[32534]: [cache ] TRACE 2024/03/19 16:09:41 batterySoc: 31.666666666666668
Mar 19 16:09:41 iWattControllerV3 evcc[32534]: [cache ] TRACE 2024/03/19 16:09:41 batteryPower: 0
Mar 19 16:09:41 iWattControllerV3 evcc[32534]: [cache ] TRACE 2024/03/19 16:09:41 batteryEnergy: 0
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.124.10

TobiasNeuschulz, 2024-03-19:
Siehe:

https://github.com/evcc-io/evcc/discussions/12103

andig, 2024-03-20:
Danke Tobias, das ist ja schade. Wärst Du so nett und würdest im Template einen Hinweis für die IQ Battery ergänzen? Gerne mit Verlinkung auf Dein Dokument?

premultiply, 2024-03-21:
Das wäre dann https://github.com/evcc-io/evcc/pull/13066

andig, 2024-03-21:
@premultiply was hat das Polling beim battery mode mit den fehlenden Leistungen zu tun? Im Zweifel müsste das jemand mal ausprobieren.

premultiply, 2024-03-21:
Überhaupt nix.
Man muss aber hier wohl regelmäßig den POST-Request absetzen um dauerhaft Daten zu erhalten.

andig, 2024-03-21:
Dann wird das so nicht funktionieren weil BatteryMode im Zweifel nie aufgerufen wird.

premultiply, 2024-03-21:
Ja, stimmt. Schade.
Hast du einen alternativen Vorschlag?

andig, 2024-03-21:
Wie wärs denn (ungetestet) mit:

```
power:
  add:
  - watchdog
  - http
```

muss vielleicht nochwas für getan werden, aber das könnte plausibel sein?

andig, 2024-03-24:
@premultiply willst Du die Variante noch probieren? 

Eine weitere Option wäre, ein generelles `heartbeat` Interface einzuführen, das `custom` Geräte implementieren können. Dann wäre das unabhängig von einer einzelnen Funktion.

Dritte Variante: statt `heartbeat` ein `setup` Interface. Damit wäre auch klar, dass `setup` verschiedene Aufgaben übernehmen könnte und im `setup` drin dann- wie in Deinem Prototyp auch- ein `watchdog` Plugin stecken müsste. Bei `heartbeat` wäre nicht so ganz klar, wohin die Watchdoglogik gehört.

Was meinst Du? Gäbe es weitere Anwendungsfälle für `setup`?

premultiply, 2024-03-24:
Bist du sicher dass das funktionieren sollte?

```
power:
  add:
  - watchdog
  - http
```

`watchdog` ist doch zum schreiben und `add` würde einen Rückgabewert erwarten, oder?

`setup` klingt für mich erstmal passender, könnte aber auch mit einem reinen Heartbeat-Interface gut leben.

Mit `setup`könnte man prinzipiell auch gewisse Grundeinstellungen einmalig umstellen die für die weitere Funktion erforderlich sind. Das wäre z. B. dieser Remote-Modus bei SolarEdge. Ist aber wirklich nicht lebensnotwendig.

andig, 2024-03-24:
Magst Du es mal auf der Basis von https://github.com/evcc-io/evcc/pull/13129 probieren? Einfach auschecken und nochmal branchen.

closed 2024-04-20 by github-actions[bot]