evcc-io/evcc #13047: Error-Meldung im EVCC Protokoll für Mennekes Charge Control Wallbox?

uw3stfs-ha, 2024-03-20:
### Describe the bug

Im EVCC-Protokoll wird für die Mennekes Amtron Charge Control 2.0 die folgende Error-Meldung gelistet.
Welcher Fehler liegt vor?

[site  ] DEBUG 2024/03/20 06:52:21 ----
[lp-1  ] DEBUG 2024/03/20 06:52:21 charge power: -15W
[site  ] DEBUG 2024/03/20 06:52:21 pv power: 58W
[site  ] DEBUG 2024/03/20 06:52:21 grid meter: -15W
[site  ] DEBUG 2024/03/20 06:52:21 grid powers: [5 -5 -15]W
[site  ] DEBUG 2024/03/20 06:52:21 grid currents: [0.183 -0.025 -0.066]A
[site  ] DEBUG 2024/03/20 06:52:21 site power: -15W
[lp-1  ] DEBUG 2024/03/20 06:52:21 charge currents: [0.183 -0.025 -0.066]A
[lp-1  ] DEBUG 2024/03/20 06:52:21 charge total import: 339.589kWh
[lp-1  ] ERROR 2024/03/20 06:52:21 charger: modbus: exception '1' (illegal function), function '3'
[mennekes] ERROR 2024/03/20 06:52:27 heartbeat: modbus: exception '2' (illegal data address), function '6'
[mennekes] ERROR 2024/03/20 06:52:35 heartbeat: modbus: exception '2' (illegal data address), function '6'
[mennekes] ERROR 2024/03/20 06:52:43 heartbeat: modbus: exception '2' (illegal data address), function '6'
[mennekes] ERROR 2024/03/20 06:52:51 heartbeat: modbus: exception '2' (illegal data address), function '6'

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# charger definitions
# name can be freely chosen and is used as reference when assigning charger to vehicle
# for documentation see https://docs.evcc.io/docs/devices/chargers

# Menekes Amtron Chare Control
chargers:
Aktueller 'EVCC.yaml' Eintrag!
 - name: my_charger
    type: template
    template: mennekes
    # Modbus TCP
    modbus: tcpip
    id: 2
    host: 192.168.178.59 # IP-Adresse oder Hostname
    port: 502 # Port

Auch schon getestet!
chargers:
  - name: my_charger
    type: template
    template: bender "mennekes"
    host: 192.0.2.2 # IP-Adresse oder Hostname
    port: 502 # Port (optional)

Mit der folgenden Fehlermeldung:

[main  ] FATAL 2024/03/20 07:24:43 cannot create charger 'my_charger': cannot create charger type 'template': cannot create charger type 'mennekes-compact': 1 error(s) decoding:
```

### Log details

```text
Keine Ahnung wie!
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

V 0.124.10

andig, 2024-03-20:
@premultiply hier failed:

	if _, err := wb.conn.WriteSingleRegister(mennekesRegHeartbeat, mennekesHeartbeatToken); err != nil {
		wb.log.ERROR.Println("heartbeat:", err)
	}

keine Ahnung, warum das passieren sollte?

andig, 2024-03-20:
@uw3stfs-ha was heisst das denn jetzt? Ist der Fehler plötzlich verschwunden?

uw3stfs-ha, 2024-03-20:
Hallo,

nein der Fehler ist nicht verschwunden, ich habe nur anhand ihrer Antwort begriffen, daß ich von dem mit was ich hier konfrontiert werde absolut zu wenig verstehe um auf Fragen wie die Ihre richtig antworten oder reagieren zu können. 
Ich habe mir den „Zusammenbau“ anhand der evcc.yaml schlicht zu einfach vorgestellt!

Aber danke für ihre Nachfrage!

andig, 2024-03-20:
Die Frage ging gar nicht an dich… wir verstehen auch grad nicht, was da los ist. Mehr Details zur Wallbox könnten helfen.

uw3stfs-ha, 2024-03-20:
Die Wallbox ist eine Mennekes Amtron Charge Control 11 mit folgenden Systemparametern:
Systeminformation

Hersteller Seriennummer
1349201.22939
Anwendungsversion
5.22.5-13500-f55572d52
Linux OS version
4.14.181-g9e0869c56d84 #3 Mon Aug 2 15:50:55 CEST 2021 (UBI)
RCMB-Version
D0660, V2.01
EEBus-Stack-Version <http://192.168.178.59/#>￼
KEO framework 16.2.1 (Release-16.2.1-0-g3472056a)
EEBus SKI Geräte Identifikation <http://192.168.178.59/#>￼
d61c51d36441c1e44cad5b7889e393bc52253a6d

Die Wallbox ist in einem Ethernet-Netzwerk mit einem SMA Home Manager 2.0 und einem Wechselrichter SMA Tripower 4.0 verbunden.
Bisher war die Wallbox mit dem SMA Home Manager gekoppelt. Nur bin ich mit der Funktionsweise und Steuerung nicht zufrieden.
Deshalb auch meine Versuche mit „evcc“
Meine Einstellungen an der Wallbox kann ich nicht mehr genau nachvollziehen, da ich da ständig am testen mit anderen „möglichen" Einstellungen im Lastmanegement bin.

Evcc läuft bei mir auf dem aktuellen Home Assistant mit dem modbus proxi add on.

Die evcc.yaml sieht wie folgt aus:

# evcc configuration
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: 192.168.178.50 # Home Assistant (Raspberri Pi 4)
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 30s # control cycle interval. Interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
sponsortoken: eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJleHAiOjE3MTExODA4MDAsImlhdCI6MTcxMDA1NzYwMCwiaXNzIjoiZXZjYy5pbyIsInN1YiI6IlRyaWFsIFNwb25zb3JzaGlwIPCfpLkiLCJzcGUiOnRydWUsInNyYyI6Im1hIn0.78UrrdrvELwzWLLrXfZbR8dP9SoN2_8tcYK4zfv1m5c

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
# telemetry: true

# log settings
log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

[... 214 lines cut ...]
      msg: "Car connected at ${pvPower:%.1fk}kW PV"
    disconnect: # vehicle connected event
      title: Car disconnected
      msg: Car disconnected after ${connectedDuration}
    soc: # vehicle soc update event
      title: Soc updated
      msg: Battery charged to ${vehicleSoc:%.0f}%
    guest: # vehicle could not be identified
      title: Unknown vehicle
      msg: Unknown vehicle, guest connected?
  services:
  # - type: pushover
  #   app: # app id
  #   recipients:
  #   - # list of recipient ids
  # - type: telegram
  #   token: # bot id
  #   chats:
  #   - # list of chat ids
  # - type: email
  #   uri: smtp://<user>:<password>@<host>:<port>/?fromAddress=<from>&toAddresses=<to>
  # - type: ntfy
  #   uri: https://<host>/<topics>
  #   priority: <priority>
  #   tags: <tags>

andig, 2024-03-23:
@uw3stfs-ha Dein Token habe ich geblockt- bitte neues erstellen.

uw3stfs-ha, 2024-03-25:
Hallo andig,
konntet ihr den Modus "ERROR" schon etwas eingrenzen? Ich habe auch schon mit dem internen evcc "Modus Proxy" getestet, jedoch mit dem selben Ergebnis im Protokoll!

[site  ] DEBUG 2024/03/25 10:12:01 ----
[lp-1  ] DEBUG 2024/03/25 10:12:01 charge power: -1818W
[lp-1  ] WARN 2024/03/25 10:12:01 charge power must not be negative: -1818
[site  ] DEBUG 2024/03/25 10:12:01 pv power: 1868W
[site  ] DEBUG 2024/03/25 10:12:01 grid meter: -1818W
[site  ] DEBUG 2024/03/25 10:12:01 grid powers: [-596 -608 -615]W
[site  ] DEBUG 2024/03/25 10:12:01 grid currents: [-2.53 -2.57 -2.6]A
[site  ] DEBUG 2024/03/25 10:12:01 site power: -1818W
[lp-1  ] DEBUG 2024/03/25 10:12:01 charge currents: [-2.53 -2.57 -2.6]A
[lp-1  ] DEBUG 2024/03/25 10:12:01 charge total import: 355.446kWh
[lp-1  ] ERROR 2024/03/25 10:12:01 charger: modbus: exception '1' (illegal function), function '3'
[mennekes] ERROR 2024/03/25 10:12:05 heartbeat: modbus: exception '2' (illegal data address), function '6'
[mennekes] ERROR 2024/03/25 10:12:13 heartbeat: modbus: exception '2' (illegal data address), function '6'
[mennekes] ERROR 2024/03/25 10:12:21 heartbeat: modbus: exception '2' (illegal data address), function '6'
[mennekes] ERROR 2024/03/25 10:12:29 heartbeat: modbus: exception '2' (illegal data address), function '6'

andig, 2024-03-25:
Ist mir ein Rätsel- bei anderen Boxen geht es ja. Da musst Du bitte mal den Support fragen.

andig, 2024-03-26:
@premultiply kann es sein, dass Compact und Charge Control einfach ganz unterschiedliche Boxen sind und hier der

    ECU-BRx and ECU-BBx

Implementiert werden müsste? Doku liegt im Repo...

karlheupel, 2024-04-08:
@andig 
Hi, sehe ich das richtig, dass ihr an der Implementierung für die Charge Control arbeitet?

EIBHomeControl, 2024-04-11:
Ich habe auch 2 Mennekes Charge Control installiert. Bei mir laufen die beiden mit evcc.
Hier mal meine Config.
- name: wallbox1
  type: template
  template: bender
  host: 192.168.2.45
  port: 502

EIBHomeControl, 2024-04-11:
Ja genau diese, aber ohne Phasenumschaltung. HIer mal der Software-Stand
```
Anwendungsversion
5.22.3-13380-4a48dd58c
Linux OS version
4.14.181-g9e0869c56d84 #3 Mon Aug 2 15:50:55 CEST 2021 (UBI)
RCMB-Version
D0569, V1.02
EEBus-Stack-Version 
KEO framework 16.2.1 (Release-16.2.1-0-g3472056a)
```

Alles andere ist von SMA bzw. BYD.

karlheupel, 2024-04-11:
Ah cool, danke du siehst die also auch im SHM / Portal von SMA wenn du von unterwegs drauf schaust oder hast du den nicht?

EIBHomeControl, 2024-04-11:
Ich sehe meine beiden Wallboxen im SunnyPortal. Gesteuert werden diese aber über evcc

premultiply, 2024-04-11:
evcc meldet ALLE Ladepunkte mit aktuellem Verbrauch und Energiebedarf an den SHM bzw. Sunny Portal. Das funktioniert mit jeder Wallbox oder sonstigem Verbraucher.

andig, 2024-04-11:
Können wir bitte zum Thema zurück kommen. Die Frage war: wird die Mennekes Charge Control unterstützt? Die Antwort scheint zu sein: ja, mittels `bender` Template. Wenn dem so ist wäre die Aufgabe, das im Template deutlich zu machen.

karlheupel, 2024-04-11:
@andig wäre top wenn dann och phasenumschaltung ginge, dann hätte ich meine Wallbox gefunden, weil sie auch noch dienstwagenabrechnung kann.

andig, 2024-04-11:
Ja, ist hier aber auch nicht das Thema. Das Bender Template kann das nicht. Falls uns jemand sagen kann welche Register es dafür braucht ließe sich auch das umsetzen. Mit der Information bitte neues Issue.

uw3stfs-ha, 2024-04-11:
Hallo andig,
ich habe in der Zwischenzeit ebenfalls die Lösung für den Betrieb der "Mennekes Charge Control 11 C2" mit evcc gefunden.
Ja, der Trick ist genau wie 'EIBHomeControl' schon festgestellt hat, in der evcc.yaml muss für die „Charge Control" als Template: „bender" eingetragen sein, auch dann, wenn in der Wallbox-Konfiguration im Modbus Registersatz an Position drei „MENNEKES“ steht. Auf diese „Besonderheit“ in der Konfiguration der "Charge Control“, sollte unbedingt in der Dokumentation der „evcc" App hingewiesen werden!

Ich bedanke mich für euere Unterstützung und schließe damit dieses Issue.

closed 2024-04-11 by uw3stfs-ha

premultiply, 2024-04-11:
https://docs.evcc.io/docs/devices/chargers#amedio-amtron-chargecontrol-professional