evcc-io/evcc #13050: Introduce time-based loadpoint metrics

naltatis, 2024-03-20:
Since this is a topic that comes up regularly, I'm creating this issue here to funnel requests.

Currently, we collect a lot of data (energy, % solar, price, co2, meter readings, ...) based on charging sessions. Since evcc's primary use case is EV charging, this is perfectly fine and will remain as is. However, the session concept does not really work for use-cases where no device is plugged in/out (heat pump, water heater, cooler, ...) and for chargers where we can't detect the connection status (smart plugs). There are workarounds to produce logged charging sessions in these scenarios (temporary PV/Stop/PV, service restart), but these are ugly.

A better solution is to also collect and store the above values on a pure time basis. This way can build hourly/daily/monthly statistics for every kind of load point. We'll probably not do high-resolution data collection (please use something like InfluxDB/Grafana for this), but enough resolution to create meaningful visualizations.

This topic is not very high on our priority list right now, but we'll build something like this eventually.

userwithoutpassword, 2024-10-31:
https://github.com/evcc-io/evcc/discussions/16999

longtom99, 2025-07-20:
Maybe it is exactly what you are planning, what i would need:
Track Power for a Heatpump(s) when power above standby (regardless of any contact or PV production).
WITH PRIORITY from PV, example i have Car (Prio 10), Heatpump for Water (Prio 20), Heatpump for heating (Prio 30).
So the PV generation power (besides other/house consumption) will be first deducted from Prio 30, then 20, then 10.
That i have the costs for each "loadpoint" according to Prio.

Andreas751, 2025-07-27:
Hi. I really appreciate the new implementation, measuring also heatpump sessions.

with version 0.205.0 I can see charging sessions for the heat pump as well, BUT:

I have a SG-Ready heat pump connected to evcc and found the daily consumption in the charging session statistics matching the reality on most days, but not on all days.

Tracing down to my own traces it looks like, the "daily" charging session starts with the first trigger from evcc to the SG ready switch.

Means all consumption earlier that day are lost in the statistics. I would assume the session starting at 00:00 every day.

Also, the length of the sessions look strange:

[image]

I could not find out, if the negative "standybypower" value might play a role.

Here comes the most important part of the yaml (I hope ;-))

---
meters:
- name: Waermepumpe-Meter
  type: template
  template: fritzdect

chargers:
- name: Waermepumpe-Schalter
  type: template
  template: fritzdect
  uri: https://fritz.box # HTTP(S) Adresse (optional)
  standbypower: -475 # Leistung oberhalb des angegebenen Wertes wird als Ladeleistung gewertet (optional)
  integrateddevice: true
  heating: true
  icon: waterheater

vehicles:
- name: Viessmann-WP
  type: custom
  title: Viessmann-WP
  icon: waterheater
  phases: 1
  soc:
    source: script
    cmd: /bin/sh -c '/etc/evcc-watertemp.sh'
    cache: 5m
 
loadpoints:
- title: Warmwasser
  charger: Waermepumpe-Schalter
  meter: Waermepumpe-Meter
  vehicle: Viessmann-WP
  enable:
    threshold: 0
    delay: 1m
  disable:
    threshold: 0
    delay: 10m # Laufzeit mindestens 10 Minuten
    #  phases: 1

---

naltatis, 2026-09-02:
The core mechanics and first visualizations (battery chart, history 🧪 page) are implemented. Closing here.

closed 2026-09-02 by naltatis