evcc-io/evcc #13056: Vehicle: add start/stop setter, tagetsoc getter

andig, 2024-03-20:
Refs https://github.com/evcc-io/evcc/pull/13055#issuecomment-2009231877

Adminius, 2024-03-20:
FinishTime? ;)

GrimmiMeloni, 2024-03-22:
I‘m actively working on the required generator fix.

andig, 2024-03-22:
@GrimmiMeloni aktuell generieren wir ja sowas:

    -t "api.SocLimiter,GetLimitSoc,func() (float64, error)"

Wenn wir stattdessen in der Lage wären, alle Methoden eines Interfaces (auch falls >1) auszulesen würde das dann so aussehen:

    -t "api.SocLimiter"

Sowohl Methode als auch Signatur würden/müssten dann per Reflect oder AST ermittelt und an den Generator übergeben werden. Wäre das denkbar? Würde auch viele/alle `go:generate` Statements deutlich vereinfachen, auch ohne dass Typsicherheit leiden würde.

GrimmiMeloni, 2024-03-22:
Hab ich auch schon überlegt. Aktuell bin ich erstmal am „backend“. Die Daten in den Strukturen sind bereits sauber, ich muss noch am Template feilen. Wenn das läuft wäre Deine Idee aber auch bei mir auf der Wishlist, denn einzelne Funktionen aus einem Interface allein lassen sich dann sowieso nicht anbinden. Es müssen dann immer alle angegeben werden. Die Generator Zeilen werden dann schnell länger als die Würmer auf Arrakis.

closed 2024-03-23 by andig