evcc-io/evcc #13069: Home Battery Discharge Control not Working with Minimum SoC

Percy2Live, 2024-03-20:
### Describe the bug

The option "prevent home battery discharge for fast charge" does work well - except if there is a minimum SoC configured for the vehicle. In that case the vehicle will automatically start to charge when plugged in - but the home battery will be used for that.

Expected behaviour: The home battery should not be used in that case, similar to what happens when the fast charge option is selected

### Steps to reproduce

1. Activate the option prevent home battery discharge for fast charge
2. Enable a minimum SoC for the vehicle in the arrival tab (e.g. 80%)
3. Plug in the vehicle with a lower SoC (e.g. 50%)

The vehicle will start to charge automatically, but the home battery will be used.
When step 2 is skipped and fast charge is selected instead, everything works like expected: The home battery is not used

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: evcc
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070
#uri: #############
interval: 10s # control cycle interval
sponsortoken: #######

log: debug

meters:

- name: Homemanager
  type: sma
  serial: 1901### # insert your own serialnumber 
    
- name: Tripower
  type: template
  template: sma-inverter
  usage: pv
  host: 192.1####P-Adresse oder Hostname
  password: ### # Anlagenpasswort / Installateurspasswort # Optional 

- name: STP_SE
  type: template
  template: sma-hybrid
  usage: pv
    
  # Modbus TCP Inverter
  modbus: tcpip
  id: 3
  host: 19####9 # Hostname
  port: 502 # Port 

- name: byd
  type: template
  template: sma-hybrid
  usage: battery
    
  # Modbus TCP Battery
  modbus: tcpip
  id: 3
  host: 1###9 # Hostname
  port: 502 # Port
    
chargers:
#- type: template
#  template: go-e-v3 
#  host: 1####3  
#  name: wbPascal
- type: template
  template: go-e-v3 
  host: 1####55  
  name: wbSandra

vehicles:
- type: template
  template: nissan 
  title: Leaf Pascal 
  user: ####  
  password: ####
  vin: #####  
  capacity: 40  
  name: ev1 

loadpoints:
- title: Ladepunkt Sandra
  charger: wbSandra
  vehicle: ev1
#  mode: now
  mode: pv
#  mode: minpv #In Kombination mit den Sunny Portal Einstellungen kann in diesem Modus auch mit reduziertem PV Anteil geladen werden, z.B. mit 30% Netzbezug
  phases: 1 # ev phases (default 3)
  enable: # pv mode enable behavior
    delay: 5m # threshold must be exceeded for this long
#Summer value
    threshold: -1600 # grid power threshold (in Watts, negative=export). If zero, export must exceed minimum charge power to enable
#Winter value
#    threshold: -1000 # grid power threshold (in Watts, negative=export). If zero, export must exceed minimum charge power to enable
  disable: # pv mode disable behavior
    delay: 10m # threshold must be exceeded for this long
#Summer value
    threshold: 300 # maximum import power (W)
#Winter value
#    threshold: 900 # maximum import power (W)
  guardDuration: 10m # switch charger contactor not more often than this (default 5m)
  minCurrent: 6 # minimum charge current (default 6A)
  maxCurrent: 32 # maximum charge current (default 16A)
  soc:
    poll:
      mode: always
      interval: 120m #interval for polling while not charging
#    min: 0
#    target: 100
    estimate: true

#- title: Ladepunkt Pascal
#  charger: wbPascal
#  vehicles: 
#  - ev1
#  mode: pv
#  resetOnDisconnect: true
#  phases: 3 # ev phases (default 3)
#  enable: # pv mode enable behavior
#    delay: 5m # threshold must be exceeded for this long
#    threshold: -1600 # grid power threshold (in Watts, negative=export). If zero, export must exceed minimum charge power to enable
#  disable: # pv mode disable behavior
#    delay: 10m # threshold must be exceeded for this long
#    threshold: 300 # maximum import power (W)
#  guardDuration: 10m # switch charger contactor not more often than this (default 5m)
#  minCurrent: 6 # minimum charge current (default 6A)
#  maxCurrent: 32 # maximum charge current (default 16A)
#  soc:
#    poll:
#      mode: charging
#      interval: 60m
#    min: 0
#    target: 100
#    estimate: true

site:
  title: Home Ladestation # display name for UI
  meters:
    grid: Homemanager # grid meter
    pv:
      - Tripower # list of pv inverters/ meters
      - STP_SE
    battery: # battery meter
      - byd
#  prioritySoC: 50 # give home battery priority up to this soc (empty to disable)
#  bufferSoC: 85 # Ist SoC der Batterie > bufferSoC, wird dieser Teil als PV Erzeugung zum Laden des Fahrzeugs genutzt (auch im PV Modus) (empty to disable)
  residualPower: -200 #-690 #Wintermodus: Wie viel Netzbezug ist ok, damit trotzdem geladen wird (z.B. -1000W)

#hems: #Versuch, die SMA Logik zur Steuerung zu verwenden. Aktuell "allowcontrol" auf false gesetzt, da es nicht so funktioniert wie gewünscht
#  type: sma
#  allowcontrol: false
#  vendorid: ###
#  deviceid: ###

# tariffs are the fixed or variable tariffs
# cheap (tibber/awattar) can be used to define a tariff rate considered cheap enough for charging
tariffs:
  currency: EUR # three letter ISO-4217 currency code (default EUR)
  grid:
    # either static grid price
    type: fixed
    price: 0.26 # EUR/kWh
  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.127 # EUR/kWh
influx:
  url: http://#####
  database: evcc
  user: #####
  password: #####
  
plant: #####
```

### Log details

```text
will follow
```

### What type of operating system are you running?

Docker container

### Version

evcc version 0.124.9

StefanSchoof, 2024-03-20:
This topic was also discussed in #11504

pustelbaer, 2024-03-20:
I know this problem too. The option is called "Avoid discharge in fast charge mode and for planned charging". 
The minimum charge to a predefined SOC on arrival is a planned charge! Then the battery should not be used. 

I would really appreciate it if this could be fixed.

premultiply, 2024-03-21:
> The minimum charge to a predefined SOC on arrival is a planned charge!

It isn't.

naltatis, 2024-03-21:
> It isn't.

Correct, but I guess we should clarify/update the label to avoid confusion.
Battery-hold while minsoc charging should work as of https://github.com/evcc-io/evcc/pull/11941

naltatis, 2024-03-21:
@Percy2Live can't reproduce it. Works as expected in my setup. Tested on nightly, but the minsoc hold features has been introduced a few releases ago. Can anyone else reproduce this? Any anomalies in your log file (unable to control battery)? 

[image]

pustelbaer, 2024-03-21:
I have a bit of difficulty in understanding why this behaviour is desirable.

I set the min SOC under the "Plan" link. There the value is labelled as "Charge planning". 
In the battery settings, the option is labelled "Prevent discharging ... during planned charging". Based on this description, I as a user conclude that the battery is not being used to charge the car.
 
When I connect my car to the wall box, evcc starts the charging at 11 kW, which is fast charging in my case. My battery and the hybrid inverter are involved in the charging process with the max. available power (approx. 6 kW). This is exactly what I want to avoid, as it put a lot of stress on the inverter. It gets very warm over time and I also want to avoid this constant load on the battery. 

From the forum posts, it seems that there are other users who are bothered by this behaviour. I can understand if some do not want to change it. How about introducing an additional option in the yaml file that allows you to change this behaviour? If the option is not present, evcc would work as before so that the compatibility is maintained.

pustelbaer, 2024-03-21:
@naltatis I'm surprised that your system behaves differently. I have a very similar configuration to percy2live: an SMA Tripower SE with BYD HVS battery and a go-e gemini wallbox. Maybe it has something to do with that?

pustelbaer, 2024-03-22:
I have now installed the latest version. lt still behaves the same. There are no errors in the log.

[image]

[image]

naltatis, 2024-03-23:
Ah interessting. I guess something with the hold implementation for your inverter is wrong. In the Screenshot we can see, that evcc assumes it has successfully "locked" your battery while minsoc charging (Batterie gesperrt). But since it still provides power there's someting wrong.

@premultiply any ideas?

premultiply, 2024-03-23:
No idea. ☹️
Should be in battery hold mode if "gesperrt" is displayed.
I don't see what should be different between manual fast charging and automatic minsoc fast charging.

Percy2Live, 2024-03-24:
For me it is similar as for Pustelbaer. The UI says "battery locked" but the battery is used for charging in MinSoC mode:
[image]
[image]

In general the battery lock works, when I use the fast charge option instead, the behaviour is as expected:
[image]

Does the MinSoC battery lock call another function as the fast charge lock?

andig, 2024-03-28:
Missing logfile

closed 2024-03-28 by andig

StefanSchoof, 2024-03-28:
On docker you can just use `docker logs` https://docs.docker.com/reference/cli/docker/container/logs/

Or you run evcc to create the log on windows or Mac to produce the needed log.

pustelbaer, 2024-03-28:
I can download the log from portainer web ui, but is it what you're looking for?

e.g.
[lp-2  ] DEBUG 2024/03/28 20:32:19 charge power: 0W
[site  ] DEBUG 2024/03/28 20:32:19 pv power: 0W
...

andig, 2024-03-28:
@StefanSchoof ein PR für die Doku wäre Klasse :)

pustelbaer, 2024-04-02:
An friend has the same problem when charging his ID.3. I would like to help to solve the error, but I need to know how to access the log file (I have already searched the file system of the Docker container completely, but cannot find it). Can someone give me a short info?
P.S. Simply closing the entry is not a solution either.

StefanSchoof, 2024-04-02:
@andig Die Doku hat dazu schon:

> If you're using Docker, you should be able to use docker logs. See the [Docker documentation](https://docs.docker.com/config/containers/logging/) for more details.

 https://docs.evcc.io/en/docs/faq/#how-do-i-create-a-log-file-for-error-analysis

pustelbaer, 2024-04-08:
Many thanks for the information. Here is a log file from today. The start of the loading process can be found at '2024/04/08 20:48:39'.
I started the process in "PV" mode by setting the min SOC to 50% on arrival. In the UI, the battery was again displayed as 'locked' but discharged. I hope it'shelpful ;-)

[_evcc_logs_20240408.txt](https://github.com/evcc-io/evcc/files/14910080/_evcc_logs_20240408.txt)

olmo1501, 2024-05-07:
Could this be related to the fact, that using modbus registers 40149 and 40151 doesn't seem to possible anymore since SMA's SHM2 update to version 20.14.13.R? Please see this rather long thread: https://www.photovoltaikforum.com/thread/220338-shm2-0-software-version-2-14-10-r-ca-m%C3%A4rz-2024/ Doesn't make sense at first glance but Homemanager seems to be overruling the inverter's Modbus parameters.

andig, 2024-05-07:
Kann gut sein- wirst Du aber nur mit SMA klären können. Kein evcc Thema...

olmo1501, 2024-05-07:
Schon klar. Aber wenn Ihr die genannten Register nutzt, wird das aktuell nicht (mehr) funktionieren. Darauf wollte ich hinaus.

kjohann7, 2024-10-13:
Bei mir wird der Stromspeicher des Hauses, obwohl "Verhindere Entladung im Schnell-Modus und bei geplantem Laden" aktiviert ist, bei folgenden Einstellungen entladen:

- Schnell laden
- Geplantes Laden Abfahrt
- Geplantes Laden Ankunft

- Ich fände es wünschenswert, wenn in keinem dieser Modi der Stromspeicher entladen würde.
- Den Text "Batterie (gesperrt)" wie man es oben in den Screenshots sehen kann, habe ich in meinem GUI noch nie gesehen.
- Eine relativ häufige Fehlermeldung in meinen Logs ist: "[site ] ERROR 2024/10/13 09:18:50 battery mode: modbus: exception '4' (server device failure), function '4' "

Kann mir jemand hilfreiche Tips geben?

Grundsätzliche Fragen sind:
1. Ist die oben gezeigte Fehlermeldung eine EVCC-Fehlermeldung oder kommt sie vom Stromspeicher oder dem Wechselrichter?
2. Ist dieser Fehler möglicherweise die Ursache für das Fehlverhalten von EVCC?
3. Sind, um die Batteriesperre nutzen zu können, besondere Einstellungen in der EVCC.yaml oder über die GUI-Konfiguration nötig?

Schon jetzt vielen Dank für Eure Unterstützung,
beste Grüße
Klaus

andig, 2024-10-13:
Das Thema hier ist seit 6 Monaten erledigt. Gerne neues auf machen.