evcc-io/evcc #13071: Smart switch with „heating: true“ option: Allow the „planner“-tab to set min and max temperature

jubudu, 2024-03-21:
**Is your feature request related to a problem? Please describe.**

I use a shelly for turning a heating pump on and off. The heating pump heats my heating-water buffer. It is not smart but cheap…!
It should keep a minimum temperature in the buffer if now solar power is available.
If excess solar power is available, it should heat up the buffer to a higher temperature. So I can buffer the excess energy for i.e. the evening or next morning.

If I use the option „heating: true“ for the shelly, I cannot define a minSoc i.e. minTemp in order to set the  minimum water temperature of the buffer.

My workaround: I define the heating-water buffer as a car and get the temperature as %-value. Using the  „planner -> arrival“, I can set the minSoc  and maxSoc as temperatures in 5 degree steps, which is a little bit too rough for this purpose. But the functionality is given and evcc keeps the min temperature and heats up the buffer, when there is excess solar power available. Fine!

**Describe the solution you'd like**

Make the „planner“ available for  „heating: true“ option, too.
Show the values as temperatures.
Rename the tab „departure“ to „timer“, and
„arrival“ to „temperature setting“, and 
the two settings to „min temperature“ and „max temperature“

naltatis, 2024-03-21:
We'll probably rework/remove the "arrival" tab. So the proposed solution will be different. But what I'm extracting here is that  you want a minsoc-like behavior when the temperature drops below a certain value, right?

Why do you also want to specify a max temperature? Right now this is what the "limit" should already do.

RenatusRo, 2024-03-21:
I pretty much achieved sth you want doing this way:

1
I modified the frontend to allow visualisation ot the planner,
so _'Make the „planner“ available for „heating: true“ option'_ 
is really a nice-to-have)

2
use PV mode, boiler defined as intended as heating and integrated device

3
wrote an external controller which takes the difference
between target temperature and current temperature
to calculate needed energy amount in kWh to achieve
target temperature next morning 8:00.
the controller disconnects and reconnects the
boiler (if not currently loading) to reset the charging session
and allow for correct counting of loaded energy with respect to 
needed energy.
send kWh and target time via API to evcc planner

this works pretty nice.
normally PV will sufficently warm up the water.
on dark days planner will target the low CO2 hours.
the plan is sent several times a day (so adjusting for 
temperature changes in both directions or even nullifying
the plan)
I slightly overcommit energy to the planner. The thermostats
will fire anyway if the target temperature has been reached
and switch off the heating.

BOFH90, 2024-11-11:
I got the same Problem as I integrated a heater for my aviary and could not get it to hold a minimum Temperature.
I "fixed" it via HomeAssistant and set the min SOC for the heater there and it works.

Only Problem is you can only do 5 Degree-Steps (0, 5, 10, 20, 25...) and nothing in between which would be a requirement for heating. 

I would be fine if you bring the minsoc setting for heater back in the configuration.

longtom99, 2025-01-06:
I also want to use 1 degree steps.
Want to use 59 degrees for my heatpump.
I would need it for the Standard-Limit.
Also it would be nice to have Degrees then instead of % in the display (for minimum/standard when set) when used for Heatpumps.

premultiply, 2025-01-06:
Mmmh. This should be better solved via a thermostat or home automation?

The task of evcc is energy management. To this end, it _permits_ or _prohibits_ individual consumers to obtain electrical energy.
The main task is the charging of electric vehicles.

premultiply, 2025-01-06:
This is correct, but exactly this does not require knowledge and control over temperatures.

BOFH90, 2025-01-06:
> This is correct, but exactly this does not require knowledge and control over temperatures.

I do this with my Warmwater-Heat pump and control the temperature and shut the SG-Ready Contact off when it reaches a certain temperature.

premultiply, 2025-01-06:
Right. The heatpump should shut off on its own when the target temperature is reached.
Exactly like a BMS would stop charging if the battery reached the target of 100% state of charge.

SG Ready interface is just a _recommendation_ for raising or lowering the energy consumption in some scenarios but it is not designed to power on or shut off anything. The interfaced device should always operate on its own.

jubudu, 2025-01-27:
> Right. The heatpump should shut off on its own when the target temperature is reached. Exactly like a BMS would stop charging if the battery reached the target of 100% state of charge.

Yes. It does. 

> SG Ready interface is just a _recommendation_ for raising or lowering the energy consumption in some scenarios but it is not designed to power on or shut off anything. The interfaced device should always operate on its own.

No, this is not the use case i.e. for Buderus heatpump boilers: 
If the SG-ready contact is activated,
the boiler keeps on heating until the max. temperature is reached. When SG-ready contact is deactivated, it stops heating and keeps min./normal temperature.

In combination with i.e. tibber it makes sense, to use SG-ready when tarif is (very) cheap to raise the temperature consuming cheap energy.