evcc-io/evcc #13074: Anpassung Hintergrund oberer Bildschirmrand

StevieC121176, 2024-03-21:
Öffnet man im hellen Design z.Bsp. Die Batterieeinstellungen wird der obere Bildchirmrand nicht an die Hintergrundfarbe angepasst und bleibt hellgrau. Beim dunklen Farbschema hingegen wird dieser angepasst. Wäre das auch beim hellen Farbschema möglich?

[image]
[image]

naltatis, 2024-03-21:
Sollte möglich sein. Ich schau mir das mal an.

StevieC121176, 2024-03-23:
Die Hintergrundfarbe ist eher dunkelgrau, nicht dieses sehr dunkle Blau aus dem dark theme.

[image]

naltatis, 2024-03-23:
Ja, perfekt bekommen wir das an dieser Stelle nicht hin (zumindest ohne manuelle Farbberechnung und so). Der Backdrop hat ne Transparenz auf den aktuellen Hintergrund. Ich würde hier jetzt ungern die genauen Farben messen. Im Dark-Mode gibt es diese Differenz heute auch, da ist sie nur etwas kleiner, weil natürlich ein dunkles Overlay auf unseren dunkelblauen Hintergrund trifft.

Ich würd das hier erstmal als "good enough" schließen, ok?

closed 2024-03-23 by naltatis

StevieC121176, 2024-03-24:
Na klar passt so, der Farbunterschied ist nicht so gravierend.

PR #13102 (2024-03-22): UI: dark menubar color on modal
fixes #13074

- switch to dark ios menu bar when modal backdrop is open
- fixed type warning (number instead of bool)