evcc-io/evcc #13075: UK CO2 data incorrect in evcc.dist,yaml

pmknowles, 2024-03-21:
### Describe the bug

The instructions for Tariff CO2 in the evcc.dist yaml say (by the way this information isn't in the documentation for tariffs)

_type: ngeso # National Grid Electricity System Operator data (United Kingdom only) https://carbonintensity.org.uk/
provides national data if both region and postcode are omitted - do not supply both at the same time!
region: 1 optional, coarser than using a postcode - see https://api.carbonintensity.org.uk/ for full list
postcode: SW1A1AA # optional_

This is incorrect an results in no information

The region details are at
https://carbon-intensity.github.io/api-definitions/#region-list

Only the 'outgoing' postcode should be used 
postcode - Outward postcode i.e. RG41 or SW1 or TF8. Do not include full postcode, outward postcode only

### Steps to reproduce

add details to yaml
tariffs:
  currency: GBP
  grid:
    type: fixed
    price: 0.27804
  co2:
    type: ngeso
    postcode: SW11AA

### Configuration details

```yaml
tariffs:
  currency: GBP
  grid:
    type: fixed
    price: 0.27804
  co2:
    type: ngeso
    postcode: SW1A1AA
```

### Log details

```text
Replacing the postcode with SW1 resolves it
```

### What type of operating system are you running?

Linux

### Version

v0.124.10

andig, 2024-03-21:
> The instructions for Tariff CO2 in the evcc.dist yaml say (by the way this information isn't in the documentation for tariffs)

@pmknowles happy to take a PR for both issues!

pmknowles, 2024-03-21:
Done. In the UK post codes have 2 parts so SW1A 1AA has 'outbound' (SW1A) which tells the sending sorting office where to direct it to and the 'inbound' part (1AA) which tells the receiving sorting office where to direct it to within the postcode area.

closed 2024-03-22 by andig