evcc-io/evcc #13084: Huawei battery control charge

mucki12, 2024-03-21:
### Describe the bug

Hallo,

nachdem dankenswerter Weise ein Test zum Laden des Hausspeicher zur Verfügung gestellt wurde:
https://github.com/evcc-io/evcc/pull/10814
ist aufgefallen, dass das Laden des Huawei Luna noch nicht klappt. „Charge from grid“ wird nicht eingeschaltet.

Genutzt wird das Template huawei-dongle-powersensor

Auszug für das Einschalten von „Charge from grid“:

```
          - source: const
            value: 1 # Enable
            set: 
              source: modbus
              {{- include "modbus" . | indent 12 }}
              register:
                address: 47087 # Charge from grid
                type: writemultiple
                encode: uint32
```

Gemäß Modbus Definition (https://www.photovoltaikforum.com/core/attachment/315028-solar-inverter-modbus-interface-definitions-pdf/ ) müsste es wohl „encode: uint16“ sein.

Wenn ich mich nicht irre müsste das dann wohl auch im Huawei Smartlogger Template geändert werde.

Könnte das jemand ändern? Habe leider keinen Plan wie ich das als PR machen kann :-(

### Steps to reproduce

1. evcc testing aus https://github.com/evcc-io/evcc/pull/10814 installieren
2. versuchen den Luna zu laden
3. Fehlermeldung
...

### Configuration details

```yaml
meters:
  - type: template
    template: huawei-dongle-powersensor 
    id: 1  
    host: 192.168.2.168  
    port: 502  
    usage: battery  
    modbus: tcpip  
    timeout: 15s  
    name: battery
    capacity: 10

chargers:
- type: battery
  name: lunacharger
  battery: battery

loadpoints:
- title: LUNA
  charger: lunacharger
```

### Log details

```text
ERROR 2024/03/21 21:12:50 charger: modbus: exception '4' (server device failure), function '16'
```

### What type of operating system are you running?

Docker container

mucki12, 2024-03-22:
@andig 
Mit dem Test Build wird AC Laden korrekt eingeschaltet. Vielen Dank!

andig, 2024-03-23:
Closed in https://github.com/evcc-io/evcc/commit/2595afff4325cd19e23436a3c939e28496fe7354

closed 2024-03-23 by andig