evcc-io/evcc #13090: Easee 3p->1p Umschaltung: schaltet gleich wieder auf 3p hoch (#12825 follow up)

kai-morich, 2024-03-22:
### Describe the bug

Zunächst wird auf 1phasig runtergeschaltet

    [easee ] TRACE 2024/03/22 10:31:02 POST https://api.easee.com/api/sites/400000/circuits/300000/settings
    [easee ] TRACE 2024/03/22 10:31:02 {"dynamicCircuitCurrentP1":16,"dynamicCircuitCurrentP2":0,"dynamicCircuitCurrentP3":0}

Die Easee hat mit der Umschaltung begonnen, in dem sie sich abschaltet und kurz danach wieder 1phasig anschalten würde.

Währenddessen ist evcc der Annahme das genug Überschuss für 3phasig vorhanden ist

    [lp-1  ] DEBUG 2024/03/22 10:31:35 available power 4341W > 4140W min 3p threshold

und schaltet wieder hoch

    [easee ] TRACE 2024/03/22 10:31:35 POST https://api.easee.com/api/sites/400000/circuits/300000/settings
    [easee ] TRACE 2024/03/22 10:31:35 {"dynamicCircuitCurrentP1":16,"dynamicCircuitCurrentP2":16,"dynamicCircuitCurrentP3":16}

was die Easee wegen 'Lastausgleich' erst nach einigen Minuten  ausführt.

Im Grafana-Dashboard ist zu sehen, das der Überschuss (grün - gelb) maximal einige 100 Watt ist.

[image]

Der unrealistische Überschuss war schon in https://github.com/evcc-io/evcc/issues/12825#issuecomment-1987221102 aufgefallen, aber zunächst wurde erst etwas anderes gefixt

### Steps to reproduce

Um die Situation unabhängig vom Sonnenstand zu erzwingen hatte ich zunächst im Modus 'schnell' mit 3p geladen, dann mit 'Min+PV' auf zunächst 3 * 6A reduziert. Ab hier beginnt das begefügte log / screenshot. Nach dem Phasetimer sollte eigentlich auf 1 * 6A runtergeschaltet werden, da nur <1500 Watt Sonnenertrag vorhanden sind. Stattdessen wird versucht wieder auf 3 Phasen hochzuschalten, was die Easee wegen 'Lastausgleich' erst nach einigem Minuten macht. Das 3phasige Wiedereinschalten ist nicht mehr im log / screenshot enthalten.

### Configuration details

```yaml
network:
  schema: http
  host: localhost
  port: 7070
log: debug
levels:
  easee: trace
sponsortoken: ***
plant: ***

meters:
- name: grid
  type: template
  template: vzlogger
  usage: grid
  host: ***
  port: 2071
  uuid: ***
- name: pv
  type: template
  template: solarlog 
  usage: pv  
  host: ***
  port: 2081

chargers:
- name: wallbox
  type: template
  template: easee
  user: ***
  password: ***
  charger: ABCDEF
  timeout: 30s
  authorize: true

loadpoints:
- title: Zuhause
[... 161 lines cut ...]
[lp-1  ] DEBUG 2024/03/22 10:32:05 charge currents: [0.009 0.004 0.008]A
[lp-1  ] DEBUG 2024/03/22 10:32:05 charge total import: 3448.002kWh
[lp-1  ] DEBUG 2024/03/22 10:32:05 charger status: B
[lp-1  ] DEBUG 2024/03/22 10:32:05 pv charge current: 0.2A = 0A + 0.2A (-138W @ 3p)
[site  ] DEBUG 2024/03/22 10:32:05 ----
[lp-1  ] DEBUG 2024/03/22 10:32:05 charge power: 0W
[site  ] DEBUG 2024/03/22 10:32:35 pv power: 1458W
[site  ] DEBUG 2024/03/22 10:32:35 grid meter: -432W
[site  ] DEBUG 2024/03/22 10:32:35 site power: -432W
[lp-1  ] DEBUG 2024/03/22 10:32:35 charge currents: [0.009 0.004 0.008]A
[lp-1  ] DEBUG 2024/03/22 10:32:35 charge total import: 3448.002kWh
[lp-1  ] DEBUG 2024/03/22 10:32:35 charger status: B
[lp-1  ] DEBUG 2024/03/22 10:32:35 pv charge current: 0.626A = 0A + 0.626A (-432W @ 3p)
[lp-1  ] DEBUG 2024/03/22 10:32:35 wake-up vehicle, attempts left: 4
[site  ] DEBUG 2024/03/22 10:32:35 ----
[lp-1  ] DEBUG 2024/03/22 10:32:35 charge power: 0W
```

### What type of operating system are you running?

Linux

### Version

0.124.10

andig, 2024-03-22:
available power 4341W > 4140W min 3p threshold

interessant. Wo kommt das her? Und warum geht es ohne Timer wieder hoch?

/cc @GrimmiMeloni @MarkusGH

kai-morich, 2024-03-22:
gibt es noch weitere traces die ich einschalten sollte?

GrimmiMeloni, 2024-03-23:
Wenn man den ganzen Lärm von der Easee mal beiseite lässt fällt auf, daß in jedem Intervall 30s nichts passiert zwischen

```
[lp-1  ] DEBUG 2024/03/22 10:31:35 charge power: 0W
[...]
[site  ] DEBUG 2024/03/22 10:32:05 pv power: 1457W
```

Spannend ist wo wir die Zeit verlieren. Vielleicht in `Site.updateMeters()` ?
@kai-morich kannst Du für Deine meters mal Trace logging einschalten? Vielleicht sehen wir dann wo genau die Zeit verloren geht.

kai-morich, 2024-03-24:
Neuer Fall in [evcc23.log](https://github.com/evcc-io/evcc/files/14735873/evcc23.log) aufgezeichnet mit ``log: trace``  und Fehler irgendwo in diesem Bereich:

    [easee ] TRACE 2024/03/24 15:22:51 {"dynamicCircuitCurrentP1":16,"dynamicCircuitCurrentP2":0,"dynamicCircuitCurrentP3":0}
    ...   
    [lp-1  ] DEBUG 2024/03/24 15:23:25 available power 4596W > 4140W min 3p threshold
    ...
    [easee ] TRACE 2024/03/24 15:23:25 {"dynamicCircuitCurrentP1":16,"dynamicCircuitCurrentP2":16,"dynamicCircuitCurrentP3":16}

Mir ist aufgefallen; das mein selbst gebauter solarlog/modbus kompatibler PV-Datenlogger nur je einen Request beantwortet und dann den Socket schließt. 15 Sekunden wird der zweite Request wiederholt. Kann dies die Berechnung durcheinander bringen?

[image]

GrimmiMeloni, 2024-03-24:
@andig da sollte mal jemand mit Modbus Kenntnis draufschauen. Die Easee hat hier keinen Einfluss.

kai-morich, 2024-03-24:
Hab mein modbus server umgebaut. Mal abwarten ob dies etwas ändert.

andig, 2024-03-24:
Seit https://github.com/evcc-io/evcc/issues/13090#issuecomment-2015522550 sehe ich keine neue Erkenntnis und v.a. auch keine Antwort auf die Frage. Da kann ich nix beitragen.