evcc-io/evcc #13092: Feed-in (Einspeisung) not available | Home Power wrong

dev-da77, 2024-03-22:
### Describe the bug

For a few days I realized that evcc did not show the feed-in (DE: Einspeisung).

PV Setup:

    SMA Home Manager 2
    SMA SBS3.7-10
    SMA STP10.0

In the SunnyPortal and HomeAssistant I see the actual value for the feed-in. But evcc didn't show the value on the UI under feed-in.

Instead the value of usage (DE: Verbrauch) is the sum of actual usage + feed-in. I can see this on the UI and get this sum value via mqtt (Home Power).

Not only the values are incorrect. MIN+PV mode is also not working correctly, because "Home Power" is higher than in reality. This lead to lower load power.

[image]

### Steps to reproduce

1. Home Batterie is full
2. PV creates more energy than needed
3. System feed-in into grid

BUT evcc don't show the feed-in.

Instead the value of usage (DE: Verbrauch) is the sum of actual usage + feed-in.

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.lan # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: *******

interval: 10s # control cycle interval

sponsortoken: *******

telemetry: false

meters:

- name: grid1
  type: custom
  power: # power (W)
    source: mqtt
    topic: hmd/sensor/SMA-Sunny-Home-Manager-2/Active-power-consumption/state
    timeout: 30s
    scale: 1
  energy: # Energy (kWh)
    source: mqtt
    topic: hmd/sensor/SMA-Sunny-Home-Manager-2/Active-power-consumption-counter/state
    timeout: 30s
    scale: 1

- name: pv1
  type: template
  template: sma-hybrid
  usage: pv
  modbus: tcpip
  id: 3
  host: <ip>
  port: 502

- name: battery1
  type: template
  template: sma-sbs-modbus
  usage: battery
  modbus: tcpip
  id: 4
  host: <ip>
  port: 502

chargers:
- name: wallbox1
  type: template
  template: go-e-v3
  host: <ip>

vehicles:
- name: ev1
  type: template
  template: tesla-command
  title: *******
  accessToken: *******
  refreshToken: *******
  vin: *******

loadpoints:
- title: myCHARGER
  charger: wallbox1
  vehicle: ev1
  phases: 3
  mincurrent: 6
  maxcurrent: 16

site:
  title: myHOME
  meters:
    grid: grid1
    pv: pv1
    battery: battery1
  residualPower: 100

mqtt:
  broker: homeassistant:1883
  topic: evcc
  user: mqttuser
  password: *******
```

### Log details

```text
[main  ] INFO 2024/03/22 14:22:50 evcc 0.124.10
[main  ] INFO 2024/03/22 14:22:50 using config file: /etc/evcc.yaml
[main  ] INFO 2024/03/22 14:22:50 starting ui and api at :7070
[db    ] INFO 2024/03/22 14:22:50 using sqlite database: /home/user/.evcc/evcc.db
[mqtt  ] INFO 2024/03/22 14:22:50 connecting evcc-1851429533 at tcp://homeassistant:1883
[mqtt  ] DEBUG 2024/03/22 14:22:50 tcp://homeassistant:1883 connected
[main  ] ERROR 2024/03/22 14:22:51 creating vehicle ev1 failed: cannot create vehicle type 'template': cannot create vehicle type 'tesla': token expired
[lp-1  ] DEBUG 2024/03/22 14:22:51 charge total import: 2864.723kWh
[site  ] INFO 2024/03/22 14:22:53 site config:
[site  ] INFO 2024/03/22 14:22:53   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2024/03/22 14:22:53     grid:      power ✓ energy ✓ currents ✗
[site  ] INFO 2024/03/22 14:22:53     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2024/03/22 14:22:53     battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2024/03/22 14:22:53   vehicles:
[site  ] INFO 2024/03/22 14:22:53     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2024/03/22 14:22:53 loadpoint 1:
[lp-1  ] INFO 2024/03/22 14:22:53   mode:        off
[lp-1  ] INFO 2024/03/22 14:22:53   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2024/03/22 14:22:53   meters:      charge ✓
[lp-1  ] INFO 2024/03/22 14:22:53     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2024/03/22 14:22:53 phase timer inactive
[lp-1  ] DEBUG 2024/03/22 14:22:53 pv timer inactive
[lp-1  ] INFO 2024/03/22 14:22:53 vehicle updated: unknown -> Tesla Model Y (unavailable)
[site  ] DEBUG 2024/03/22 14:22:53 ----
[lp-1  ] DEBUG 2024/03/22 14:22:53 charge power: 0W
[site  ] DEBUG 2024/03/22 14:22:53 pv power: 2343W
[site  ] DEBUG 2024/03/22 14:22:53 battery soc: 100%
[site  ] DEBUG 2024/03/22 14:22:53 battery power: -23W
[site  ] DEBUG 2024/03/22 14:22:53 grid meter: 0W
[site  ] DEBUG 2024/03/22 14:22:53 site power: 77W
[lp-1  ] DEBUG 2024/03/22 14:22:53 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/03/22 14:22:53 charge total import: 2864.723kWh
[lp-1  ] DEBUG 2024/03/22 14:22:53 charger status: B
[lp-1  ] INFO 2024/03/22 14:22:53 car connected
[lp-1  ] DEBUG 2024/03/22 14:22:53 pv timer elapse
[lp-1  ] DEBUG 2024/03/22 14:22:53 pv timer inactive
[lp-1  ] DEBUG 2024/03/22 14:22:53 charge total import: 2864.723kWh
[site  ] DEBUG 2024/03/22 14:23:23 ----
[lp-1  ] DEBUG 2024/03/22 14:23:23 charge power: 0W
[site  ] DEBUG 2024/03/22 14:23:23 pv power: 2303W
[site  ] DEBUG 2024/03/22 14:23:23 battery soc: 100%
[site  ] DEBUG 2024/03/22 14:23:23 battery power: -23W
[site  ] DEBUG 2024/03/22 14:23:23 grid meter: 0W
[site  ] DEBUG 2024/03/22 14:23:23 site power: 77W
[lp-1  ] DEBUG 2024/03/22 14:23:23 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/03/22 14:23:23 charge total import: 2864.723kWh
[lp-1  ] DEBUG 2024/03/22 14:23:23 charger status: B
[site  ] DEBUG 2024/03/22 14:23:53 ----
[lp-1  ] DEBUG 2024/03/22 14:23:53 charge power: 0W
[site  ] DEBUG 2024/03/22 14:23:53 pv power: 2255W
[site  ] DEBUG 2024/03/22 14:23:53 battery soc: 100%
[site  ] DEBUG 2024/03/22 14:23:53 battery power: 439W
[site  ] DEBUG 2024/03/22 14:23:53 grid meter: 18W
[site  ] DEBUG 2024/03/22 14:23:53 site power: 557W
[lp-1  ] DEBUG 2024/03/22 14:23:53 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/03/22 14:23:53 charge total import: 2864.723kWh
[lp-1  ] DEBUG 2024/03/22 14:23:53 charger status: B
```

### What type of operating system are you running?

Linux

### Version

0.124.10

dev-da77, 2024-03-22:
SMA Home Manager geht nur direkt, wenn im gleichen Subnetz (wegen Speedwire Schnittstelle). SMA Komponenten sind bei mir aber in einem anderen Subnetz als EVCC und HOME ASSISTANT. Darum mußte ich den SMA HUB dazwischen nehmen und Grid Power per MQTT bereitstellen.

// Offtopic: _Neu erstellen der Tesla Token hat geholfen, die Fehlermeldung zu beseitigen. Hatte ich schonmal versucht, da hatte es nicht geholfen._

dev-da77, 2024-03-22:
MQTT Explorer:
SMA Werte zeigen die richtigen Werte an - also auch die Einspeisung.
EVCC Werte sind wie auf der UI inkorrekt.

dev-da77, 2024-03-22:
Richtig,

Am Tag ist dieser meistens eh Null, da nichts aus dem Netz gezogen wird.

Von wo liest evcc den Wert für die Einspeisung? Ich gehe davon aus, dass von SMA SUNNY TRIPOWER (STP10.0)?!

mucki12, 2024-03-22:
>Erwartet evcc, dass der Wert positiv (für Bezug) und negativ (für Einspeisung) wird? Das ist bei Active power consumption nicht der Fall.

Als ich grid noch per MQTT übergeben habe war es so: 
Wirkleistung vom Power Meter ist bei mir negativ bei Bezug und positiv bei Einspeisung. Musste ich dann so einbinden:
```
- name: grid
  type: custom
  power: #Power # default value, optionally override # Leistung (W)
    source: mqtt
    topic: huawei/netzbezug
    timeout: 30s
    scale: -1
```

Also ja - evcc erwartet für den Bezug wohl einen positiven Wert.

dev-da77, 2024-03-22:
Ich geh im Moment davon aus, dass ich das Plugin CALC verwenden muss und die beiden Werte Active-power-consumption und Active-power-supply zu einem Grid Power Signal verknüpfen muss, so dass

  Grid Power = Active-power-consumption - Active-power-supply

Bei Netzbezug: Grid Power = Active-power-consumption - 0 = +Netzbezug
Bei Einspeisung: Grid Power = 0 - Active-power-supply = -Einspeisung

Muss nur noch schauen, wie das geht, wenn jemand helfen kann?

dev-da77, 2024-03-22:
Sieht gut aus :-)

- name: grid1
  type: custom
  power: # power (W)
    source: calc
    add:
    - source: mqtt
      topic: hmd/sensor/SMA-Sunny-Home-Manager-2/Active-power-consumption/state
      timeout: 30s
      scale: 1
    - source: mqtt
      topic: hmd/sensor/SMA-Sunny-Home-Manager-2/Active-power-supply/state
      timeout: 30s
      scale: -1