evcc-io/evcc #13093: resetOnDisconnect with mode on Easee is broken

dm82m, 2024-03-22:
### Describe the bug

If you unplug the car the mode and car is not reset. It stays like it was and leads to direct charge if another car is plugged. It worked before so I guess the failure started with current version.

### Steps to reproduce

1. disconnect the car
2. check the settings in evcc

### Configuration details

```yaml
Mode: off
```

### Log details

```text
No logs, request if you need something
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.124.10

dm82m, 2024-03-22:
Nein. Ich hab auch nichts an der config geändert und bisher hat alles funktioniert. Also unveränderte config und der bug muss mit einem der letzten releases gekommen sein.

VolkerK62, 2024-03-22:
>mit einem der letzten releases gekommen sein.

da wäre gut zu wissen, mit welchem.
Wenn du jetzt wieder zurückgehst, funktioniert wieder alles wie gewohnt?

dm82m, 2024-03-22:
Ich kann ja in Home Assistant leider kein add-on downgrade machen. Kann es also leider auch nicht rekonstruieren…

dm82m, 2024-03-22:
@GrimmiMeloni kannst du das bei dir mit der easee evtl. Mal reproduzieren?

closed 2024-03-22 by andig

dm82m, 2024-03-22:
@andig ist das schon gefixt in der aktuellen nightly oder warum close?

andig, 2024-03-22:
> NOTE Due to amount of issues raised we will close incomplete issues (unclear description what the issue actually is, missing log file, unsupported platform) without further comments.

GrimmiMeloni, 2024-03-22:
@dm82m bitte teile mal deine Konfiguration. 
Bonusfrage: hast Du zwischen abstecken und anstecken mindestens ein Interval abgewartet?

GrimmiMeloni, 2024-03-23:
Ok, 1,5h sollten ausreichen. 😅 
Auch die Konfig sieht auf den ersten Blick richtig aus. Teile bitte mal ein Debug Log das den Fehler zeigt.

dm82m, 2024-03-23:
Ich sagte ja schon, dass das mit der identische findig schon funktioniert hat. Was soll ich auf debug Level stellen? Dann mach ich das, auch wenn’s mir Home Assistant addon wieder eine Qual wird an die Logs zu kommen.