evcc-io/evcc #13094: Victron Batterylife SoC Limit wird beim Schnellladen verändert

detlefh68, 2024-03-22:
### Describe the bug

Hallo,

ich habe das evcc seit 2 Tagen installiert und teste die Ladung. Leider habe ich nur eine sehr kleine PV Einspeisung und das Wetter ist schlecht, daher habe ich mich auf das Schnell laden konzentriert.

Als Batterie Speicher Speicher für das Haus habe ich einen Victron MultiPlus-II GX mit einer 4,8 KwH Batterie. Die soll auch das Haus versorgen und ist viel zu klein um ein Auto zu laden. Da ist evcc genau richtig, es kann beim Laden des Autos das Entladen der Batterie unterbinden. Das funktioniert soweit auch.
Doch leider wird beim Starten des Schnell Ladens der Batterielife SoC Wert verändert. Es war auf 50% und evcc hat diesen dann auf 76% eingestellt. Beim Beenden des Ladevorgangs ist auch der verstellte Wert nicht wieder auf den ursprünglichen Wert zurück geändert worden. 
Ich nehme an, dass das Limit beim Ladestart auf den aktuellen Doc Wert der Batterie gestellt wird (dieser war etwas über 70%). 

Zudem wird beim Laden auch der der "ESS Batterie Life State" auch auf "Discharge disabled" gestellt und nicht wieder freigegeben.

[image]

Die Batterie Einstellungen:

[image]

Was kann ich einstellen, damit der Wert wieder auf den ursprünglichen Wert (vor dem Ladevorgang) gestellt und das Entladen der Batterie wieder freigegeben wird? 

Über eine Hilfe würde ich mich sehr freuen!

Gruß 
detlefh68 

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

interval: 30s # control cycle interval

# trial token, valid until 2024-03-23
sponsortoken: 

# unique installation id
plant: 

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
# sponsortoken:

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
# telemetry: true

log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

# modbus proxy for allowing external programs to reuse the evcc modbus connection
# each entry will start a proxy instance at the given port speaking Modbus TCP and
# relaying to the given modbus downstream device (either TCP or RTU, RS485 or TCP)
modbusproxy:
  #  - port: 5200
  #    uri: solar-edge:502
  #    # rtu: true
  #    # readonly: true # use `deny` to raise modbus errors

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for documentation see https://docs.evcc.io/docs/devices/meters
meters:
  - name: grid1 
    type: template
    template: shelly-3em 
    usage: grid  
    host: 192.x.x.x
  - name: pv1
    type: template
    template: hoymiles-opendtu
    usage: pv
    host: 192.x.x.x
  - name: battery1
    type: template
    template: victron-energy 
    usage: battery  
    host: 192.x.x.x 
    port: 502  
    capacity: 4.8
    minsoc: -1
#    minsoc: 20  
#    maxsoc: 95  
    
# charger definitions
# name can be freely chosen and is used as reference when assigning charger to vehicle
# for documentation see https://docs.evcc.io/docs/devices/chargers
chargers:
  - name: wallbox
    type: template
    template: smaevcharger
    host: 192.x.x.x
    user: evcc 
    password: evcc!SMA#aW7 
  - name: ebike
    type: template
    template: tasmota
    host: 192.x.x.x # IP-Adresse oder Hostname
    user: # Standard-User ist admin (optional)
    password: # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen) (optional)
    channel: 1 # Schaltkanal (1-8)
    standbypower: 15 # Leistung oberhalb des angegebenen Wertes wird als Ladeleistung gewertet (optional) 
    integrateddevice: true
    icon: bike

# vehicle definitions
# name can be freely chosen and is used as reference when assigning vehicle to loadpoint
# for documentation see https://docs.evcc.io/docs/devices/vehicles
vehicles:
  - type: template
    template: vw 
    title: ID.3  
    icon: car  
    user:  
    password:   
    vin:  
    capacity: 58  
    phases: 3  
    cache: 15m  
    mode: off  
    minCurrent: 6  
    maxCurrent: 16  
    timeout: 10s  
    name: id3

# site describes the EVU connection, PV and home battery
site:
  title: Mein Zuhause
  meters:
    grid: grid1
    pv: 
      - pv1 
    battery:
      - battery1
  residualPower: 100 # 100W Netzbezug im PV Modus erlaubt

# loadpoint describes the charger, charge meter and connected vehicle
loadpoints:
  - title: Garage 
    charger: wallbox
    vehicle: id3 
    mode: off 
    enable:
      # einschalten, wenn 1 Minute lang mindestens 2000 W Überschuss vorhanden ist
      delay: 1m
      threshold: -800
    disable:
      # ausschalten, wenn 30 Minuten lang mehr als 2000 W aus dem Netz bezogen werden
      delay: 30m
      threshold: 2000    
  - title: E-Bikes
    charger: ebike
    mode: off 
#    mincurrent: 0.5 # 9A*220V=1980W minimum charge current (default 6A)
#    maxcurrent: 4 # 10A*220V=2200W maximum charge current (default 16A)
#    phases: 1

# tariffs are the fixed or variable tariffs
tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.3784
  feedin:
    type: fixed
    price: 0.0
  co2:
    type: grünstromindex
    zip: xxxxx

# mqtt message broker
mqtt:
  broker: mqtt.home.lan:1883
  topic: evcc # root topic for publishing, set empty to disable
  clientid: evcc
  user: 
  password: 

# influx database
influx:
  # url: http://localhost:8086
  # database: evcc
  # user:
  # password:

# eebus credentials
eebus:
  # uri: # :4712
  # interfaces: # limit eebus to specific network interfaces
  # - en0
  # certificate: # local signed certificate, required, can be generated via `evcc eebus-cert`
  #   public: # public key
  #   private: # private key

# push messages
messaging:
  events:
    start:
      title: Ladevorgang gestartet
      msg: Ladevorgang wurde mit Modus "${mode}" gestartet
    stop:
      title: Ladevorgang beendet
      msg: ${chargedEnergy:%.1fk}kWh in ${chargeDuration} geladen
    connect:
      title: Fahrzeug verbunden
      msg: Ladepunkt ${loadpoint}
    disconnect:
      title: Fahrzeug getrennt
      msg: Ladepunkt ${loadpoint} nach ${connectedDuration} getrennt
  services:
```

### Log details

```text
...ist beim HA Addon nur eingeschränkt möglich
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

v0.124.10

andig, 2024-03-22:
Ein zurücksetzen geht nicht, aber du kannst den Standardwert konfigurieren. Siehe

    evcc configure

/cc @VolkerK62 sollen wir die Batteriesteuerung per `minsoc` nochmal dokumentieren?

closed 2024-03-22 by andig

detlefh68, 2024-03-22:
Bei meiner ersten Testinstalltion unter Proxmox habe ich folgendes erhalten....

  - name: battery1
    type: template
    template: victron-energy 
    usage: battery  
    host: 192.x.x.x 
    port: 502  
    capacity: 4.8
    minsoc: -1
#    minsoc: 20  
#    maxsoc: 95  
    
minsoc und maxsoc waren nach "configure" eingetragen => damit ist das gleiche passiert. Bat Life SoC Limit verstellt und bat Life State auf disabled gestellt. Daher habe ich für den 2.Test minsoc auf -1 gestellt => ändert nichts :-( 
Wenn diasabled wird, kann doch auch wieder enabled werden, oder muss ich das nach jedem Schnell Laden wieder manuell  enablen?

andig, 2024-03-22:
>Wenn diasabled wird, kann doch auch wieder enabled werden, oder muss ich das nach jedem Schnell Laden wieder manuell enablen?

Kann er aktuell nicht. Du kannst *entweder*  die aktive Batteriesteuerung deaktivieren *oder* den `minsoc` auf die gewünschten 50% stellen.

detlefh68, 2024-03-22:
ok ich stellen für den nächsten Test minsoc im evcc.yaml bei der Batterie ein. 
Dann wird nach dem Laden das SoC Limit auf diesen Wert eingestellt?
Wie ich die aktive Batterie Steuerung dektiviere ist mir im Moment nicht ganz klar. Ich weis nicht wo und wie ich das aktiviere. 
Ist halt blöd, dass es von Hand verstellt werden muss.

andig, 2024-03-22:
Muss es nicht. Du darfst sie nur nicht einschalten.

VolkerK62, 2024-03-22:
@andig 
>/cc @VolkerK62 sollen wir die Batteriesteuerung per minsoc nochmal dokumentieren?

Wegen Nichtbetroffenheit verfolge ich das Thema nur am Rande.
Ist `minsoc` ein Victron spezifisches Thema? Dann könnte man das im Template für die Doku "sichtbar" machen?

andig, 2024-03-22:
Das betrifft tatsächlich alle Batterien, bei denen hold/charge über den `limitsoc` anstatt über einen Mode gesteuert wird:

[image]

detlefh68, 2024-03-22:
@andig 

du meinst hier deaktivieren?

[image]

andig, 2024-03-22:
Genau. Der Haken kommt nicht von alleine ;)

detlefh68, 2024-03-22:
Danke 🙂

Wenn ich es richtig verstanden habe...Die Sperre zum Entladen wird über (beim Victron) Batterylife SoCLimit realisiert. D.h. es wird ein Wert größer aktueller SoC gesetzt und der von mir eingestellte minsoc (z.B. 20%) in der evcc.yaml beim Beenden gesetzt. 

Da der Victron ja immer nach Ladezyklus diesen Wert täglich verändert, wir durch das Auto Laden immer ein fix Wert gesetzt. Dann müsste ich ja abhängig von der Wetterlage und Jahreszeit diesen Wert in der evcc.yaml anpassen?

Wäre es super, wenn evcc den aktuelle limit Wert beim Beginn des Ladens liest und anschließend wieder setzt....

Habe ich den Mechanismus der Ladesperre in etwa verstanden?😬

premultiply, 2024-03-22:
Die bessere Variante wäre auch hier ein Weg ohne die Manipulation von SoC-Schwellen.

detlefh68, 2024-03-23:
Ich habe jetzt minsoc auf 20 gestellt. Beim Starten des Ladenvorgangs wir minsoc auf den aktuellen soc Wert der Batterie geändert (heute auf 62%) und das soc limit auf Aktuelle soc+5%, also 67%.

Nach dem Laden wird auf den von mir auf minsoc gestellten Wert aus der evcc.yaml gestellt. doch leider bleibt des aktive soc limit auf 67%. 

Das hat zur Folge, das die Hausbatterie immer nur auf den Wert entladen wird (soc+5%), der vor dem Autoladen war.

=> minsoc funktioniert, aber das Änerden des aktiven SoC Limit im Victron ist schlecht. 

ist es nicht möglich das SoC Limit der Batterie von evcc nicht anzutasten?

andig, 2024-03-23:
> Nach dem Laden wird auf den von mir auf minsoc gestellten Wert aus der evcc.yaml gestellt. doch leider bleibt des aktive soc limit auf 67%.

@philipptrenz hast Du eine Ahnung was hier los ist? minsoc vs. "aktives soc Limit"?

detlefh68, 2024-03-24:
Der Victron hat zwei SoC Grenzen,  minsoc und Aktives SoC Limit....

[image]

...minsoc ist der Wert, der beim Laden auf den den aktuellen SoC Wert gestellt wird, damit das Entladen gesperrt ist.
Dieser wird nach dem Laden (Auto) wieder auf den im der evcc.yaml definierten Wert gestellt. Das funktioniert auch gut.

Der zweite Wert "Aktives SoC Limit" ist der Wert der beim Batterielife Betrieb vom Victron verstellt wird. Wird die Batterie am Tag auf annähernd 100% geladen, singt der Wert um 5%. Wird die Batterie bei schlechtem Wetter nur wenig geladen, also unter 100%, steigt der Wert täglich wieder um 5%. Somit soll vermieden werden, dass sich die Batterie über eine längere Zeit im sehr hohen oder tiefen Ladezustand befindet.

Durch das Anpassen von minsoc beim Start des Ladevorgangs (Auto) wird das "aktive SoC Limit" mit verstellt (vermutlich darf im Victron minsoc nicht größer als das SoC Limit sein).
Das hast in einer Zeit (Sommer), in der das Wetter gut ist und die Batterie ordentlich geladen wird, ist diese Limit sehr niedrig. Letzen Sommer war em immer fast auf 5%, Da heist die Batterie wurde nachts Immer entladen und den Grundbadarf des Hauses bis zum Morgen gedeckt.

Wenn ich dann am Nachmittag das Auto Lade und der die Batterie einen hohen Ladezustand hat, wird mein SoC Limit auch mit angehoben (macht öffentlich der Virctron selbstständig) und die Batterie nur bis zu dem Limit entladen. 

Man müsste dann immer nach einem Ladevorgang des Autos überprüfen wie hoch das Limit war und manuell wieder anpassen.

Daher kam meine Frage, ob evcc nicht auch diesen Wert wieder auf den alten Wert zurückstellt, wenn ein Ladezyklus des Autos beendet ist.

Ich hoffe ich habe meine Gedanken einigermaßen verständlich erklärt.

Viele Grüße
detlefh68

andig, 2024-03-24:
> Daher kam meine Frage, ob evcc nicht auch diesen Wert wieder auf den alten Wert zurückstellt, wenn ein Ladezyklus des Autos beendet ist.

Das geht in den Templates. Wir könnten es mit auf `minsoc` stellen?

detlefh68, 2024-03-24:
Im sonnigen Sommer wäre das kein Problem, bei eine schlechten Wetterlage eher schlecht, da das Limit ja dann auch auf einen Kleinen Wert gestellt wird und die Batterie  wenig geladen stehen bleibt, weil die Sonne fehlt.

Super wäre beim Start lesen und dann wieder zurück stellen...fall so etwas beim "victron-energy" Temlate möglich ist.

Das Stellen auf minsoc würde das manuell nachstellen bei schlechter "Sonne" nach sich ziehen. Dieser Automatisch Batterie Modus beim Victron ist schon komfortabel :-)

andig, 2024-03-24:
> Super wäre beim Start lesen und dann wieder zurück stellen...fall so etwas beim "victron-energy" Temlate möglich ist.

Oben fehlte das "nicht". Nein und nein.

detlefh68, 2024-03-24:
Das ist schade :-(

Wenn das Wiederherstellen des Limits nicht möglich ist, wäre ein stellen auf minsoc besser.

Wäre das ein Sache für die Zukunft, feature request?

Dann muss ich versuchen dan Ladevorgang des Autos vom evcc im Homeassistant zu erfassen und dort in einer Automatisierung den Limit Wert Wiederherzustellen. Wird das Starten eines Ladevorgangs irgendwie per MQTT mitgeteilt?

Ich habe evcc ja erste ein paar Tage installiert und teste es und es gefällt mir soweit sehr gut.
So wie es im Moment die Hausbatterie beeinflußt in es nicht schlecht und mit Vorsicht und Korrigieren zu nutzen.
Ich hoffe, dass die Interaktion mit dem Victron System irgendwann besser läuft....

detlefh68, 2024-03-24:
Das Verhalten der Victron Batterie scheint es nicht nur bei mir zu geben.

In diesen Beiträgen scheint es sich um das gleiche Verhalten zu handeln
#11286 
#11252

Wenn Batterielife am Vitron an ist, müsste es die Veränderungen am minsoc ignorieren.

Es ist nicht so glücklich das Sperren der Batterieentladung über minsoc durchzuführen.

andig, 2024-03-24:
Du kannst gerne einen alternativen Vorschlag machen. Das ist der der umgesetzt ist...

detlefh68, 2024-03-25:
Ich habe die Batterie jetzt erst mal so eingestellt 

[image]

Und probiere erst nochmal weiter wie es sind bei den nächsten Ladevorgängen des Autos verhält. 

Das Limit habe ich versucht über eine Homeassistant Automation beim Start zu lesen und wieder zurückzuschreiben. 

Eine Frage habe ich noch...
Wenn ich minsoc in der Konfiguration auf -1 stelle, wird dann nichts an den Victron übertragen?

Trotzdem erst mal Danke für deine Hilfe 😊

tuxedo0801, 2025-10-15:
Servus @detlefh68 ,

der Thread hier ist zwar schon etwas älter, aber: Hast du einen Weg gefunden bei dem Battery Life aktiv bleiben kann? Hab das Problem bei meiner Anlage in Verbindung mit EVCC auch entdeckt. 
EIgentlich hatte ich einen "battery life min soc" von 10% oder so. Dank EVCC ist der nun bei >80%... trotz eigentlich ausreichend schönen Wetters. 

Die Battery Life Funktion ist schon was feines bei Victron... Gerade in der dunkleren Jahreszeit. 

Grüble schon ob man das hier nicht anders steuern könnte. Stecke aber noch nicht ausreichend tief in der EVCC Thematik drin, bin auch noch am experimentieren...

detlefh68, 2025-10-16:
Hallo tuxedo0801,

ich speichere der Wert in einer er HA Automation das Limit zwischen und schreibe den Wert nach der Ladesitzung zurück.