evcc-io/evcc #13101: Keba P30 - Überschussladen + RFID: Automatisiertes Laden für 10sek.

Magoo2, 2024-03-22:
Hallo zusammen,
bei der Keba P30 gibt es das Problem, dass Überschussladen nicht automatisch startet. Ist die Keba mit einer RFID gesichert, wird der Token nicht lang genug von der Box gespeichert und es kann später kein automatischer Start der Ladung erfolgen.
Der Parameter "Identifier" hilft leider auch nicht. 
Details im Thread: https://github.com/evcc-io/evcc/discussions/13014

Feature request: 
Die Keba soll für ein paar Sekunden nach den anstecken auf minimaler Ladung starten und dann die Ladung beenden.

horstvogel, 2024-03-22:
Keba P30 mit RFID funktioniert bei mir 1a. 

Vor dem Einstecken die Karte vorhalten und dann läuft das.

Magoo2, 2024-03-22:
Servus,

hast du eine P30 X? Gibt mehr Leute bei denen das im Modus PV nicht funktioniert.

67-Peter, 2024-03-23:
Also bei mir läuft/lief die Keba 30x in Verbindung mit RFID und dem Template "keba" (udp) super gut. Das beste daran war , das über diesen Mode meine RFID an die Box übertagen wurde (Keba hängt bei mir in der Garage und ich bin der einzige mit eine E-Auto). Ich braucht meine RFID Karte gar nicht mehr :-). Leider wird in diesem Modus kein Phasenumschalten unterstützt.
Ich bin darum diesen Monat auf das Template "keba-modbus" umgestiegen, leider habe ich damit ähnliche Probleme wie du.
Eben hatte ich auch wieder jede Menge Überschuss, musste aber erstmal in die Garage wieder die Karte vor die Wallbox halten.

Ich werde leider wieder zu "keba" (UDP) wechseln und auf die Phasenumschaltung verzichten.
Ich habe aber auch in einem Kommentar von [andig](https://github.com/andig) gelesen, das "keba" (udp) irgendwann eingestellt werden soll.

Also wenn du keine Phasenumschaltung hast , nimm das "keba" Template  mit dem Parameter "rfid".

Magoo2, 2024-03-23:
@67-Peter Danke für die Info, dass ist richtig interessant.
Tatsächlich kann ich UDP nicht nutzen, weil die Box im WLAN hängt. Ursprünglich hatte ich UDP aber da gab es immer wieder Fehler bei der Kommunikation, so dass es nicht nutzbar war. 
Das Thema über Modbus zu lösen, wäre das Ziel. Wie gesagt, die Box muss nur einmal kurz den Ladevorgang starten und dann beenden. Sobald er einmal gestartet ist, vergisst die Box den Token nicht mehr und alles läuft.