evcc-io/evcc #13105: charge total import: timeout & charge rater: charge meter error timeout

hp-8472, 2024-03-23:
### Describe the bug

Wir haben einen SUNGROW SH10RT zusammen mit einer Wallbox Pulsar Plus bei uns aufgestellt.
Nach dem Start scheint alles gut zu funktionieren, aber schon nach gut zwei Minuten bekomme ich immer wieder charger timeout und ab da ständig.
[image]

Ich habe eine ganz ähnliche Problematik hier bei den [Diskussionen ](https://github.com/evcc-io/evcc/discussions/10296)gefunden, nur leider gibt es dabei keinen echten Workaround.

### Steps to reproduce

1. sudo systemctl start evcc

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: trace
levels:
  cache: error

# unique installation id
plant: fad75693ec47bd2c3ea7a19031d64cbcbb1308083be527e330ece9c2421a0882

interval: 30s # control cycle interval

meters:
- type: template
  template: sungrow-hybrid
  id: 1
  host: 192.168.178.97
  port: 502
  usage: grid
  modbus: tcpip
  timeout: 1m
  name: grid1
- type: template
  template: sungrow-hybrid
  id: 1
  host: 192.168.178.97
  port: 502
  usage: pv
  modbus: tcpip
  timeout: 1m
  name: pv2
- type: template
  template: sungrow-hybrid
  id: 1
  host: 192.168.178.97
  port: 502
  usage: battery
  modbus: tcpip
  timeout: 1m
  name: battery3

chargers:
- type: template
  template: pulsarplus
  timeout: 2m
  name: wallbox4

loadpoints:
- title: Rechts
  charger: wallbox4
  mode: pv

[... 387 lines cut ...]
[modbus] TRACE 2024/03/23 00:47:43 modbus: recv 00 41 00 00 00 05 01 04 02 00 0c
[modbus] TRACE 2024/03/23 00:47:43 modbus: send 00 42 00 00 00 06 01 04 32 e2 00 02
[modbus] TRACE 2024/03/23 00:47:44 modbus: recv 00 42 00 00 00 07 01 04 04 0e 9b 00 00
[modbus] TRACE 2024/03/23 00:47:44 modbus: send 00 43 00 00 00 06 01 04 32 de 00 01
[modbus] TRACE 2024/03/23 00:47:44 modbus: recv 00 43 00 00 00 05 01 04 02 01 a9
[site  ] DEBUG 2024/03/23 00:47:44 battery soc: 43%
[site  ] DEBUG 2024/03/23 00:47:44 battery power: 729W
[modbus] TRACE 2024/03/23 00:47:44 modbus: send 00 44 00 00 00 06 01 04 32 d1 00 02
[modbus] TRACE 2024/03/23 00:47:44 modbus: recv 00 44 00 00 00 07 01 04 04 00 07 00 00
[site  ] DEBUG 2024/03/23 00:47:44 grid meter: -7W
[modbus] TRACE 2024/03/23 00:47:44 modbus: send 00 45 00 00 00 06 01 04 32 ec 00 02
[modbus] TRACE 2024/03/23 00:47:44 modbus: recv 00 45 00 00 00 07 01 04 04 d8 5b 00 00
[site  ] DEBUG 2024/03/23 00:47:44 site power: 732W
[lp-1  ] ERROR 2024/03/23 00:47:44 charge total import: timeout
[lp-1  ] DEBUG 2024/03/23 00:47:44 charger status: B
[lp-1  ] DEBUG 2024/03/23 00:47:44 pv charge current: 0A = 0A + -1.06A (732W @ 3p)
```

### What type of operating system are you running?

Linux

### Version

0.124.10

andig, 2024-03-23:
> [lp-1  ] ERROR 2024/03/23 00:47:13 charge meter: timeout

Deine Pulsar liefert keine Zählerwerte.

hp-8472, 2024-03-23:
OK, kann ich etwas tun?

andig, 2024-03-23:
Probier mal Sufu welche Erfahrungen andere Anwender mit der Box haben. Ich hab das Gerät nicht.

b0e, 2024-03-23:
Hi, 

ich hatte ein ähnliches Verhalten mit meiner ABL ehm2 und die gleichen Fehlermeldungen. Meine Wallbox sendet nur alle 300 Sekunden MeterValues, z.B.
```
[ocpp  ] TRACE 2024/03/23 11:54:47 received JSON message from ABL_2W224000650: [2, "35188287", "MeterValues" ,{"connectorId":1,"transactionId":18,"meterValue":[{"timestamp":"2024-03-23T10:54:46.767+00:00","sampledValue":[{"value":"19091.351","context":"Sample.Periodic","format":"Raw","measurand":"Energy.Active.Import.Register","location":"Outlet","unit":"kWh"},{"value":"7208.7","context":"Sample.Periodic","format":"Raw","measurand":"Power.Active.Import","location":"Outlet","unit":"W"}]}]}]
```

evcc erwartet wohl, dass spätestens nach 2 Minuten wieder Werte kommen, weshalb nach 2min immer die Fehler kommen. Lass doch evcc mal längere Zeit laufen und such in den Logs nach den MeterValues (siehe oben). Sollte deine Wallbox auch nur alle 5 oder 10 Minuten Daten senden, kannst du einfach den timeout am Loadpoint entsprechend hochstellen.

Was nicht ganz dazu passt: Deine Wallbox sagt sie sendet im Interval von 5 Sekunden die Samples (MeterValueSampleInterval), aber vielleicht hält sich deine WB ja nicht dran?!

Ich hoff das führt nicht in eine falsche Richtung.

Lg, Achim

hp-8472, 2024-03-23:
Vielen Dank für die Hinweise

Damit habe ich es jetzt lösen können, und es ging wie folgt:
An der Wallbox habe ich die OCPP Funktionalität deaktiviert, nach dem Neustart habe ich dann festgestellt dass die Wallbox gesperrt gewesen ist, warum auch immer... 
Daraufhin habe ich die Wallbox entsperrt und die OCPP Verbindung zum evcc wiederherstellen lassen.
Seit dem läuft es ohne Probleme :)

closed 2024-03-23 by hp-8472

andig, 2024-03-23:
Wunderbar und danke an @b0e für den Support!