evcc-io/evcc #13110: MQTT charger type custom - chargeLimitSet

aichingerk, 2024-03-23:
### Describe the bug

Hallo, wenn der Charger von Hand mit der Ladestromstärke verstellt worden ist, wird bei Schnell Laden oder Günstiges laden keine neue Information über MQTT an den Charger gesendet. 

### Steps to reproduce

1. Stromstärke auf 6A gestellt am Charger
2. EVCC auf schnell laden stellen
3. keine neue Information von der Ladestromstärke von evcc über mqtt an den Charger
4. somit wird nur mit 6A geladen statt mit 16A

### Configuration details

```yaml
network:
  schema: http
  #  host: evcc.local
  port: 7070

interval: 30s # control cycle interval

log: info
levels:
  #site: debug
  #cache: error
  #db: error
  #lp-1: debug
  #Cupra_Born: debug
  #mqtt: debug
  #charger: debug

meters:
  # Amis Zähler selbst, aber test mit GoodWe
  #- name: Amis-Normal
  #  type: custom
  #  power:
  #    source: mqtt
  #    topic: amis/normal
  #    jq: '."1.7.0"'
  # timeout: 30s # don't accept values older than timeout
  #   scale: # 0.001 # floating point factor applied to result, e.g. for Wh to kWh conversion

  - name: GoodWe-Grid
    type: template
    template: goodwe-hybrid
    usage: grid
    modbus: tcpip
    id: 247
    host: 192.168.4.60 # Hostname
    port: 502 # Port

  - name: GoodWe-PV
    type: template
    template: goodwe-hybrid
    usage: pv
    # Modbus TCP
    modbus: tcpip
    id: 247
    host: 192.168.4.60 # Hostname
    port: 502 # Port

  - name: GoodWe-Batterie
    type: template
    template: goodwe-hybrid
    usage: battery
    # Modbus TCP
    modbus: tcpip
    id: 247
    host: 192.168.4.60 # Hostname
    port: 502 # Port
    capacity: 6.6 # in kWh

chargers:
  - name: NRGKick
    type: custom
    status: # charger status A..F status A..F: https://evsim.gonium.net/#der-control-pilot-cp
      source: mqtt
      topic: evcc/nrgkick/chargerStatusValue
      timeout: 1m # don't accept values older than timeout
    enabled: # charger enabled state (true/false or 0/1)
      source: mqtt
      topic: evcc/nrgkick/chargingStateValue
      timeout: 1m
    enable: # set charger enabled state (true/false or 0/1)
      source: mqtt
      topic: evcc/nrgkick/chargeStartStopSet
      payload: ${enable:%d}
      timeout: 1m
    maxcurrent: # set charger max current (A)
      source: mqtt
      topic: evcc/nrgkick/chargeLimitSet
      payload: ${maxcurrent:%d}
      timeout: 1m
      scale: 1

vehicles:
  - name: Cupra_Born
    type: template
    template: cupra
    title: Born 50
    capacity: 58
    user: xxxx # user
    password: xxxx # password
    vin: xxxx # WREN...
    cache: 5m

site:
  title: HomeAK # display name for UI
  meters:
    grid: GoodWe-Grid
    pv: GoodWe-PV
    battery: GoodWe-Batterie
  residualPower: 50 # https://docs.evcc.io/docs/reference/configuration/site/#residualpower

loadpoints:
  - title: Garage # display name for UI
    charger: NRGKick # Wallbox Referenz
    vehicle: Cupra_Born # Referenz auf Standardfahrzeug
    mode: pv # charge mode (off, now, minpv, pv)
    enable: # pv mode enable behavior
      delay: 1m # threshold must be exceeded for this long
      threshold: 0 # grid power threshold (in Watts, negative=export). If zero, export must exceed minimum charge power to enable
    disable: # pv mode disable behavior
      delay: 2m # threshold must be exceeded for this long
      threshold: 0 # maximum import power (W)
    guardDuration: 1m # switch charger contactor not more often than this (default 5m)

# tariffs are the fixed or variable tariffs
tariffs:
  grid:
    type: awattar
    region: at
    charges: 0.012 # optional, additional charges per kWh
    tax: 0.2 # optional, additional tax (0.1 for 10%)

# mqtt message broker
mqtt:
  broker: localhost:1883
  topic: evcc # root topic for publishing, set empty to disable
  user: xxxx
  password: xxxx

# push messages
# Message config using evcc go-text-template rendering, evcc variables and sprig-functions
messaging:
  events:
    start: # charge start event
      title: Charge of {{.vehicleTitle}} started
      msg: |
        Wallbox {{.title}} started charging {{.vehicleTitle}} in {{ toString .mode | upper }} mode.
        --------------------------
        evcc Status {{printf `(%d-%02d-%02d %02d:%02d:%02d)` now.Year now.Month now.Day now.Hour now.Minute now.Second}}
        Netz-Leistung: {{round (divf .gridPower 1000) 3 }} kW
        Solar-Leistung: {{round (divf .pvPower 1000) 3 }} kW
        Eigenverbrauch: {{round (divf .homePower 1000) 3 }} kW
        {{if .batteryConfigured}}Batteriespeicher-Status: {{round (divf .batteryPower 1000) 3 }} kW ({{.batterySoc }} %){{end}}
    stop: # charge stop event
      title: Charge of {{.vehicleTitle}} finished
      msg: |
        Wallbox {{.title}} finished charging {{.vehicleTitle}} 
        with {{round (divf .chargedEnergy 1000) 2 }} kWh in {{.chargeDuration}}.
        --------------------------
        evcc Status {{printf `(%d-%02d-%02d %02d:%02d:%02d)` now.Year now.Month now.Day now.Hour now.Minute now.Second}}
        Netz-Leistung: {{round (divf .gridPower 1000) 3 }} kW
        Solar-Leistung: {{round (divf .pvPower 1000) 3 }} kW
        Eigenverbrauch: {{round (divf .homePower 1000) 3 }} kW
        {{if .batteryConfigured}}Batteriespeicher-Status: {{round (divf .batteryPower 1000) 3 }} kW ({{.batterySoc }} %){{end}}
    connect: # vehicle connect event
      title: "{{.vehicleTitle}} connected on wallbox {{.title}}"
      msg: |
        {{.vehicleTitle}} connected on wallbox {{.title}} at {{round (divf .pvPower 1000) 2 }} kW PV.
        --------------------------
        evcc Status {{printf `(%d-%02d-%02d %02d:%02d:%02d)` now.Year now.Month now.Day now.Hour now.Minute now.Second}}
        Netz-Leistung: {{round (divf .gridPower 1000) 3 }} kW
        Solar-Leistung: {{round (divf .pvPower 1000) 3 }} kW
        Eigenverbrauch: {{round (divf .homePower 1000) 3 }} kW
        {{if .batteryConfigured}}Batteriespeicher-Status: {{round (divf .batteryPower 1000) 3 }} kW ({{.batterySoc }} %){{end}}
    disconnect: # vehicle connected event
      title: "{{.vehicleTitle}} disconnected of wallbox {{.title}}"
      msg: |
        {{.vehicleTitle}} disconnected of wallbox {{.title}} after {{.connectedDuration}}.
        --------------------------
        evcc Status {{printf `(%d-%02d-%02d %02d:%02d:%02d)` now.Year now.Month now.Day now.Hour now.Minute now.Second}}
        Netz-Leistung: {{round (divf .gridPower 1000) 3 }} kW
        Solar-Leistung: {{round (divf .pvPower 1000) 3 }} kW
        Eigenverbrauch: {{round (divf .homePower 1000) 3 }} kW
        {{if .batteryConfigured}}Batteriespeicher-Status: {{round (divf .batteryPower 1000) 3 }} kW ({{.batterySoc }} %){{end}}
```

### Log details

```text
ist nachvollziehbar
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.124.10

andig, 2024-03-23:
> wenn der Charger von Hand mit der Ladestromstärke verstellt worden ist

Die Lösung wäre doch, das nicht zu tun?

aichingerk, 2024-03-23:
OK, baue meine anderen Automatisierungen so um, dass der chargeLimitSet nicht verändert wird und beobachte es, ob es wirklich an dem liegt. Cool, wäre es halt schon gewesen, wenn man eine Meldung geschickt bekommt Lade schnell und MQtt einmal auf max Load gesetzt werden würde.

closed 2024-03-23 by aichingerk

andig, 2024-03-23:
Ist immer blöd wenn zwei Steuerungen gegeneinander arbeiten...