evcc-io/evcc #13130: UI Problem nach update auf 0.124.10 - evcc als App auf dem Home-Bildschirm eines iPhone 15 - Oberer Zeilen verdeckt und Burgermenü nicht nutzbar

wedojan, 2024-03-24:
### Describe the bug

Hallo Zusammen,
ich benutze evcc als "App" auf dem Home-Bildschirm des iPhone. 
Man hat die evcc-Website sozusagen als eigenes Symbol auf dem Home-Bildschirm.
(So gehts: URL im Safari öffnen -> auf "Teilen"-Symbol klicken ->  Zum Home-Bildschirm")

Seit dem Update auf 0.124.10 wird nun auf dem iPhone 15 der obere Teil von der Urzeit, "apples dynamic island" (der schwarze Bereich) und der Verbindungsstatus überlagert. (Siehe Bild) 
Auch kann man in diesem Bereich nicht auf das Burger-Menü tippen.

[image]

Öffnet man die Seite direkt im, so wird alles korrekt dargestellt.

An dieser Stelle noch einmal einen **großen Dank** für dieses tolle Projekt an die engagierten Programmierer.
Ihr leistet damit einen großen Beitrag für Gesellschaft und Umwelt!

Gruß wedojan

### Steps to reproduce

1. Auf iphone 15: evcc-url im Safari öffnen -> auf "Teilen"-Symbol klicken ->  Zum Home-Bildschirm"
2. Auf dem Home-Bildschirm die "evcc-App" öffnen

### Configuration details

```yaml
.
```

### Log details

```text
.
```

### What type of operating system are you running?

Docker container

andig, 2024-03-24:
Ist im Nightly behoben

closed 2024-03-24 by andig

wedojan, 2024-03-24:
Vielen Dank für die schnelle Antwort. 
Wie in #12941 beschrieben, genügte das Löschen und Neuanlegen der evcc-App-Kachel auf dem Home-Bildschirm

grusslev, 2024-08-01:
Funktioniert einwandfrei auf meinem iPhone SE.