evcc-io/evcc #13136: Add option to obtain vehicle SOC through car manufacturer API

andreas-bulling, 2024-03-24:
The SOC calculation done by evcc does not match the one I can get through the Hyundai/Kia app/API. This is despite the fact that I have configured the battery size correctly. As a result, evcc stops the charging prematurely.

It'd be nice if evcc could read out the SOC directly from the car or if I could configure a "source" from which the SOC is taken (in my case the Hyundai/Kia API and the corresponding HomeAssistant integration)

panzerdev, 2024-03-24:
Turn off estimation and configure bluelink. What's missing?

https://docs.evcc.io/docs/reference/configuration/loadpoints#estimate

https://docs.evcc.io/docs/devices/vehicles#hyundai-bluelink

closed 2024-03-24 by andig

andreas-bulling, 2024-03-24:
I have bluelink configured already. And estimate I did not disable as the configuration says "not recommended"... So is it safe to disable?

panzerdev, 2024-03-24:
Sure. It's just going to update the soc in the polling interval you have configured and not constantly. Also no estimation is of course done. You can also change a car without car integration so no worries about anything not being safe