evcc-io/evcc #13145: Grünstromindex HTTP 429

olmo1501, 2024-03-25:
### Describe the bug

I have configured CO2 tariff and yesterday i recognized that no CO2 values are available anymore. 

Running `systemctl status evcc` shows:
`Mär 25 10:15:52 pv evcc[3780238]: [gsi   ] ERROR 2024/03/25 10:15:52 unexpected status: 429 (Too Many Requests)`
`Mär 25 10:15:52 pv evcc[3780238]: [main  ] ERROR 2024/03/25 10:15:52 failed configuring co2 tariff: cannot create tariff type 'grünstromindex': unexpected status: 429 (Too Many Requests)`

Who is running those API-calls? Is it my evcc instance or do you run those centralized? Is there anything I can do about it?

### Steps to reproduce

1. Configure CO2 tariff
2. Check if UI shows CO2 or just the cost per kWh
3.
...

### Configuration details

```yaml
tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.3197
  feedin:
    type: fixed
    price: 0.1051
  co2:
    type: grünstromindex
    zip: <schnippschnapp>
```

### Log details

```text
[main  ] INFO 2024/03/25 10:30:31 evcc 0.124.10
[main  ] INFO 2024/03/25 10:30:31 using config file: /etc/evcc.yaml
[main  ] INFO 2024/03/25 10:30:31 starting ui and api at :7070
[db    ] INFO 2024/03/25 10:30:31 using sqlite database: /root/.evcc/evcc.db
[mqtt  ] INFO 2024/03/25 10:30:31 connecting evcc-194965399 at tcp://192.168.241.2:1883
[mqtt  ] DEBUG 2024/03/25 10:30:31 tcp://192.168.241.2:1883 connected
[mqtt  ] DEBUG 2024/03/25 10:30:31 tcp://192.168.241.2:1883 subscribe SMA-PV/status/3015604826
[gsi   ] ERROR 2024/03/25 10:31:22 unexpected status: 429 (Too Many Requests)
[main  ] ERROR 2024/03/25 10:31:22 failed configuring co2 tariff: cannot create tariff type 'grünstromindex': unexpected status: 429 (Too Many Requests)
[semp  ] WARN 2024/03/25 10:31:23 SEMP_BASE_URL unspecified, using http://192.168.51.5:7070 instead
[site  ] INFO 2024/03/25 10:31:23 site config:
[site  ] INFO 2024/03/25 10:31:23   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2024/03/25 10:31:23     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2024/03/25 10:31:23     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2024/03/25 10:31:23     battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✓
[site  ] INFO 2024/03/25 10:31:23   vehicles:
[site  ] INFO 2024/03/25 10:31:23     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[site  ] INFO 2024/03/25 10:31:23     vehicle 2: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[site  ] WARN 2024/03/25 10:31:23 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2024/03/25 10:31:23 ----
[site  ] DEBUG 2024/03/25 10:31:23 pv power: 2494W
[site  ] DEBUG 2024/03/25 10:31:23 battery soc: 49%
[site  ] DEBUG 2024/03/25 10:31:23 battery power: -2216W
[site  ] DEBUG 2024/03/25 10:31:23 grid meter: 1W
[site  ] DEBUG 2024/03/25 10:31:23 grid currents: [0.356 0.242 0.297]A
[site  ] DEBUG 2024/03/25 10:31:23 site power: -2115W
[main  ] DEBUG 2024/03/25 10:31:27 vehicle status: B (VW e-up!)
```

### What type of operating system are you running?

Linux

### Version

0.124.10

mr42hh, 2024-03-25:
I get 
`failed configuring co2 tariff: cannot create tariff type 'grünstromindex': unexpected status: 500 (Internal Server Error)`
too.

olmo1501, 2024-03-25:
That's different than mine. "Internal Server Error" is most likely due to a problem on Corrently's site while "Too many requests" means that more calls to Corrently's API than allowed have been issued.

Buty maybe they are related? Let's wait for the Dev's feedback.

zoernert, 2024-03-25:
Due to the server outage in Strassbourg Datacenter the free  version of the GrünstromIndex is currently not available. We hope that out data center provider fixes the issues shortly. Meanwhile we could only provided limited free access.

olmo1501, 2024-03-25:
Those Strassburg datacenter don't seem to be a good choice. Fire at OVH in 2021 and now water in a battery room at Server4You...

zoernert, 2024-03-25:
> Those Strassburg datacenter don't seem to be a good choice. 

We are using several... but for a free service there is no budget for the more expensive ones. However, it might be good to look for some sponsors as there are thousands of requests to our free service per hour.

zoernert, 2024-03-25:
Update 12:30 UTC: Limited service is restored for free service allowing 1000 instead of 10 requests per minute. Hope that this is enough for the evcc community and ioBroker users.

olmo1501, 2024-03-25:
Don't get me wrong, I'm not critizising. Instead I'm grateful for these services. Thanks for the update.

closed 2024-03-25 by olmo1501

andig, 2024-03-25:
@zoernert we're very happy with using fly.io

Adminius, 2025-01-05:
Im getting since few days 429 also:

> [site ] WARN 2025/01/05 17:21:29 planner: tariff not available: cannot create tariff type 'template': cannot create tariff type 'grünstromindex': unexpected status: 429 (Too Many Requests)