evcc-io/evcc #13150: Tesla: Limit Charging current through Tesla API

Duell10111, 2024-03-25:
### Describe the bug

I am unsure if this should already supported or not therefore this could be an bug report or an feature request.

The problem is that the normal wallboxes only support minimum Current setting as 6A on their site with 3 phases, where as for example a Tesla Vehicle allows 5A as their minimum.
So when there is not enough PV power available for 4,1 kW it would be nice to let evcc start with 3,5 kW which is indeed currently possible by setting 5A as minimum current for the site.

But as the Wallbox do not understand anything under 6A, I currently have to limit it to 5A by myself on vehicle site through the Tesla App.

As evcc already has an integration for the Tesla API, it would be nice if evcc could handle this in this case automatically.
As I looked into the evcc code, I was unsure if evcc is not already setting the current value on vehicle site.

Below I put the logs from my evcc instance in the scenario described above.

### Steps to reproduce

1. Set min current to 5A in site settings
2. Have only enough PV power available to charge the vehicle with 5A on 3 phases (3,5 kW)
3. Set the PV Charge mode
4. Charging starts as evcc thinks it can charge with 5A, but stops after time window as not enough PV while charging with 4,1 kw instead of the desired 3,5 kW

### Configuration details

```yaml
sponsortoken: TOKEN

interval: 10s # control cycle interval

meters:
- type: template
  template: solaredge-hybrid 
  usage: grid  
  host: 192.168.178.10  
  port: 1502  
  timeout: 10s  
  name: grid1
  modbus: tcpip
  id: 1
- type: template
  template: solaredge-hybrid 
  usage: pv  
  host: 192.168.178.10  
  port: 1502  
  timeout: 10s  
  name: pv2
  modbus: tcpip
  id: 1
- type: template
  template: solaredge-hybrid 
  usage: battery  
  host: 192.168.178.10
  port: 1502  
  timeout: 10s  
  name: battery3
  modbus: tcpip
  id: 1

chargers:
  - name: keba
    type: template
    template: keba-modbus 

    # Modbus TCP
    modbus: tcpip
    id: 255
    host: 192.168.178.79
    port: 502 # Port 

vehicles:
- name: Tesla_Model_3_P
[... 173 lines cut ...]
[site  ] DEBUG 2024/03/25 13:27:55 battery soc: 54%
[site  ] DEBUG 2024/03/25 13:27:55 battery power: 526W
[site  ] DEBUG 2024/03/25 13:27:55 grid meter: -13W
[site  ] DEBUG 2024/03/25 13:27:55 site power: 563W
[lp-1  ] DEBUG 2024/03/25 13:27:55 charge currents: [6.01 6.04 6.02]A
[lp-1  ] DEBUG 2024/03/25 13:27:55 detected active phases: 3p
[lp-1  ] DEBUG 2024/03/25 13:27:55 charge total import: 6230.244kWh
[lp-1  ] DEBUG 2024/03/25 13:27:55 charger status: C
[lp-1  ] DEBUG 2024/03/25 13:27:55 soc estimated: 78.04% (vehicle: 78.00%)
[lp-1  ] DEBUG 2024/03/25 13:27:55 vehicle soc: 78%
[lp-1  ] DEBUG 2024/03/25 13:27:55 vehicle soc limit: 79%
[lp-1  ] DEBUG 2024/03/25 13:27:55 vehicle range: 375km
[lp-1  ] DEBUG 2024/03/25 13:27:55 pv charge current: 4.18A = 5A + -0.816A (563W @ 3p)
[lp-1  ] DEBUG 2024/03/25 13:27:55 projected site power 563W >= 200W disable threshold
[lp-1  ] DEBUG 2024/03/25 13:27:55 pv disable in 9m40s
[lp-1  ] DEBUG 2024/03/25 13:27:55 pv disable timer remaining: 9m40s
```

### What type of operating system are you running?

Linux

### Version

0.124.10

mucki12, 2024-03-25:
evcc setzt bereits heute die Stromstärke über die Tesla API (ab 1A). Allerdings bis jetzt nur in der Kombination TWC3+Tesla.
Bei dir erfolgt die Steuerung über die Keba und die kann wie alle anderen auch nicht unter 6A.

Duell10111, 2024-03-25:
@mucki12 
Ah okay, gut zu wissen dann ist das hier doch ein Feature Request und kein Bug Report. :D

andig, 2024-03-25:
> Ah okay, gut zu wissen dann ist das hier doch ein Feature Request und kein Bug Report. :D

Nein, die KEBA kann das nicht. Tesla ohne Keba wäre dann sowas wie https://github.com/evcc-io/evcc/pull/13062.

Duell10111, 2024-03-25:
> > Ah okay, gut zu wissen dann ist das hier doch ein Feature Request und kein Bug Report. :D
> 
> Nein, die KEBA kann das nicht. Tesla ohne Keba wäre dann sowas wie #13062.
> 
> @premultiply da das Fahrzeug das kann wäre es steuerungstechnisch auch spannend zu überlegen, ob sich das sinnvoll alternativ über _entweder_ Charger _oder_ Fahrzeug steuern ließe. Wäre aber keine Prio für mich?

Ja genau wäre auf jeden Fall interessant das auch in das Tesla Template zu integrieren. :D

Duell10111, 2024-03-28:
@andig 
Kleine Verständnisfrage, im Tesla Controller ist ja bereits das Setzten und abrufen der Current implementiert (siehe: https://github.com/evcc-io/evcc/blob/198e94e69537ca76b1a289daae459c8fedc8f8e4/vehicle/tesla/controller.go#L43)
es wird aber nur über die CMD Befehle aktuell benutzt hier(https://github.com/evcc-io/evcc/blob/198e94e69537ca76b1a289daae459c8fedc8f8e4/cmd/vehicle.go#L57).

Also es wird aktuell nicht aktiv vom Lademanagement verwendet, oder? :D

andig, 2024-03-28:
@Duell10111 OT

Duell10111, 2024-03-28:
@andig 
Ich habe mal einen Draft MR erstellt, indem die Stromstärke auch auf Fahrzeugseite gesetzt wird.
Testen kann ich aktuell nicht da ich keine Tesla Client ID und Secret habe. :(

Man könnte das vlt. noch anpassen das es auf Fahrzeugseite nur gesetzt wird wenn das Fahrzeug niedrigere Ampere unterstützt als die Wallbox, aber dafür müsste man diese Werte irgendwie herbekommen von den Wallbox bzw. Fahrzeug...

umrath, 2024-04-11:
> evcc setzt bereits heute die Stromstärke über die Tesla API (ab 1A). Allerdings bis jetzt nur in der Kombination TWC3+Tesla. Bei dir erfolgt die Steuerung über die Keba und die kann wie alle anderen auch nicht unter 6A.

Das "alle" ist definitiv so nicht richtig.
go-e kann definitiv mit 5A umgehen. Man kann es zwar nicht an der Wallbox selbst einstellen - aber wenn das Auto auf 5A reduziert, macht die go-e das problemlos mit.

Ich kann mir gut vorstellen, dass andere Wallboxen das auch können.

mucki12, 2024-04-11:
@umrath 
Danke für die (ungewollte?) Bestätigung. Die go-e wird auf 6A geregelt und da die Steuerung über das Fahrzeug erfolgt ruft dieses 5A ab.

andig, 2024-04-11:
> Ich kann mir gut vorstellen, dass andere Wallboxen das auch können.

Dieser Satz kein Inhalt da das mit der WB absolut nichts zu tun hat. Genauso könntest Du auch sagen, dass sie sogar bei Vollmond lädt.

josefzahner, 2024-04-18:
@andig auch meine Easee kann mit 5A umgehen, siehe unten. Warum solls auch nicht gehen, wenn die Batterien vom Auto voll sind und das Auto immer weniger Strom anfordert gehts ja auch runter mit dem Strom. Das auf Wallbox Seite nicht weniger eingestellt werden kann (wegen dem Standard) und die meisten Autos mind. 6A erfordern ist das eine, aber Tesla kann nunmal mit weniger als 6A Laden - da wärs doch super wenn EVCC das auch runterregeln könnte bei Bedarf. 
Für mich siehts so aus als ob wohl die meisten Wallboxen mit weniger als 6A umgehen können, wenn vom Auto weniger gefordert wird. Von daher ist das ein sehr guter Input. Ich möchte nämlich auch mit weniger als 6A dreiphasig laden können! Fragen wir doch mal anders, kann den jemand mit einer NICHT **go-e** oder **easee** mal probieren ob er weiter runter kommt als 6A mit dem Tesla? In der App kann man ja easy 5A einstellen (weniger geht ja nicht so einfach - zumindest nicht in der App). Dann haben wir nämlich mal Fakten auf dem Tisch.

[image]
[image]

andig, 2024-04-18:
> Warum solls auch nicht gehen, wenn die Batterien vom Auto voll sind und das Auto immer weniger Strom anfordert gehts ja auch runter mit dem Strom. 

Weil der IEC Standard 6A als unteres Limit der Signalisierung WB->Auto hat. Ganz einfach.

josefzahner, 2024-04-18:
Richtig, du schreibst ja selber WB-> Auto, aber das Auto kann ja selber noch weiter runter als die 6A Laden, sonst würdest du nämlich einen kleine PlugIn Akku gar nie ganz voll kriegen...

andig, 2024-04-18:
Ja, das habe ich nun schon mindestens 2x geschrieben. Und exakt *weil* das mit der WB überhaupt gar nichts zu tun hat- genauso wenig wie mit Vollmond- ist es auch sinnlos, das mit anderen WBen zu probieren. Dort ist das nämlich exakt genauso (wenns keine ISO WB ist). Aber wir drehen uns da im Kreis...

josefzahner, 2024-04-18:
Ok fair enough, das hat nichts mit der Wallbox zu tun, ich dachte genau das sei dein Kritikpunkt. 

Dann verstehe ich aber nicht warum man dann scheinbar in EVCC mit Tesla und einem Tesla Wallbox die Steuerung übers Auto zulässt, das aber bei allen anderen Boxen tunlichst nicht möchte.

josefzahner, 2024-04-18:
Ok dann fassen wir mal zusammen, für mich hats von aussen so ausgesehen als ob EVCC die Steuerung nur über die Wallbox erlauben möchte - und explizit nicht das Auto nicht steuern möchte. Und da bin ich nicht der einzige, wenn man Forenbeiträge ausserhalb anschaut. Aber scheinbar ist ja hier der richtige Ort um das zu diskutieren und die Annahme war wohl falls - dann habe ich nichts gesagt. Weiter gehts :-)

closed 2024-05-15 by github-actions[bot]