evcc-io/evcc #13158: Unavailable car is logged with (unavailable)

mansurs, 2024-03-26:
### Describe the bug

Usually my car is name `BMW iX3`, but yesterday I had a timeframe, where the car was named `BMW iX3 (unavailable)`, because I have chosen it manually from the list, but it still wasn't recognised. This got logged as a new car in the "Ladevorgänge" interface and inside influx db. See screenshots.

I would expect it should be logged as the original name, instead of a new car with (unavailable) suffix.

[image]
[image]

### Steps to reproduce

1. Plugin a car, which is known by evcc
2. Evcc should not be able to communicate to the car manufacturer api and try to recognise it
3. Chose it manually from the list, so it should be shown as "Car name (unavailable)"
4. Check logs

### Configuration details

```yaml
Not needed
```

### Log details

```text
Not needed
```

### What type of operating system are you running?

Linux

### Version

0.124.10

andig, 2024-03-26:
Fixed in master

closed 2024-03-26 by andig

VolkerK62, 2024-03-26:
and you can change the name within the session details

mansurs, 2024-03-26:
Ah nice! Thanks for the hint!