evcc-io/evcc #13162: Generic automatic 1P3P switching

schulzh, 2024-03-26:
**Is your feature request related to a problem? Please describe.**
While some EVSEs support automatic 1P3P switching, many of the ones which do not could be made to support it by adding a contactor to phases 2 and 3. This is even suggested in the [FAQ](https://docs.evcc.io/docs/faq#meine-wallbox-hat-keine-phasenumschaltung-kann-ich-trotzdem-1-phasig-laden), but manually with a switch instead of automatically.

I've done this myself, I've got a Heidelberg Energy Control and have added a Contactor (Siemens 5TT5 8000), operated by a Shelly Pro 1 to control the additional phases. My charging logic (currently custom in node-red) switches phases by first setting the charging current to 0, waiting until the current has indeed reached 0, enabling/disabling the contactor and then setting the charging current again.

This works for both my Polestar 2 and a BMW i4 of a friend. I assume this will work with most (if not all) other cars the same way, as this should be what the other EVSEs which natively support 1P3P do.

**Describe the solution you'd like**
A method of adding a 1P3P switch to an EVSE that does not natively support it would be great. The configuration should be generic, such that different types of switches can be configured (e.g. shelly, zigbee, modbus, gpio, etc.).
It could be limited (soft or hard) to wallboxes which have been tested, and/or where current can be read out, to only switch once the current has reached 0.

**Describe alternatives you've considered**
A different approach than implementing this generically is to implement it on a per-wallbox basis. I've noticed that the current coding for 1P3P switching is always specific to a family of hardware and not generic, so this would be more in-line with the current implementation and less refactoring work. However, since the actual phase switching is not specific to any EVSE, this would result in code/logic duplication.

**Additional context**
I'm don't know if this has come up before; from a quick search I did not find a generic feature request.
I'm also not sure if this would be "too dangerous" or "too adventeours" to add natively to evcc. I think a precedent has been made by suggesting the manual operation in the FAQ however, which hints that the users of evcc are considered to be technically founded enough to operate such an addition to their charging equipment.
I'd also be interested in working on this (no guarantees though), as I'd like to switch to evcc from node-red and this is currently a blocker for me.

Raudi1, 2024-03-26:
I plan to do something similar with a Shelly Pro 3. Because I only have a small PV, I have to charge on 1 phase with less than 6A. That only works in EVCC with a Tesla and TWC3 which doesn't support automatic 1p/3p switching. I have a Go-e Gemini which does support automatic switching though.Right now when the energy is cheap (I use Tibber) or I need to charge faster, I have to manually switch on the circuit breakers. And afterwards I have to switch them back off.
I have read somewhere, that some cars don't like switching the phases and you could damage the on board chargers. Some EVSE could be problematic as well. I manually simulated the desired behavior and it worked without any issues. Even switching off under load didn't result in an error (did it for testing purposes only).

andig, 2024-03-26:
> I've done this myself, I've got a Heidelberg Energy Control and have added a Contactor (Siemens 5TT5 8000), operated by a Shelly Pro 1 to control the additional phases. My charging logic (currently [...]

This is dangerous. We won't support it. You can already do this in a custom charger.

schulzh, 2024-03-26:
>This is dangerous
What exactly? My implementation? The hardware chosen? The concept in general?

andig, 2024-03-26:
Security of electrical belongs in hardware, not software.

schulzh, 2024-03-27:
Thats fair, thanks for the answer.

>if you accept the terms of service

what exactly do you mean with this?

TheTrueRandom, 2025-07-29:
I came across the same problem (and happen to have the exact same wallbox @schulzh). I added a custom charger to do the phase switching, which works nicely. [Here's the repository](https://github.com/TheTrueRandom/evcc-heidelberg-wallbox-p1p3).

I would still prefer if the logic can be added to evcc, so more wallboxes could be supported. Obviously with a big disclaimer and detailed explaining the risks. I think it's more dangerous what I (and probably others) did, than if a whole community could verify and maintain a feature to make it as safe as possible, if this was part of evcc.

People will always build it and it doesn't make the world to a safer place just because evcc doesn't implement it. Rather we could provide a detailed description of the risks and clearly state it when the feature is being used (not like the current "tos" property, which I still don't know what's it's purpose, nowhere I found a description of which terms I am accepting by setting it to true)

andig, 2025-08-01:
See https://github.com/evcc-io/evcc/issues/13162#issuecomment-2021342264