evcc-io/evcc #13163: Citroen: login not possible after changes done by PSA

ThoHilde, 2024-03-26:
### Describe the bug

It seems like PSA changed the login procedure for the api (see https://github.com/flobz/psa_car_controller/discussions/779)
Since then evcc cannot login and the car is charged as an guest car with unknown soc.

It is possible to login with PSA Controller but not as easy as entering user/pass.

### Steps to reproduce

1. Configure PSA (Citroen)
2. Start evcc
3. check web ui or logs

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error
  ec4: trace

# unique installation id
plant: ec0508860b30906bcb5f51e4b48c310453800def509086de0937c733a2737d62

interval: 10s # control cycle interval

sponsortoken: *****************

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 192.168.178.220  
  port: 502  
  usage: grid  
  modbus: tcpip  
  timeout: 10s  
  name: grid1
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 192.168.178.220  
  port: 502  
  usage: pv  
  modbus: tcpip  
  timeout: 10s  
  name: pv2
#- type: template
#  template: sungrow-hybrid 
#  id: 1  
#  host: 192.168.178.220  
#  port: 502  
#  usage: battery  
#  modbus: tcpip  
#  timeout: 10s  
#  name: battery3
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 192.168.178.66  
  port: 502  
  usage: pv  
  modbus: tcpip  
  timeout: 10s  
  name: pv4

chargers:
- type: template
  template: heidelberg 
  id: 1
  host: 192.168.178.196  
  port: 8123
  modbus: tcpip  
  name: wallbox6

#chargers:
#type: template
#template: heidelberg 
#  id: 1  
#  host: 192.168.178.196  
#  port: 8123  
#  modbus: rs485tcpip  
#  name: wallbox6

vehicles:
- name: ec4
  type: template
  template: citroen 
  title: ec4  
  user: ***@*******.de 
  password: ******************
  vin: VR*********  
  capacity: 50

loadpoints:
- title: Garage
  vehicle: ec4
  charger: wallbox6
  mode: off

site:
  title: Zuhause
  meters:
    grid: grid1
    pv:
    - pv2
    - pv4
#    battery:
#    - battery3

mqtt:
  broker: localhost:1883
  topic: evcc # root topic for publishing, set empty to disable publishing
  # clientid: foo
  user: homeassistant
  password: *******
```

### Log details

```text
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/03/26 20:02:32 evcc 0.124.10
[main  ] INFO 2024/03/26 20:02:32 using config file: /config/evcc.yaml
[main  ] INFO 2024/03/26 20:02:32 starting ui and api at :7070
[db    ] INFO 2024/03/26 20:02:33 using sqlite database: /data/evcc.db
[mqtt  ] INFO 2024/03/26 20:02:33 connecting evcc-1982962816 at tcp://localhost:1883
[mqtt  ] DEBUG 2024/03/26 20:02:33 tcp://localhost:1883 connected
[main  ] ERROR 2024/03/26 20:02:33 creating vehicle ec4 failed: cannot create vehicle type 'template': cannot create vehicle type 'citroen': login failed: oauth2: "server_error" "Internal Server Error"
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.124.10

andig, 2024-03-26:
Yo, das ist bekannt...

closed 2024-03-26 by andig

ThoHilde, 2024-03-26:
> Yo, das ist bekannt...

... und kann/wird gefixt werden? Oder haben die das so verkompliziert, dass es unwahrscheinlich ist?

StefanSchoof, 2024-03-27:
Siehe https://github.com/evcc-io/evcc/discussions/12231

ThoHilde, 2024-03-27:
Danke. PSA Controller läuft auf meinem HAOS sowieso. Wusste nicht, dass ich das per Template in evcc integrieren kann. Habe meine config umgestellt und es funktioniert wieder.