evcc-io/evcc #13165: Tariff tibber - no longer receiving prices, error "smartCost: no matching rate"

mrandt, 2024-03-27:
### Describe the bug

Since yesterday my evcc instance no longer receives price info for tibber. The upcoming hours in charge planning display "unknown" and in error notifications and log I find "smartCost: no matching rate" error.

This worked fine before. I double-checked my access token (which is valid) and also added the homeid (which I previously had not used) - but the behaviour is still the same.

I am not sure if this is a bug in evcc or if the tibber API is to blame.

How could I check?

### Steps to reproduce

1. Configure tarriff tibber 
2. Try to see the dynamic prices, e.g. in charge planning

[image]

[image]

### Configuration details

```yaml
tariffs:
  currency: EUR # (default EUR)

  grid:
    type: tibber
    token: "XYZ"
    homeid: "e1234-..."

#  grid:
#    # static grid price
#    type: fixed
#    price: 0.32 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.082 # [currency]/kWh
```

### Log details

```text
[site  ] ERROR 2024/03/27 10:34:25 smartCost: no matching rate
```

### What type of operating system are you running?

Docker container

### Version

evcc version 0.124.10

BeckerWdf, 2024-03-27:
This is an issue in the Tibber API. It does simply not provide rates. See: https://github.com/evcc-io/evcc/discussions/13160
Nevertheless their App seems to not use the API because it still shows the rates correctly.

mrandt, 2024-03-27:
It was in fact a glitch in Tibber API. I chatted with their tech support and within a few hours they told me it was fixed now.

I can confirm it is working again and had nothing to do with evcc.

closed 2024-03-27 by mrandt

BeckerWdf, 2024-03-27:
> It was in fact a glitch in Tibber API. I chatted with their tech support and within a few hours they told me it was fixed now.
> 
> I can confirm it is working again and had nothing to do with evcc.

Did they tell you why this was not mentioned on their status page?