evcc-io/evcc #13180: failed configuring loadpoints

TerrorSource, 2024-03-27:
### Describe the bug

failed configuring loadpoints
missing loadpoints

### Steps to reproduce

1. use "evcc configure"
2. only add "SMA Speedwire", skip all other options
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070  
network:  
  schema: http  
  host: evcc.local # .local suffix announces the hostname on MDNS  
  port: 7070  
  
log: debug  
levels:  
  cache: error  
  
# unique installation id  
plant: *****  
  
interval: 30s # control cycle interval  
  
meters:  
- type: template  
  template: sma-inverter-speedwire   
  usage: pv    
  host: 192.168.1.15    
  password: *****  
  name: pv1  
  
site:  
  title: House  
  meters:  
    pv:  
    - pv1
```

### Log details

```text
[main  ] INFO 2024/03/28 00:26:13 evcc 0.124.10
[main  ] INFO 2024/03/28 00:26:13 using config file: /app/evcc.yaml
[main  ] INFO 2024/03/28 00:26:13 starting ui and api at :7070
[db    ] INFO 2024/03/28 00:26:13 using sqlite database: /root/.evcc/evcc.db
[main  ] FATAL 2024/03/28 00:26:14 failed configuring loadpoints: missing loadpoints
[main  ] FATAL 2024/03/28 00:26:14 will attempt restart in: 5m0s
```

### What type of operating system are you running?

Docker container

mdkeil, 2024-03-28:
You have to config a charger!

naltatis, 2024-03-28:
> only add "SMA Speedwire", skip all other options

Yes, don't skip all other steps. Without a charger/loadpoint evcc doesnt work right now.
We could add a warning in the configurator, but since we are working on replacing it with web UI we'll likely not invest effort there.

closed 2024-03-28 by naltatis

TerrorSource, 2024-03-28:
> > only add "SMA Speedwire", skip all other options
> 
> Yes, don't skip all other steps. Without a charger/loadpoint evcc doesnt work right now. We could add a warning in the configurator, but since we are working on replacing it with web UI we'll likely not invest effort there.

I don’t have a Charger for my car. 
I want to have Eccv as a connector between my SMA with Speedwire and HomeAssistant. 
Except the SMA, there is nothing I would like to configure in evcc. 

Is there a dummy charger or one that does not work but let me use evcc this way?

naltatis, 2024-03-28:
Yes, there is. See: https://docs.evcc.io/en/docs/faq/#can-i-try-out-evcc-without-all-the-components-pv-system-charger-

TerrorSource, 2024-03-28:
> Yes, there is. See: https://docs.evcc.io/en/docs/faq/#can-i-try-out-evcc-without-all-the-components-pv-system-charger-

Thanks, i got somethings working now.
Can i remove more options/info from the yaml then i already did?

And how do i make evcc send mqtt data to HomeAssistant?

```
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

interval: 30s # control cycle interval

mqtt:
   broker: 192.168.1.12:1883
   topic: evcc # root topic for publishing, set empty to disable
   user: mqtt-user
   password: mqtt-password

meters:
  - name: pv1
    type: template
    template: sma-inverter-speedwire
    usage: pv
    host: 192.168.1.11
    password: sma-password

  - name: meter_charger_1
    type: custom
    power:
      source: js
      vm: shared
      script: state.loadpoints[0].chargepower;

chargers:
  - name: charger_1
    type: custom
    status: # charger status A..F
      source: script
      cmd: /bin/sh -c 'echo b'
    enabled: # charger enabled state (true/false or 0/1)
      source: script
      cmd: /bin/sh -c 'echo false'
    enable: # set charger enabled state (true/false or 0/1)
      source: script
      cmd: /bin/sh -c 'echo false'
    maxcurrent: # set charger max current (A)
      source: script
      cmd: /bin/sh -c 'echo 10'

vehicles:
  - name: vehicle_1
    type: template
    template: offline
    title: vehicle_1
    icon: waterheater

site:
  title: Home
  meters:
    pv: pv1

loadpoints:
  - title: Carport
    charger: charger_1
    mode: pv
    meter: meter_charger_1
    vehicle: vehicle_1

```

naltatis, 2024-03-28:
Yes, see the docs. evcc can write to mqtt. homeassistant can read from it.