evcc-io/evcc #13189: tesla and tesla-command in debug log always say token expired

damarges, 2024-03-28:
### Describe the bug

Running evcc lets me start the charging of my tesla (probably it simply starts the charge in the wallbox) but my SoC from my car is not visible. Regardless of using tesla or tesla-command template. The debug log always give me this error:

> evcc  | [main  ] ERROR 2024/03/28 14:17:49 creating vehicle ev4 failed: cannot create vehicle type 'template': cannot create vehicle type 'tesla': token expired 

or if using tesla-command it changes of course to tesla-command.

Thats how it looks:
[image]

I am using 0.124.10 (nightly)

### Steps to reproduce

1. Run evcc in docker.
2. Oben WebUI and have a look at the cars section
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error
  ev4: debug
  tesla-command: debug

# unique installation id
plant: ********
interval: 10s # control cycle interval
sponsortoken: ****************

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

meters:
- type: template
  template: goodwe-wifi
  usage: grid
  uri: 192.168.0.23
  name: grid1
- type: template
  template: goodwe-wifi
  usage: pv
  uri: 192.168.0.23
  name: pv2
- type: template
  template: goodwe-wifi
  usage: battery
  uri: 192.168.0.23
  name: battery3

chargers:
- type: template
  template: go-e-v3
  host: 192.168.0.76
  name: wallbox5

vehicles:
- type: template
  template: tesla
  title: Model Y
  accessToken: *****
  refreshToken: *****
  vin: LRWYGCFS0PC********
  capacity: 60
  name: ev4

loadpoints:
- title: Go-E Charger Gemini (11kW)
  charger: wallbox5
  mode: off
  vehicle: ev4
```

### Log details

```text
/app # evcc vehicle
[main  ] INFO 2024/03/28 14:23:29 evcc 0.124.10 (a5c7b9fd)
[main  ] INFO 2024/03/28 14:23:29 using config file: /etc/evcc.yaml
[db    ] INFO 2024/03/28 14:23:29 using sqlite database: /root/.evcc/evcc.db
[main  ] ERROR 2024/03/28 14:23:29 creating vehicle ev4 failed: cannot create vehicle type 'template': cannot create vehicle type 'tesla': token expired
.Soc:     vehicle not available: cannot create vehicle type 'template': cannot create vehicle type 'tesla': token expired
Capacity: 60.0kWh
Features: [Offline Retryable]
```

### What type of operating system are you running?

Docker container

### Version

evcc version 0.124.10 (a5c7b9fd)

damarges, 2024-03-28:
Additional info. I tried to create new tokens, here is the error I get now:
`evcc  | [main  ] ERROR 2024/03/28 14:29:38 creating vehicle ev4 failed: cannot create vehicle type 'template': cannot create vehicle type 'tesla': Get "https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/users/region": net/http: request canceled while waiting for connection (Client.Timeout exceeded while awaiting headers)
`

andig, 2024-03-28:
Your token is expired, thats why…