evcc-io/evcc #13194: SMA Speedwire wrong data

TerrorSource, 2024-03-28:
### Describe the bug

evcc retrieves the wrong data from the SMA inverter([SB 3000TL-21) via speedwire

### Steps to reproduce

1. setup evcc with config:

2. go to http://192.168.1.9:7070/api/state and check pvEnergy value
3. the "pvEnergy" value is the same as the Total Generated energy, not the daily generated energy
...
[image]

### Configuration details

```yaml
meters:
  - name: pv1
    type: template
    template: sma-inverter-speedwire
    usage: pv
    host: 192.168.1.111
    password: sma-password
```

### Log details

```text
api/state json:

    "pv": [
      {
        "power": 0,
        "energy": 33120.642
      }
    ],
    "pvConfigured": true,
    "pvEnergy": 33120.642,
    "pvPower": 0,
    "residualPower": 0,
    "siteTitle": "Home",
    "smartCostType": null,
    "statistics": {
      "30d": {
        "avgCo2": 0,
        "avgPrice": 0.08000000000000002,
        "chargedKWh": 0.5213336094754222,
        "solarPercentage": 100.00000000000001
      },
      "365d": {
        "avgCo2": 0,
        "avgPrice": 0.08000000000000002,
        "chargedKWh": 0.5213336094754222,
        "solarPercentage": 100.00000000000001
      },
      "total": {
        "avgCo2": 0,
        "avgPrice": 0.08000000000000002,
        "chargedKWh": 0.5213336094754222,
        "solarPercentage": 100.00000000000001
      }
```

### What type of operating system are you running?

Docker container

andig, 2024-03-29:
> not the daily generated energy

I would expect total

closed 2024-03-29 by andig

TerrorSource, 2024-03-29:
> > not the daily generated energy
> 
> I would expect total

Might be helpfull to give some more information then a typical "i would expect total" and close the issue.

How can i make sure that evcc does read out the dialy yield so i can forward that via MQTT?
What is the use of "total" in general? It isn't helpfull when you want to do load-balancing with charging stations

andig, 2024-03-29:
Loadbalancing is generally not done on energy and not yet supported in evcc.

TerrorSource, 2024-03-29:
> Loadbalancing is generally not done on energy and not yet supported in evcc.

What about the question/issue getting the daily yield from the inverter?

premultiply, 2024-03-29:
evcc does not care about special daily production calculation.

You can calculate it by yourself by reading total energy once a day at the same time and  do some very magic math like "value of today - value of yesterday". 😉