evcc-io/evcc #13200: Corsa-e (unavailable)

Sm4rtH0me, 2024-03-29:
### Describe the bug

Hallo zusammen, 

ich habe seit längerer Zeit das Problem, dass mir bei meinem Opel Corsa e in der Oberfläche neben dem Namen (unavailable) angezeigt wird. Geladen wird das Fahrzeug trotzdem - nur kann man nicht einstellen, bis wieviel Prozent das Fahrzeug geladen werden soll. Ich dachte zunächst, dass es an dem fehlenden Connect Paket von Opel liegt, welches man gegen monatliche Gebühr dazubuchen muss. Hat allerdings an dem Zustand nichts verändert.

Vielleicht hat jemand einen Tipp hierzu.

### Steps to reproduce

Opel Corsa-e Modelljahr 2023

### Configuration details

```yaml
GNU nano 4.8                                                                                                   /etc/evcc.yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

# unique installation id
plant: 2883d4862d050e58a56285b7311005f65e5e7bc2dff4077d7be0b512b06999f8

interval: 10s # control cycle interval

sponsortoken: eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJleHAiOjE3NjI0NzE1NDAsImlhdCI6MTY2Nzg2MzU0MCwiaXNzIjoiZXZjYy5pbyIsInN1YiI6IlNtNHJ0SDBtZSJ9.rhiixM6Uuy4oZFwZPdBxNKiZmVCRvZJglN-7hqBD-R0

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: huawei-dongle-powersensor
  id: 1
  host: 192.168.1.59
  port: 502
  usage: grid
  modbus: tcpip
  name: grid1
- type: template
  template: huawei-dongle-powersensor
  id: 1
  host: 192.168.1.59
  port: 502
  usage: pv
  modbus: tcpip
  name: pv2
- type: template
  template: huawei-dongle-powersensor
  id: 1
  host: 192.168.1.59
  port: 502
  usage: battery
  modbus: tcpip
  name: battery3
- type: template
  template: huawei-dongle-powersensor
  id: 2
  host: 192.168.1.59
  port: 502
  usage: pv
  modbus: tcpip
  name: pv4

chargers:
- type: template
  template: easee
  user: fam@cumia.de
  password: '*CuMiA32415*'
  charger: EH3MYPYU
  name: wallbox6
```

### Log details

```text
[main  ] INFO 2024/03/29 11:38:58 evcc 0.124.10
[main  ] INFO 2024/03/29 11:38:58 using config file: /etc/evcc.yaml
[main  ] INFO 2024/03/29 11:38:58 starting ui and api at :7070
[db    ] INFO 2024/03/29 11:38:59 using sqlite database: /home/marco/.evcc/evcc.db
[main  ] ERROR 2024/03/29 11:39:00 creating vehicle ev6 failed: cannot create vehicle type 'template': cannot create vehicle type 'opel': login failed: oauth2: "server_error" "Internal Server Error"
[lp-1  ] DEBUG 2024/03/29 11:39:01 charge total import: 3035.256kWh
[lp-2  ] DEBUG 2024/03/29 11:39:01 charge total import: 362.957kWh
[site  ] WARN 2024/03/29 11:39:01 battery configured but residualPower is missing or <= 0 (add residualPower: 100 to site), see https://docs.evcc.io/en/docs/reference/configuration/site#residualpower
[site  ] WARN 2024/03/29 11:39:01 vehicle 'Corsa-e (unavailable)' does not support automatic detection
[site  ] INFO 2024/03/29 11:39:01 site config:
[site  ] INFO 2024/03/29 11:39:01   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2024/03/29 11:39:01     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2024/03/29 11:39:01     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2024/03/29 11:39:01     pv 2:      power ✓ energy ✓ currents ✗
[site  ] INFO 2024/03/29 11:39:01     battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2024/03/29 11:39:01   vehicles:
[site  ] INFO 2024/03/29 11:39:01     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[site  ] INFO 2024/03/29 11:39:01     vehicle 2: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2024/03/29 11:39:01 loadpoint 1:
[lp-1  ] INFO 2024/03/29 11:39:01   mode:        pv
[lp-1  ] INFO 2024/03/29 11:39:01   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2024/03/29 11:39:01   meters:      charge ✓
[lp-1  ] INFO 2024/03/29 11:39:01     charge:    power ✓ energy ✓ currents ✓
[lp-2  ] INFO 2024/03/29 11:39:01 loadpoint 2:
[lp-2  ] INFO 2024/03/29 11:39:01   mode:        pv
[lp-2  ] INFO 2024/03/29 11:39:01   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-2  ] INFO 2024/03/29 11:39:01   meters:      charge ✓
[lp-2  ] INFO 2024/03/29 11:39:01     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2024/03/29 11:39:01 phase timer inactive
[lp-1  ] DEBUG 2024/03/29 11:39:01 pv timer inactive
[lp-2  ] DEBUG 2024/03/29 11:39:01 phase timer inactive
[lp-2  ] DEBUG 2024/03/29 11:39:01 pv timer inactive
[main  ] FATAL 2024/03/29 11:39:01 could not open port- check that evcc is not already running (listen tcp :7070: bind: address already in use)
```

### What type of operating system are you running?

Linux

### Version

v0.124.10

andig, 2024-03-29:
Duplikat...

closed 2024-03-29 by andig

Sm4rtH0me, 2024-03-29:
> Duplikat...

Duplikat? Darf ich erfahren, was du damit meinst?

Dankeschön.

andig, 2024-03-29:
Naja, normal... aber leider haben wir aktuell keine Handhabe :(