evcc-io/evcc #13201: Validate charger phases

andig, 2024-03-29:
Ich hänge mich in diese Diskussion jetzt auch mal mit rein ;-)

Trotz der letzten Erweiterungen von Tinkerforge kommt es hin und wieder vor (z. B. nach einem Firmware-Update), dass die vom WARP Energy Manager tatsächlich geschalteten Phasen nicht dem letzten bekannten Stand von EVCC entsprechen.
EVCC geht dann davon aus, dass z. B. 3-phasig geladen werden soll (und löst somit auch keine neue Phasenumschaltung auf 3-phasig aus), WEM hat aber keine oder nur eine Phase geschaltet.

@andig, siehst du irgendeine Möglichkeit, dass EVCC in/vor "scalePhases()" den internen Zustand mit dem des WEM (oder jeder anderen Wallbox mit automatischer Phasenumschaltung) synchronisiert?

Besten Dank und Gruß Thomas

_Originally posted by @poohnet in https://github.com/evcc-io/evcc/discussions/8461#discussioncomment-8950931_

GrimmiMeloni, 2024-03-29:
Siehe mein Kommentar hier: https://github.com/evcc-io/evcc/issues/12773#issuecomment-1987263008

Ich glaube beim Phasenabgleich sollte nicht das Minimum gewinnen, sondern der Charger.

poohnet, 2024-03-29:
Wie kann ich helfen? Gibt es schon Ideen/Konzepte, wie und wo der Phasenabgleich EVCC <=> WB stattfinden soll?

poohnet, 2024-03-29:
@andig, @GrimmiMeloni:
Ich habe mich mal ein bisschen durch den Code gewühlt und hier eine Idee (evtl. sogar mögliche Lösung?) skizziert: https://github.com/poohnet/evcc/commit/4c8520306927284e9ce1f25e9f47960d45030931

Zum Testen habe ich eine neue Debug-Ausgabe in "updateChargerStatus()" eingebaut und die Phasen unabhängig von EVCC direkt in WARP umgeschaltet:

```
[lp-1  ] DEBUG 2024/03/29 20:40:35 charge power: 0W
[site  ] DEBUG 2024/03/29 20:40:35 pv power: 764W
[site  ] DEBUG 2024/03/29 20:40:35 battery soc: 70%
[site  ] DEBUG 2024/03/29 20:40:35 battery power: 0W
[site  ] DEBUG 2024/03/29 20:40:35 grid meter: -309W
[site  ] DEBUG 2024/03/29 20:40:35 grid currents: [-2.65 0.691 1.39]A
[site  ] DEBUG 2024/03/29 20:40:35 site power: -209W
[lp-1  ] DEBUG 2024/03/29 20:40:35 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/03/29 20:40:35 charge total import: 1434.510kWh
[lp-1  ] DEBUG 2024/03/29 20:40:35 charger status: A
[lp-1  ] DEBUG 2024/03/29 20:40:35 charger phases: 3          <==
[site  ] DEBUG 2024/03/29 20:40:50 ----
[lp-1  ] DEBUG 2024/03/29 20:40:50 charge power: 0W
[site  ] DEBUG 2024/03/29 20:40:50 pv power: 764W
[site  ] DEBUG 2024/03/29 20:40:50 battery soc: 70%
[site  ] DEBUG 2024/03/29 20:40:50 battery power: 0W
[site  ] DEBUG 2024/03/29 20:40:50 grid meter: -303W
[site  ] DEBUG 2024/03/29 20:40:50 grid currents: [-2.65 0.692 1.37]A
[site  ] DEBUG 2024/03/29 20:40:50 site power: -203W
[lp-1  ] DEBUG 2024/03/29 20:40:50 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/03/29 20:40:50 charge total import: 1434.510kWh
[lp-1  ] DEBUG 2024/03/29 20:40:50 charger status: A
[lp-1  ] DEBUG 2024/03/29 20:40:50 charger phases: 1          <==
```

Was haltet ihr von dieser Idee?

GrimmiMeloni, 2024-03-30:
> @GrimmiMeloni zu schauen ob richtig gesetzt wurde (oder WB einen Neustart gemacht hat) ist doch erstmal ein ganz anderes Problem als Deins. Dafür neues Issue?

Du hast Recht. Mein Problem ist (wenn überhaupt) nachgelagert.

Frage hierzu, bzw. auch dem konkreten Vorschlag von @poohnet. Erfüllt nicht der existierende Code in https://github.com/GrimmiMeloni/evcc/blob/48772bf1428c6bf4b359ba71cfb391b110eaf39b/core/loadpoint.go#L1319 genau diesen Zweck bereits?

poohnet, 2024-03-30:
Die Prüfung der Phasenspannungen kann funktionieren, wenn der Stromzähler hinter den Schützen hängt. Es gibt aber Situationen (z. B. bei mir oder der kommenden WARP3), wo dies nicht der Fall ist, d. h. die Wallbox immer drei angeschlossene Phasen sieht.
Darüber hinaus habe ich bei unserer Zoe festgestellt, dass bei einphasigem Laden auf L1 trotz abgeschaltetem Schütz für L2+L3 auf L2 eine Spannung > 100V anliegt, die anscheinend von dem „besonderen“ Renault-Lader verursacht wird.

GrimmiMeloni, 2024-03-30:
Verstehe, die Erkennung funktioniert dann nur bei einigen Boxen.

Was ist sonst mit der [Implementierung im Loadpoint](https://github.com/GrimmiMeloni/evcc/blob/48772bf1428c6bf4b359ba71cfb391b110eaf39b/core/loadpoint.go#L1302) die auch heute schon "detected active phases" ausspuckt. Die setzt den Wert zwar nicht auf dem Charger, aber zumindest die Logik die Phasen zu erkennen ist vorhanden. Könnten wir hier sonst ggf. einhaken?

GrimmiMeloni, 2024-03-30:
OK, das funktioniert auch nur bedingt, weil dafür ein Ladevorgang aktiv sein muß.
So fern nicht noch jemand eine andere Lösung im bestehenden Code sieht, scheint mir der Vorschlag von @poohnet gut zu sein.

andig, 2024-03-30:
Das gehört m.E. in syncCjarger wo auch der Strom geprüft wird.

poohnet, 2024-03-30:
Gehe ich recht in der Annahme, dass wir hier dann mittels "SetPhases()" **PhasesEnabled** auf den von der Wallbox zurückgegebenen Wert setzen müssen?

````
PhasesConfigured = "phasesConfigured" // configured phases (1/3, 0 for auto on 1p3p chargers, nil for plain chargers)
PhasesEnabled    = "phasesEnabled"    // enabled phases (1/3)
PhasesActive     = "phasesActive"     // active phases as used by vehicle (1/2/3)
````

poohnet, 2024-03-30:
Kann das tatsächlich alles sein?!?
````
diff --git a/core/loadpoint.go b/core/loadpoint.go
index 26c5db52..da51ab99 100644
--- a/core/loadpoint.go
+++ b/core/loadpoint.go
@@ -666,6 +666,14 @@ func (lp *Loadpoint) syncCharger() error {
 		return fmt.Errorf("charger enabled: %w", err)
 	}
 
+	if ps, ok := lp.charger.(api.PhaseSwitcherState); ok {
+		phases, err := ps.GetPhases()
+		if err == nil && phases != lp.GetPhases() {
+			lp.log.WARN.Printf("charger phases out of sync: expected %d, got %d", lp.GetPhases(), phases)
+			lp.setPhases(phases)
+		}
+	}
+
 	shouldBeConsistent := lp.chargerUpdateCompleted() && lp.phaseSwitchCompleted()
 
 	if shouldBeConsistent {
 ````
Auf den ersten Blick **scheint** EVCC das nun korrekt mitzubekommen, wenn ich die Phasen direkt am WARP umschalte und im Log gibt's eine passende Warnung :-)
````
[lp-1  ] WARN 2024/03/30 16:35:23 charger phases out of sync: expected 3, got 1
````

andig, 2024-03-30:
So hätte ich das erwartet. Falls GetPhases == 0 sollte man das nicht machen, den Status gibts beim Start von evcc.

poohnet, 2024-03-30:
Ok, ist drin. Ich werde das in den nächsten Tagen mal intensiv testen, insbesondere hinsichtlich Reboot von WARP, WEM und EVCC.

Frohe Ostern!

andig, 2024-03-31:
@poohnet magst Du einen PR machen?

poohnet, 2024-03-31:
Kann ich gerne machen. Ist das denn so schon ausreichend oder brauchen wir vorher noch Doku und/oder Tests?

PR #13230 (2024-03-31): Warp: validate charger phases
Fix #13201
closed 2024-03-31 by andig