evcc-io/evcc #13212: MQTT: Topic evcc/loadpoints/X/vehicleTitle is not longer published

cschlipf, 2024-03-29:
### Describe the bug

Previously the currently connected vehicle title could be read through MQTT by reading evcc/loadpoints/X/vehicleTitle. In 0.124.10 this is no longer published. Only vehicleName is still there.

### Steps to reproduce

1. Launch MQTT Explorer
2. Connect a vehicle to your loadpoint.
3. Look for vehicleTitle under the Loadpoints. It's missing now.

### Configuration details

```yaml
.
```

### Log details

```text
.
```

### What type of operating system are you running?

Linux

### Version

0.124.10

cschlipf, 2024-03-29:
Same for minSoc

VolkerK62, 2024-03-30:
Das läuft jetzt über den Namen
`evcc/loadpoints/X/vehicleName`   

und darüber findest du auch "minSoc"
`evcc/vehicles/"vehicleName"/minSoc`

cschlipf, 2024-03-30:
Hi @VolkerK62 

habe ich beides gefunden. Hilft mir aber nicht viel.

- vehicleName ist der interne Identifier, nicht der Title für den End-User
- minSoc am vehicle bringt mir nicht viel, wenn ich herausfinden muss, wie hoch der minSoc von dem Fahrzeug ist, das gerade an diesem Loadpoint angekoppelt wurde (deswegen habt ihr ja z.B. auch limitSoc am Loadpoint - derselbe Use Case)

Ja, natürlich könnte ich nun mit eigenem Code umgehen und über den vehicleName den Title und den minSoc herausfinden. Geht halt in Home Assistant dann nicht so einfach in einer Integration, da ich dann noch dynamisch eine MQTT Entity erzeugen müsste oder irgendwie direkt auf MQTT Topic lesen müsste per Code und den Wert in eine eigene Entity schreiben. Ein enormer Rückschritt für eine einfache Automation oder ein Dashboard, das die Werte pro Loadpoint versucht anzuzeigen.

sahomm, 2024-03-30:
Ich kann hier @cschlipf nur beipflichten. 

Die Integration in IOBroker ist hier leider in gleichem Maße betroffen. 

EVCC ist ein sehr zuverlässiges Mittel und eröffnet somit so viele Möglichkeiten alles noch smarter zu machen, wäre schade wenn diese Möglichkeiten somit sterben würden. 

Beste Grüße und schöne Osterfeiertage

Saho

cschlipf, 2024-03-30:
Scheint als hätte der Change auch Auswirkung auf die Message Templates. Auch hier scheint vehicleTitle nicht mehr als Parameter verfügbar zu sein:

[image]

andig, 2024-03-31:
Also Duplikat zu https://github.com/evcc-io/evcc/issues/12862?

cschlipf, 2024-03-31:
> Also Duplikat zu #12862?

Sehe es nicht unbedingt als Duplikat, mehr als Erweiterung. Hier geht es in erster Linie um MQTT und auch um minSoc, nicht nur vehicleTitle. Aber ja, da scheint es eine Überschneidung zu geben.

andig, 2024-03-31:
Die Überschneidung ist `vehicleTitle`. Unklar warum das manchmal funktionieren sollte. Alle Vehicle Variablen müssen jetzt fürs Messaging künstlich erzeugt werden. Machen wir in verlinktem Issue.

closed 2024-03-31 by andig

andig, 2024-03-31:
Die Idee wäre, alle Variablen die mit vehicle anfragen von dort zu suchen 🔦. Entweder sls Variable oder als Funktion; ich muss mal schauen was geht.

andig, 2024-03-31:
Siehe https://github.com/evcc-io/evcc/pull/13229

andig, 2024-03-31:
> In 0.124.10 these are no longer published.

To summarise: they are published under the vehicle. If anything is still missing please let me know.

cschlipf, 2024-03-31:
> > In 0.124.10 these are no longer published.
> 
> To summarise: they are published under the vehicle. If anything is still missing please let me know.

D.h. es ist nicht mehr wie früher möglich, direkt am Loadpoint die Daten von verbundenen Vehicle abzufragen?

Ich verstehe ja, dass ihr die Daten nicht duplizieren wollt. Wie ich versucht es oben zu beschrieben, ist das aber in Tools wie HA mit erheblichen Mehraufwand verbunden. Ich habe keine Ahnung wie ich nun das alte Verhalten realisieren kann ohne mit Code eine eigene Integration zu schreiben, da nun deutlich mehr Logik benötigt wird.

Früher konnte ich die Daten des verbunden Vehicles einfach am Loadpoint abfragen. Jetzt muss ich über den Vehicle Name das entsprechende Vehicle in der Vehicle Liste finden und die Daten dort auslesen. Für mich ein leider ein drastischer Rückschritt der Benutzbarkeit der MQTT API :-(

cschlipf, 2024-03-31:
Beispiel einer Automation: Ich will den Hausspeicher abschalten, wenn das Auto im PV modus unter minSoc ist. Also habe ich bisher geschaut ich ob der soc des Fahrzeuges am Loadpoint und dem minSoc des Fahrzeuges am selben Loadpoint ist. Falls ja, habe ich den Speicher abgeschaltet.

Diese Automation ist nun kaputt. SOC hängt jetzt nur am Loadpoint, minSoc hängt nur am Vehicle - wie finde ich nun heraus, welcher Loadpoint zum Vehicle gehört? Mit etwas Scripting möglich - aber ging vorher einfach ohne.

Noch blöder: vehicleName ist Teil vom MQTT topic. D.h. man muss in der MQTT Integration von Home Assistant immer MQTT Entities je nach vehicleName definieren. Richtig doof: In Home Assistant hast Du Zugriff auf den Entity Name - aber nicht darauf auf welchen MQTT Topic die Entity Definition basiert. Ich kann also nicht einmal irgendein Entity nach den Vehicle Namen fragen um das Vehicle herauszufinden:
[image]

Sorry, ich bin da jetzt völlig lost, was ich nun machen kann um wieder auf den minSoc des mit dem Loadpoint verbunden Fahrzeuges zu kommen.

andig, 2024-03-31:
> D.h. es ist nicht mehr wie früher möglich, direkt am Loadpoint die Daten von verbundenen Vehicle abzufragen?

So isses.

cschlipf, 2024-03-31:
Sehr schlecht :-(

Gibt es dann wenigstens die Chance, dass wir Daten, die zum Vehicle gehören auch alle zu diesem einen Vehicle hinzufügen anstatt sie über Loadpoint und Vehicle zu verstreuen? Dann hätte ich zumindest die Möglichkeit zu schauen, ob ein VehicleSOC eines Fahrzeuges unter seinem minSoc ist. So wie das aktuell verteilt ist, ist das aber wirklich sehr schlecht.

Ist halt nicht nur HA. Es ist generell ein Problem, wenn man MQTT benutzen will mit Daten, die zueinander gehören, nun aber lose gekoppelt verteilt sind.

andig, 2024-03-31:
> Sehr schlecht :-(

Ja, man könnte stattdessen auch einfach gar nichts ändern. Es ist halt ein Zwischenschritt. Perspektivisch wird alles ans vehicle wandern. Die Diskussion um v0.x hatten wir oft genug, die wärme ich jetzt nicht wieder auf.

cschlipf, 2024-03-31:
OK, wenn denn dann auch vehicleSoc, vehicleRange, vehicleTargetSoc, vehicleOdometer nach vehicle wandern, kann ich damit wieder leben. Hoffentlich bald. Bis dahin ist die Automation, die ich auch mit anderen share kaputt.

Was dann noch unschön ist, dass der vehicleName Teil des Topics ist. Um seine Konfiguration mit anderen zu teilen, wäre es sinnvoller auch bei Vehicles wie bei Loadpoints eine Numerierung zu haben und der name wandert dort dann in ein Topic.

Macht es Sinn dafür einen Issue einzugeben, oder wird der eh rejected, weil es von Euch eine bewusste Idee war den vehicleName ins Topic reinzuschreiben?

cschlipf, 2024-03-31:
Und ändern ist OK. Aber bitte als 'Breaking change' markieren.

andig, 2024-03-31:
Das ist anscheinend sinnlos: https://github.com/evcc-io/evcc/pull/11181.

binderth, 2024-04-08:
Bitte diese Anpassungen als "breaking changes" kennzeichnen. mit v0.125.0 ist die API wieder geändert worden.

andig, 2024-04-08:
@binderth nicht dass ich wüsste.

cschlipf, 2024-04-08:
@binderth Von 0.124.10 auf 0.125 sind mir bisher keine Änderungen aufgefallen. Was wurde denn geändert?

binderth, 2024-04-08:
@andig der Kommentar oben bezog sich ja auf pre 0.125.0
@cschlipf ich habe seit v0.125.0 die Änderung von `loadpoints[0].vehicleTargetSoc` auf `loadpoints[0].limitSoc`

cschlipf, 2024-04-08:
> @andig der Kommentar oben bezog sich ja auf pre 0.125.0 @cschlipf ich habe seit v0.125.0 die Änderung von `loadpoints[0].vehicleTargetSoc` auf `loadpoints[0].limitSoc`

Das war schon in 0.124.10 so.

Wegen den laufenden Änderungen habe ich ein HA Dashboard, das ich nun nach jedem Update prüfe. Hier ist meine MQTT Topic Konfiguration für HA seit 0.124.10: https://github.com/ha-puzzles/homeassistant-puzzlepieces/tree/main/installation/evcc-mqtt-integration

cschlipf, 2024-04-08:
Ja, MQTT Topic Änderungen werden leider nie als Breaking Chance gekennzeichnet. Deswegen habe ich das hier auch kritisiert und bin da voll und ganz Deiner Meinung.