evcc-io/evcc #13214: SMA Tripower Smart Energy - battery kann nicht über gui konfiguriert werden

koaschten, 2024-03-29:
### Describe the bug

Ich stelle gerade die Konfiguration von evcc.yaml zu GUI um und wollte das Battery Meter hinzufügen.
via yaml wird sauber SoC und Power ausgewertet.
[image]

Habe dies dann in der evcc.yaml via # auskommentiert und wollte es dann via gui hinzufügen.

Bei klick auf prüfen wird aber weder SoC noch Power angezeigt. Auch nach Speichern und Neustart sieht es noch wie unten im Bild aus.
[image]

Disclaimer: es werden auch für pv (sunny boy 3 und Sunny TriPower 8 Smart Energy) keine Werte geholt, kwh und kw sind 0, gut das könnte daran liegen, dass es aktuell draussen dunkel ist, ich werde es morgen bei Solar Produktion noch mal testen.

PS: wäre toll wenn man im GUI auch einen Display Namen angeben könnte.

### Steps to reproduce

1. Bestehende battery meter config in yaml kommentieren und yaml speichern
2. evcc neu starten
3. battery über gui hinzufügen
4. prüfen und speichern
5. neustarten
6. es werden keine battery meter Daten angezeigt

### Configuration details

```yaml
- name: my_battery #STP8SE
  type: template
  template: sma-hybrid
  usage: battery
  # Modbus TCP
  modbus: tcpip
  id: 3
  host: 192.168.1.41 # Hostname
  port: 502 # Port
```

### Log details

```text
wenn ich evcc mit --log debug starte werden keine via GUI konfigurierten Geräte angezeigt im Config GUI

pi@evcc:/etc $ sudo systemctl stop evcc
pi@evcc:/etc $ evcc --log debug -c /etc/evcc.yaml
[main  ] INFO 2024/03/29 22:04:17 evcc 0.124.10 (6446d920)
[main  ] INFO 2024/03/29 22:04:17 using config file: /etc/evcc.yaml
[main  ] INFO 2024/03/29 22:04:17 starting ui and api at :7070
[db    ] INFO 2024/03/29 22:04:18 using sqlite database: /home/pi/.evcc/evcc.db
[lp-1  ] WARN 2024/03/29 22:04:20 deprecated: mincurrent setting is ignored, please remove
[lp-1  ] WARN 2024/03/29 22:04:20 deprecated: maxcurrent setting is ignored, please remove
[lp-1  ] DEBUG 2024/03/29 22:04:21 charge total import: 799.270kWh
[semp  ] WARN 2024/03/29 22:04:21 SEMP_BASE_URL unspecified, using http://192.168.1.98:7070 instead
[site  ] INFO 2024/03/29 22:04:21 site config:
[site  ] INFO 2024/03/29 22:04:21   meters:      grid ✓ pv ✗ battery ✗
[site  ] INFO 2024/03/29 22:04:21     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2024/03/29 22:04:21   vehicles:
[site  ] INFO 2024/03/29 22:04:21     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[site  ] INFO 2024/03/29 22:04:21     vehicle 2: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[lp-1  ] INFO 2024/03/29 22:04:21 loadpoint 1:
[lp-1  ] INFO 2024/03/29 22:04:21   mode:        off
[lp-1  ] INFO 2024/03/29 22:04:21   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2024/03/29 22:04:21   meters:      charge ✓
[lp-1  ] INFO 2024/03/29 22:04:21     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2024/03/29 22:04:21 phase timer inactive
[lp-1  ] DEBUG 2024/03/29 22:04:21 pv timer inactive
[site  ] WARN 2024/03/29 22:04:21 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2024/03/29 22:04:21 ----
[lp-1  ] DEBUG 2024/03/29 22:04:21 charge power: 0W
[site  ] DEBUG 2024/03/29 22:04:21 grid meter: -4W
[site  ] DEBUG 2024/03/29 22:04:21 grid powers: [96 -136 36]W
[site  ] DEBUG 2024/03/29 22:04:21 grid currents: [0.777 -0.901 0.711]A
[site  ] DEBUG 2024/03/29 22:04:21 pv power: 54W
[site  ] DEBUG 2024/03/29 22:04:21 site power: 46W
[lp-1  ] DEBUG 2024/03/29 22:04:21 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/03/29 22:04:21 charge total import: 799.270kWh
[lp-1  ] DEBUG 2024/03/29 22:04:21 charger status: A
[lp-1  ] INFO 2024/03/29 22:04:21 car disconnected
[lp-1  ] DEBUG 2024/03/29 22:04:21 set session soc limit: 0
[lp-1  ] DEBUG 2024/03/29 22:04:21 set session energy limit: 0
[site  ] DEBUG 2024/03/29 22:04:31 ----
[lp-1  ] DEBUG 2024/03/29 22:04:31 charge power: 0W
[site  ] DEBUG 2024/03/29 22:04:31 grid meter: 0W
[site  ] DEBUG 2024/03/29 22:04:31 grid powers: [94 -133 40]W
[site  ] DEBUG 2024/03/29 22:04:31 grid currents: [0.763 -0.889 0.718]A
[site  ] DEBUG 2024/03/29 22:04:31 pv power: 50W
[site  ] DEBUG 2024/03/29 22:04:31 site power: 50W
[lp-1  ] DEBUG 2024/03/29 22:04:31 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/03/29 22:04:31 charge total import: 799.270kWh
[lp-1  ] DEBUG 2024/03/29 22:04:31 charger status: A
[site  ] DEBUG 2024/03/29 22:04:41 ----
[lp-1  ] DEBUG 2024/03/29 22:04:41 charge power: 0W
[site  ] DEBUG 2024/03/29 22:04:41 grid meter: 5W
[site  ] DEBUG 2024/03/29 22:04:41 grid powers: [94 -133 44]W
[site  ] DEBUG 2024/03/29 22:04:41 grid currents: [0.765 -0.891 0.744]A
[site  ] DEBUG 2024/03/29 22:04:41 pv power: 45W
[site  ] DEBUG 2024/03/29 22:04:41 site power: 55W
[lp-1  ] DEBUG 2024/03/29 22:04:41 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/03/29 22:04:41 charge total import: 799.270kWh
[lp-1  ] DEBUG 2024/03/29 22:04:41 charger status: A
[site  ] DEBUG 2024/03/29 22:04:51 ----
[lp-1  ] DEBUG 2024/03/29 22:04:51 charge power: 0W
[site  ] DEBUG 2024/03/29 22:04:51 grid meter: 14W
[site  ] DEBUG 2024/03/29 22:04:51 grid powers: [99 -133 49]W
[site  ] DEBUG 2024/03/29 22:04:51 grid currents: [0.781 -0.89 0.763]A
[site  ] DEBUG 2024/03/29 22:04:51 pv power: 36W
[site  ] DEBUG 2024/03/29 22:04:51 site power: 64W
[lp-1  ] DEBUG 2024/03/29 22:04:51 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/03/29 22:04:51 charge total import: 799.270kWh
[lp-1  ] DEBUG 2024/03/29 22:04:51 charger status: A
[site  ] DEBUG 2024/03/29 22:05:01 ----
[lp-1  ] DEBUG 2024/03/29 22:05:01 charge power: 0W
[site  ] DEBUG 2024/03/29 22:05:01 grid meter: -5W
[site  ] DEBUG 2024/03/29 22:05:01 grid powers: [96 -133 32]W
[site  ] DEBUG 2024/03/29 22:05:01 grid currents: [0.776 -0.893 0.733]A
[site  ] DEBUG 2024/03/29 22:05:01 pv power: 55W
[site  ] DEBUG 2024/03/29 22:05:01 site power: 45W
[lp-1  ] DEBUG 2024/03/29 22:05:01 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/03/29 22:05:01 charge total import: 799.270kWh
[lp-1  ] DEBUG 2024/03/29 22:05:01 charger status: A
[site  ] DEBUG 2024/03/29 22:05:11 ----
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.124.10 (6446d920)

naltatis, 2024-03-30:
Thema Titel-Feld ist bereits in https://github.com/evcc-io/evcc/issues/6781 adressiert @andig 

Wir haben aktuell noch einen Bug, dass Default-Werte nicht automatisch übernommen werden. Siehe https://github.com/evcc-io/evcc/issues/11934
Kannst du mal versuchen, ob es bei dir funktioniert, wenn du alle Felder selbst ausfüllst?

koaschten, 2024-04-01:
> Thema Titel-Feld ist bereits in #6781 adressiert @andig
> 
> Wir haben aktuell noch einen Bug, dass Default-Werte nicht automatisch übernommen werden. Siehe #11934 Kannst du mal versuchen, ob es bei dir funktioniert, wenn du alle Felder selbst ausfüllst?

Ich hab noch mal versucht das Battery Meter des Sunny TriPower Smart Energy hinzuzufügen als  SMA Smart Energy Hybrid Wechselrichter, jedes Feld gelöscht im Gui und von Hand ausgefüllt mit den entsprechenden Werten die ich in der YAML benutzt habe. Funktioniert leider auch nicht.

koaschten, 2024-04-01:
> > Funktioniert leider auch nicht.
> 
> Das ist recht unspezifisch ;)

Was ist daran unspezifisch? Die Frage war, funktioniert es wenn du von Hand die Default Werte ausfüllst, daraufhin habe ich genau das gemacht und geantwortet das es dann auch nicht funktioniert?

Gerne helfe ich bei der Fehlersuche. Allerdings debugge ich nicht täglich EVCC und ein zielgerichtetes "mach mal bitte das" wäre hilfreich(er).

andig, 2024-04-21:
@naltatis was brauchst du?

andig, 2024-04-21:
> "mach mal bitte das" wäre hilfreich(er).

@koaschten schau mal bitte in die Entwicklerkonsole/Netzwerk was da an JSON geschickt wird wenn Du auf speichern clickst.

Generell hilft immer eine Fehlermeldung oder konkrete Beobachtung, nicht nur die Abwesenheit von "geht" ;)

closed 2024-05-16 by andig

koaschten, 2024-05-18:
Soo, ich bin gerade mal wieder vor Ort. Habe einen Sunnyboy3, einen STP15000TL und den STP8SE eingebunden. Das funktioniert jetzt.

Was mich irritiert, muss ich den STP8SE als Battery und PV einbinden? Wenn ich das mache meckert EVCC während des Batterie laden über negative Erzeugungswerte und selbst wenn man den Battery Treshold unterhalb des aktuellen SOC setzt versucht EVCC nicht mehr der Batterie den Ladestrom zu klauen.
[image]

Und selbst wenn am Gridmeter -3,5kW anliegen dreht EVCC die wallbox nicht weiter hoch
[image]
[image]

und ja, da ist eine Differenz von ein paar kW (3,5kWp je Ost/West), weil zwei alte Wechselrichter ohne Auslesemöglichkeit noch am Anschlusspunkt hängen.

```
[site ] DEBUG 2024/05/18 08:47:35 ----
[lp-1 ] DEBUG 2024/05/18 08:47:35 charge power: 8210W
[lp-1 ] DEBUG 2024/05/18 08:47:35 charge currents: [12.7 12.7 12.7]A
[site ] DEBUG 2024/05/18 08:47:35 pv power: 10343W
[site ] DEBUG 2024/05/18 08:47:35 battery soc: 100%
[site ] DEBUG 2024/05/18 08:47:35 battery power: 3447W
[site ] DEBUG 2024/05/18 08:47:35 grid meter: -3871W
[site ] DEBUG 2024/05/18 08:47:35 grid powers: [-390 -2468 -1013]W
[site ] DEBUG 2024/05/18 08:47:35 grid currents: [-2.31 -11.3 -4.73]A
[site ] DEBUG 2024/05/18 08:47:35 site power: -374W
```
Chargepower: 8210
PV Power: 10343
Grid -3871

es werden also grundlos 3,8kW ins Netz geschoben? und aus irgendeinem Grund die Erzeugung des Hybridwechselrichters trotz SOC 100 als battery power mitgerechnet und deswegen die Ladeleistung runtergeregelt? sehr kurios.

Ich bin heute am Samstag noch bis 16:00 vor Ort.

andig, 2024-05-18:
Ich verstehe nur Bahnhof. Die Regelung ist nur vom Gridmeter abhängig. Und dem ist egal, was dahinter hängt.

koaschten, 2024-05-18:
wenn die Regelung nur vom Grid Meter abhängt
- wieso wird die Batterie geladen, wenn die Batterie Steuerung in EVCC wie folgt eingestellt ist und der Hausbatterie SOC über 80% ist? Zugleich aber die Wallbox noch nicht die 11kW erreicht hat? eigentlich müsste EVCC doch hochregeln bis die 11kW der Wallbox erreicht sind, solange der Hausbatterie SoC über 50% ist?
[image]

- Wie trage ich über die GUI einen SMA Hybrid Wechselrichter korrekt ein? Einmal als PV und einmal als Batterie? Nur als Batterie? Aktuell wird halt die Leistung des Hybridwechselrichters als Batterie Entladen angezeigt obwohl die Batterie voll ist und es PV Erzeugung sein müsste. 
[image]

- und woher kommen diese 3,x kW Eigenverbrauch aus dem vorherigen Screenshot? Laut SHM sind die beiden großen Verbraucher Wärmepumpe und Wallbox aktuell nicht aktiv in der Live Ansicht der Verbraucheransicht.
[image]

-----
Und um eventuell noch etwas mehr Verwirrung rauszunehmen, die Anlage sieht wie folgt aus:
SMA STP15000TL, 2 Strings, Ost/West (als PV über GUI angelegt)
SMA STP8SE 2 Strings, Süd mit 16.6kWh Batterie (als PV und Batterie über GUI angelegt)
SMA SB3 1 String Süd (als PV über GUI angelegt)
Sunny Home Manager am Hausanschluss (Grid Meter)

Zwei große Verbraucher im SHM eingebunden Wallbox (via EVCC) und Vaillant Wärmepumpe

Sowie 2 nicht gemonitorte Wechselrichter mit jeweils ca. 3,5kWP Ost und West
-----

Ich bin heute und morgen bis jeweils 16:00 an der Anlage. Und bin wirklich interessiert das zu verstehen und eventuelle Haken die drin sind in der Konfiguration zu verstehen und zu korrigieren.

andig, 2024-05-18:
Das gehört besser in eine Diskussion- hier ging es um GUI-Konfiguration.

koaschten, 2024-05-18:
ok. ich mach dann ein neuen Thread in den Diskussionen auf und verweise auf meinen vorherigen Post?