evcc-io/evcc #13215: Warp 3 Charger von Tinkerforge über Modbus TCP

StevieC121176, 2024-03-29:
Wäre es möglich den neuen Warp 3 Charger von Tinkerforge in EVCC zu implementieren? Soweit ich gelesen hab ist dieser nun über Modbus TCP steuerbar (Doku der Register sh. Link ab Seite 32) Die Phasenumschaltung sollte nun ohne Warp Energy Manager funktionieren.

https://warp-charger.com/documents/WARP3_Betriebsanleitung.pdf

andig, 2024-03-30:
Läuft der mit dem Warp2 Template nicht?

StevieC121176, 2024-03-30:
Testen kann ich das erst kw16 wenn ich ihn bekomme, allerdings geht da die Phasenumschaltung jetzt ohne Warp Energy Manager. Sollte das Template dann trotzdem funktionieren?

andig, 2024-03-30:
Warum sollte es nicht? Gibts neue Topics?

StevieC121176, 2024-03-30:
Die Seite wird noch überarbeitet. 
meine Überlegung war, als Endanwender wäre es vielleicht einfacher und besser ohne noch ein zusätzliches Programm wie mosquitto zu installieren. Das ist allerdings mein persönliches empfinden.
In kw16 bekomme ich den Warp 3 charger pro, dann kann ich das alte Template probieren.

poohnet, 2024-03-30:
Das alte Template sollte weiter funktionieren, man muss nur für "topic" und "energymanager" das gleiche Topic angeben ("warp2/ABC")...

LeoWinterDE, 2024-03-30:
Nur die Doku müsste ggf. angepasst werden.

Rel #13193

andig, 2024-04-12:
> Das alte Template sollte weiter funktionieren, man muss nur für "topic" und "energymanager" das gleiche Topic angeben ("warp2/ABC")...

Ein PR für ein vereinfachtes Warp3 Template wäre Klasse.

StevieC121176, 2024-04-12:
Es wäre klasse wenn es ohne sich Zusatzsoftware wie mosquitto zu installieren funktionieren würde. (via Modbus z.Bsp.)

andig, 2024-04-12:
Gerne PR. Da es ohne funktioniert keine Prio für uns. Beispiele dafür gibts in Hülle und Fülle, jemand müsste einfach nur die richtigen Register in einer neuen Datei eintragen.

poohnet, 2024-04-13:
> Ein PR für ein vereinfachtes Warp3 Template wäre Klasse.

Möchtest du dafür ein komplett eigenes Template? Letztendlich wäre es m. E. ausreichend, wenn man im vorhandenen Template für WARP2 für die Phasenumschaltung weiterhin das WEM-Topic (Parameter "energymanager") angibt und für WARP3 einen neuen bool-Parameter "warp3" o. ä. spendiert.

> Es wäre klasse wenn es ohne eine Zusatzsoftware wie mosquitto zu installieren funktionieren würde. (via Modbus z.Bsp.)

Wenn ich das richtig sehe, dann funktioniert die externe Phasenumschaltung (noch) nicht per Modbus TCP.

StevieC121176, 2024-04-14:
> > Es wäre klasse wenn es ohne eine Zusatzsoftware wie mosquitto zu installieren funktionieren würde. (via Modbus z.Bsp.)
> 
> Wenn ich das richtig sehe, dann funktioniert die externe Phasenumschaltung (noch) nicht per Modbus TCP.

Ja da hast du Recht. Wenn ich mir die Modbus Register anschaue findet man da nichts von Phasenumschaltung. Ich werde das mal ansprechen im entsprechenden Forum.
Ich hoffe ich bekomme meine Box nächste Woche das ich das testen kann.

poohnet, 2024-04-14:
Hab gestern Abend schon nachgefragt und den Tipp bekommen, die HTTP-API zu verwenden.

StevieC121176, 2024-04-14:
> Hab gestern Abend schon nachgefragt und den Tipp bekommen, die HTTP-API zu verwenden.

Ja , laut Doku kann man damit am meisten machen, sogar die RGB LED steuern und somit vielleicht sogar den EVCC Modus und die Ladegeschwindigkeit visualisieren. Da die Box kein Display hat wäre das sicher nicht schlecht.

andig, 2024-04-14:
> Ja da hast du Recht. Wenn ich mir die Modbus Register anschaue findet man da nichts von Phasenumschaltung. Ich werde das mal ansprechen im entsprechenden Forum.

Wir machen mal zu bis es da was brauchbares gibt.

closed 2024-04-14 by andig