evcc-io/evcc #13226: Offer API to either get charging history or amount of charged energy per vehicle

NobodysNightmare, 2024-03-30:
**Is your feature request related to a problem? Please describe.**
So far I've been charging my one and only car through a simple wall plug attached to a Fritz DECT Meter, and collected usage metrics this way. I have some own tooling that will eventually tell me how much electricity and gasoline went into the car over a given time period and how much of the electricity was from my PV. Whenever I needed to charge the car of a friend, I'd unplug the Fritz Meter in the middle, so that their electricity would not count towards statistics of my own car.

Soon I'll upgrade to using a proper wallbox (a warp 3) and using EVCC. I already saw that "guest car" is a built-in feature of EVCC, as well as a log of charging sessions. However, what seems to be missing (as far as I read the docs) is a way to automatically/programmatically extract corresponding information.

**Describe the solution you'd like**
I could imagine two kinds of APIs that I could utilize to meet my needs.

**Option 1: A "meter-like" API per vehicle**

This would be an API that allows querying the total consumed Watt-Hours for a given vehicle right now. E.g. the "meter" would stand still while the car is not charging (or not even connected), and increase while it's charging. Something along the lines of (MQTT):

````
evcc/vehicles/<name>/total_charged_energy
````

This solution would be a drop-in replacement for what I'm currently using. I'd not even need EVCCs abilities to judge how much PV-juice and how much grid-juice went into the car, since the rest of my tools could figure that out anyways. I guess a pro of that API is that it could be pretty universal (to be point of being usable by other "meter readers" as well). But a con might be, that it might not be compatible with the data that EVCC is storing right now (judging from docs mostly), e.g. I am not sure if a running total of Wh per vehicle is even readily available yet.

**Option 2: An API to fetch charging sessions for a given vehicle**

This solution might be closer to the current design of EVCC. If I could fetch charging sessions programmatically (ideally per vehicle filtering by a timeframe), I could also get all the data I need. In this case I'd be dependent on EVCCs ability to calculate the amount of pv-energy vs. grid-energy, because I wouldn't know when exactly the car got how much energy. Using those I'd also be able to figure out cost per kWh (and I think EVCC could do that as well, if properly configured).

This would be thinkable as:

````
GET /api/loadpoints/<id>/history?from=2024-01-01&to=2024-02-01&vehicle=prius
````

(query parameters optional, but they bring along the idea of filtering that I'd otherwise do on the client side)

**Describe alternatives you've considered**
One cheap alternative (the one I'll be going with until I find a better solution) is to just read the meter of the wallbox and not help friends (or only handing them the Schuko charger). I might also be lucky with the Warp chargers abilities to track charges, but I didn't yet dig into this deeply.

naltatis, 2024-03-30:
Have you looked at `/api/sessions`? I think it solves all of your needs. You'd have to do client side filtering. But at a home use scale this should not be an issue.

For pure time-based stats, take a look at this issue https://github.com/evcc-io/evcc/issues/13050

closed 2024-03-30 by naltatis

NobodysNightmare, 2024-03-30:
Thank you. I definitely missed this endpoint so far. It will probably solve
my needs.

Thanks!