evcc-io/evcc #13242: Bluelink Vehicle Identification Status Updates not working correctly

h4sk33r, 2024-03-31:
### Describe the bug

Hello, 

I have been running evcc for quite some time now. Been driving a Hyundai Kona 2020 so far. Everything was working fine. 
evcc identified the car during charging. The SOC was displayed correctly and charging limits were used.

Now i recently got a new Hyundai Kona 2023. I have removed the "old" Kona from my .yaml and added the new one. Now, however neither vehicle recognition or other status updates are working. Looking in the logfiles the Car-SOC and charging target are reported as 0%, the range is not available at all and vehicle recognition is not automatically detecting the car either.

It's the only car in the evcc.yaml

I have since been trying to change the configuration based on other bugs/solutiuons reported here. e.g increasing log-levels and deactivation SOC-Estimation.

Looking at the logs, attached below, it looks like the Kona does not switch into the correct status as it is reported as status A, while the charger is in B. While the car is attached to the charger right now.

I have setup a linux system just to reproduce the error. Logs are from this system and not HomeAssistant where evcc is normally running.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
levels:
  site: debug
  lp-1: trace
  lp-2: debug
  cache: debug
  db: error

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for documentation see https://docs.evcc.io/docs/devices/meters
meters:
  - name: pv
    type: custom
    power:
      source: mqtt
      topic: evccinput/sensor/tlj2bkm02g_pv_all_power/state
    energy:
      source: mqtt
      topic: evccinput/sensor/solarenergy_produced/state
  - name: grid
    type: custom
    power:
      source: mqtt
      topic: evccinput/sensor/evcc_grid_power/state
  - name: battery
    type: custom
    power:
      source: mqtt
      topic: evccinput/sensor/evcc_battery_power/state
    soc:
      source: mqtt
      topic: evccinput/sensor/tlj2bkm02g_state_of_charge/state
    capacity: 7.5

# charger definitions
# name can be freely chosen and is used as reference when assigning charger to vehicle
# for documentation see https://docs.evcc.io/docs/devices/chargers
chargers:
  - name: Wallbox
    type: template
    template: easee
    user: deprecated
    password: deprecated
[... 226 lines cut ...]
[site  ] DEBUG 2024/03/31 21:17:41 grid meter: 360W
[site  ] DEBUG 2024/03/31 21:17:41 site power: 560W
[lp-1  ] DEBUG 2024/03/31 21:17:41 charge currents: [0.008 0.009 0.011]A
[lp-1  ] DEBUG 2024/03/31 21:17:41 charge total import: 1523.381kWh
[lp-1  ] DEBUG 2024/03/31 21:17:41 charger status: B
[main  ] DEBUG 2024/03/31 21:17:41 vehicle status: A (Kona SX2)
[lp-1  ] DEBUG 2024/03/31 21:17:43 max charge current: 16A
[site  ] DEBUG 2024/03/31 21:18:11 ----
[lp-1  ] DEBUG 2024/03/31 21:18:11 charge power: 0W
[site  ] DEBUG 2024/03/31 21:18:11 pv power: 0W
[site  ] DEBUG 2024/03/31 21:18:11 battery soc: 10%
[site  ] DEBUG 2024/03/31 21:18:11 battery power: 0W
[site  ] DEBUG 2024/03/31 21:18:11 grid meter: 350W
[site  ] DEBUG 2024/03/31 21:18:11 site power: 550W
[lp-1  ] DEBUG 2024/03/31 21:18:11 charge currents: [0.008 0.009 0.011]A
[lp-1  ] DEBUG 2024/03/31 21:18:11 charge to
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.124.10

andig, 2024-03-31:
Please share

    evcc vehicle --log trace

h4sk33r, 2024-03-31:
Thank you very much for looking into it andig,

do I need to remove anything from the log for security reasons (VIN etc.)? I have never used bluelinky, so I can't tell if that one works.

best regards.

andig, 2024-04-02:
Anscheinend hat sich bei Deinem Fahrzeug das API geändert. Der Code sollte damit jetzt konzilianter umgehen. Kann ab morgen im Nightly getestet werden.

closed 2024-04-02 by andig

VolkerK62, 2024-04-03:
da fehlt der komplette Teil `vehicleStatus.evStatus`
`battery` ist die 12V Batterie

```
{
    "retCode": "S",
    "resCode": "0000",
    "resMsg": {
        "vehicleStatusInfo": {
            "vehicleStatus": {
                "windowRearCurtain": {},
                "transCond": true,
                "airCtrlOn": false,
                "engine": false,
                "doorLock": true,
                "doorOpen": {
                    "frontLeft": 0,
                    "frontRight": 0,
                    "backLeft": 0,
                    "backRight": 0
                },
                "trunkOpen": false,
                "defrost": false,
                "lowFuelLight": false,
                "acc": false,
                "ign3": false,
                "hoodOpen": false,
                "steerWheelHeatStep2": 0,
                "steerWheelHeat": 0,
                "sideBackWindowHeat": 0,
                "dte": {
                    "value": 437,
                    "unit": 1
                },
                "tirePressureLamp": {
                    "tirePressureLampAll": 0,
                    "tirePressureLampFL": 0,
                    "tirePressureLampFR": 0,
                    "tirePressureLampRL": 0,
                    "tirePressureLampRR": 0
                },
                "seatHeaterVentState": {
                    "frSeatHeatState": 2,
                    "flSeatHeatState": 2,
                    "rlSeatHeatState": 2,
                    "rrSeatHeatState": 2
                },
                "battery": {
                    "batSoc": 87,
                    "bacState": 0,
                    "sjbDeliveryMode": 2
                },
                "lampWireStatus": {
                    "stopLamp": {
                        "leftLamp": false,
                        "rightLamp": false
                    },
                    "headLamp": {
                        "headLampStatus": false,
                        "leftLowLamp": false,
                        "rigthLowLamp": false,
                        "leftHighLamp": false,
                        "rightHighLamp": false
                    },
                    "turnSignalLamp": {
                        "leftFrontLamp": false,
                        "rightFrontLamp": false,
                        "leftRearLamp": false,
                        "rightRearLamp": false
                    }
                },
                "windowOpen": {
                    "frontLeft": 0,
                    "frontRight": 0,
                    "backLeft": 0,
                    "backRight": 0
                },
                "windowOpen2": {
                    "frontLeft": 0,
                    "frontRight": 0,
                    "backLeft": 0,
                    "backRight": 0
                },
                "tailLampStatus": 0,
                "hazardStatus": 0,
                "smartKeyBatteryWarning": false,
                "fuelLevel": 0,
                "washerFluidStatus": false,
                "breakOilStatus": false,
                "sleepModeCheck": false,
                "time": "20240401002523",
                "airCleaning": {
                    "airPurifierStatus": 1,
                    "fineDustStep": 0,
                    "fineDustColor": 0
                }
            }
        }
    },
    "msgId": "3fa4b238-5110-4482-8438-766f73c22073"
}
```

h4sk33r, 2024-04-03:
Guten Morgen, 
kann ich noch irgendwie helfen. Braucht ihr noch spezifische Logfiles oder ähnliches?

h4sk33r, 2024-04-03:
Bluelinky hat scheinbar auch noch keine Lösung:
https://github.com/Hacksore/bluelinky/issues/271

VolkerK62, 2024-04-03:
@andig  hilft das weiter?   https://github.com/Hyundai-Kia-Connect/hyundai_kia_connect_api/pull/480

andig, 2024-04-03:
That should help! I will need login credentials for testing, please email to info@evcc.io.

h4sk33r, 2024-04-03:
> That should help! I will need login credentials for testing, please email to [info@evcc.io](mailto:info@evcc.io).

Done. Thanks for testing.