evcc-io/evcc #13245: Get request vehicle types (openWB) does not return list

Bucky2k, 2024-04-01:
### Describe the bug

OpenWB requests the types of vehicles supported by EVCC by http:  

https://cloud.evcc.io/api/vehicles

Currently, this does not return any data

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
OpenWB 1.9
```

### Log details

```text
OpenWB, no relevant log
```

### What type of operating system are you running?

Linux

andig, 2024-04-01:
@Bucky2k gibts irgendeine Antwort oder Timeout? Statuscode?

Bucky2k, 2024-04-01:
Leider sind die Logs leer. Nur diese Nachricht nach etwa 10 Sekunden:

[image]

Bucky2k, 2024-04-01:
[image]

Bucky2k, 2024-04-01:
Hier die Abfrage:
```
$('#soc_evcc_load_vehicles_lp1').click(function(){
										$('#soc_evcc_load_vehicles_lp1').removeClass("bg-danger");
										$('#soc_evcc_load_vehicles_lp1').removeClass("bg-success");
										$('#soc_evcc_load_vehicles_lp1').addClass("bg-warning");
										$(this).find('.fa-sync').addClass('fa-spin');
										$("#soc_evcc_select_vehicle_lp1").empty();
										$.ajax({
											type: "GET",
											url: "https://cloud.evcc.io/api/vehicles",
											success: function(vehicledata){
												$('#soc_evcc_load_vehicles_lp1').removeClass("bg-warning");
												$('#soc_evcc_load_vehicles_lp1').addClass("bg-success");
												var vehicleid = $('#soc_evcc_type_lp1').val();
												$("<option/>").val('').text("-- Bitte auswählen --").appendTo('#soc_evcc_select_vehicle_lp1');
												vehicledata.forEach(function(vehicle){
													newVehicle = $("<option/>").val(vehicle.id).text(vehicle.name);
													if( vehicleid == vehicle.id ){
														newVehicle.attr('selected','selected');
													}
													newVehicle.appendTo('#soc_evcc_select_vehicle_lp1');
												});
												$('#soc_evcc_select_vehicle_lp1').attr('readonly', false);
											},
											error: function(errMsg) {
												$('#soc_evcc_load_vehicles_lp1').removeClass("bg-warning");
												$('#soc_evcc_load_vehicles_lp1').addClass("bg-danger");
												alert("Fahrzeuge konnten nicht abgerufen werden!");
											},
											complete: function(){
												$('#soc_evcc_load_vehicles_lp1').find('.fa-sync').removeClass('fa-spin');
											}```

closed 2024-04-01 by andig